project-chip/connectedhomeip #6839: Failed to establish BLE connection on M5 during provisioning

cjandhyala, 2021-05-14:
Failed to establish BLE connection on M5 during provisioning . 

M5log
----

I (1609) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0500', EndPoint ID: '0x01', Attribute ID: '0x0002'
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40I (1629) app-devicecallbacks: Unhandled cluster ID: 1280

I (1639) app-devicecallbacks: Current free heap: 134272

I (1639) chip[ZCL]: Failed to send IAS Zone update. Err 0x70
I (1649) chip[DIS]: Init admin pairing table with server storage.
V (1659) chip[IN]: TransportMgr initialized
I (1659) chip[DL]: CHIPoBLE advertising started
I (1659) chip[DL]: NVS set: chip-counters/GlobalMCTR = 1000 (0x3E8)
E (1669) chip[DL]: Long dispatch time: 815 ms
I (1679) chip[IN]: local node id is 0x0000000000bc5c01
I (1679) chip[DL]: Starting ESP WiFi layer
I (1689) phy_init: phy_version 4660,0162888,Dec 23 2020
I (1699) wifi:mode : sta (24:6f:28:8f:d7:58)
W (1709) wifi:Haven't to connect to a suitable AP now!
I (1709) chip[DL]: Done driving station state, nothing else to do...
W (1709) wifi:Haven't to connect to a suitable AP now!
I (1719) chip[DL]: Done driving station state, nothing else to do...
V (1719) chip[BLE]: Assigned local session key ID 0
V (1729) chip[BLE]: Waiting for PBKDF param request
I (1729) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable, device name CHIP-4000)
I (1739) chip[SVR]: Server Listening...
I (1749) chip[DL]: Device already advertising, stop active advertisement and restart
I (1749) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0402', EndPoint ID: '0x01', Attribute ID: '0x0000'
GAP procedure initiated: stop advertising.
I (1769) app-devicecallbacks: Unhandled cluster ID: 1026
I (1779) app-devicecallbacks: Current free heap: 130076

GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (1789) app-devicecallbacks: PostAttributeChangeCallback - Cluster ID: '0x0101', EndPoint ID: '0x01', Attribute ID: '0x0000'
I (1799) app-devicecallbacks: Current free heap: 129944

I (1809) chip[DL]: WIFI_EVENT_STA_START
W (1819) wifi:Haven't to connect to a suitable AP now!
I (1819) app-devicecallbacks: Unhandled cluster ID: 257
I (1819) chip[DL]: Done driving station state, nothing else to do...
I (1839) app-devicecallbacks: Current free heap: 129944

I (1839) app-devicecallbacks: Current free heap: 129944

I (1849) chip[ZCL]: Failed to read AutoRelockTime attribute: 0x86
I (1859) all-clusters-app: Short Manual(decimal) setup code: 34970112332
I (1859) all-clusters-app: Long Manual(decimal) setup code:  749701123309050177298
I (1869) all-clusters-app: QR CODE Text: 'CH:C8XA062600K50324800'
I (1879) all-clusters-app: Copy/paste the below URL in a browser to see the QR CODE:
	https://dhrishi.github.io/connectedhomeip/qrcode.html?data=CH%3AC8XA062600K50324800
I (36719) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name CHIP-4000)
I (36719) chip[DL]: Device already advertising, stop active advertisement and restart
GAP procedure initiated: stop advertising.
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (157459) chip[DL]: BLE GAP connection established (con 0)
I (157459) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, non-connectable, device name CHIP-4000)
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (157779) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (157789) chip[BLE]: local and remote recv window sizes = 3
I (157789) chip[BLE]: selected BTP version 3
I (157789) chip[BLE]: using BTP fragment sizes rx 128 / tx 20.
I (159999) chip[DL]: BLE GAP connection terminated (con 25473
E (159999) chip[IN]: Failed to establish BLE connection: Ble Error 6000011 (0x005B8D8B): BLE transport protocol fired abort
V (160009) chip[IN]: Clearing BLE pending packets.
I (160009) chip[BLE]: Releasing end point's BLE connection back to application.
I (160019) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable, device name CHIP-4000)
I (160029) chip[DL]: Device already advertising, stop active advertisement and restart
GAP procedure initiated: stop advertising.
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800

Chip-tool
----------

% ./out/host/chip-tool pairing ble-wifi MYSSID MYPASSWD 87564321 20202021 4000
[1621032375251] [0x6b69b0] CHIP: [DL] _Init
[1621032375256] [0x6b69b0] CHIP: [IN] TransportMgr initialized
[1621032375256] [0x6b69b0] CHIP: [IN] local node id is 0x000000000001b669
[1621032375256] [0x6b69b0] CHIP: [ZCL] Using ZAP configuration...
[1621032375256] [0x6b69b0] CHIP: [ZCL] deactivate report event
[1621032375256] [0x6b69b4] CHIP: [DL] CHIP task running
[1621032375256] [0x6b69b0] CHIP: [BLE] NewConnection
[1621032375258] [0x6b69b0] CHIP: [BLE] Assigned local session key ID 0
[1621032375258] [0x6b69b0] CHIP: [DL] Chip stack locking error at '../../third_party/connectedhomeip/src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621032375258] [0x6b69b0] CHIP: [IN] Message appended to BLE send queue
[1621032375258] [0x6b69b0] CHIP: [BLE] Sent PBKDF param request
[1621032375315] [0x6b69b5] CHIP: [BLE] CBManagerState: ON
[1621032376734] [0x6b69b5] CHIP: [BLE] Connecting to device with discriminator: 4000
[1621032377478] [0x6b69b5] CHIP: [IN] BleConnectionComplete: endPoint 0x1094de740
[1621032377478] [0x6b69b5] CHIP: [DL] Chip stack locking error at '../../third_party/connectedhomeip/src/system/SystemLayer.cpp:204'. Code is unsafe/racy
[1621032392478] [0x6b69b4] CHIP: [BLE] connect handshake timed out, closing ep 0x1094de740
[1621032392478] [0x6b69b4] CHIP: [IN] Failed to establish BLE connection: Ble Error 6021 (0x00001785): Connect handshake timed out
[1621032392478] [0x6b69b4] CHIP: [IN] Clearing BLE pending packets.
[1621032392478] [0x6b69b4] CHIP: [BLE] BtpEngine unsub failed
[1621032392478] [0x6b69b4] CHIP: [BLE] Auto-closing end point's BLE connection.
[1621032405257] [0x6b69b4] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1621032405257] [0x6b69b4] CHIP: [BLE] PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1621032405257] [0x6b69b4] CHIP: [TOO] Secure Pairing Failed
[1621032405257] [0x6b69b0] CHIP: [CTL] Shutting down the commissioner
[1621032405257] [0x6b69b0] CHIP: [CTL] Shutting down the controller
[1621032405258] [0x6b69b4] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1621032405258] [0x6b69b0] CHIP: [DL] System Layer shutdown
[1621032405258] [0x6b69b0] CHIP: [DL] Inet Layer shutdown
[1621032405258] [0x6b69b0] CHIP: [DL] BLE layer shutdown

fengbowe, 2021-05-16:
I am seeing the same issue on ESP32-WROVER-KIT_V4.1

dhrishi, 2021-05-17:
@cjandhyala @fengbowe  Assuming you are trying with the default CHIP code (no changes), can you please try the command:
```
% ./out/host/chip-tool pairing ble-wifi MYSSID MYPASSWD 112233 20202021 3840
```
a. fabric ID is 112233
b. setup-pin-code is 20202021 by default (configurable through menuconfig)
c. discriminator is 3840 by default which is 0xF00 (configurable through menuconfig)

RahulEspressif, 2021-05-17:
Hi @cjandhyala  @fengbowe  ,  We are debugging one internal issue, which we believe has the same root cause for the connection issue being observed here. As a workaround , can you please try to disable "Host flow control" , via menuconfig  and confirm if this change helps you.

dhrishi, 2021-05-17:
>can you please try to disable "Host flow control" , via menuconfig and confirm if this change helps you.

`idf.py menuconfig` (if using CMake) or `idf make menuconfig` (if using automake) -> Component config -> Bluetooth -> NimBLE Options -> Enable Host Flow control (disable this)

fengbowe, 2021-05-17:
> > can you please try to disable "Host flow control" , via menuconfig and confirm if this change helps you.
> 
> `idf.py menuconfig` (if using CMake) or `idf make menuconfig` (if using automake) -> Component config -> Bluetooth -> NimBLE Options -> Enable Host Flow control (disable this)

This works. Thanks

dhrishi, 2021-05-21:
@fengbowe @cjandhyala The IDF release/v4.2 branch is updated with the BLE connection fix. Please pull in the latest changes on that branch, update the submodules (git submodule update --init --recursive) and test. With this, you should not require to disable the host flow control from menuconfig

dhrishi, 2021-05-25:
@cjandhyala  Can you please close this if it is working fine. Also, as mentioned above, the fix is already on the latest IDF branch `release/v4.2` so you don't need to make any changes here.

dhrishi, 2021-07-06:
@cjandhyala Closing this now, assuming it works fine

closed 2021-07-06 by dhrishi

Darshanr5051, 2023-05-30:
Hello, I am using esp-idf V5.0 and From menuconfig -> BLE-> host flow is also disabled still i am getting  BLE GAP Connection terminated (Con 0 reason 0x213) error ! Any idea what to do ?