project-chip/connectedhomeip #6845: Compilation fail for all-clusters-app on ESP32

fengbowe, 2021-05-15:
#### Problem
compilation fail during linking:

/Users/fengbowe/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/chip/lib/libCHIP.a(libCHIPDataModel.CommandHandler.cpp.o):(.literal._ZN4chip3app14CommandHandler25ProcessCommandDataElementERNS0_18CommandDataElement6ParserE+0x8): undefined reference to `chip::app::DispatchSingleClusterCommand(unsigned short, unsigned char, unsigned char, chip::TLV::TLVReader&, chip::app::Command*)'
/Users/fengbowe/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/chip/lib/libCHIP.a(libCHIPDataModel.CommandHandler.cpp.o): in function `chip::app::CommandHandler::ProcessCommandDataElement(chip::app::CommandDataElement::Parser&)':
/Users/fengbowe/Documents/CHIP/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/app/CommandHandler.cpp:101: undefined reference to `chip::app::DispatchSingleClusterCommand(unsigned short, unsigned char, unsigned char, chip::TLV::TLVReader&, chip::app::Command*)'
/Users/fengbowe/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/chip/lib/libCHIP.a(libCHIPDataModel.CommandSender.cpp.o): in function `chip::app::CommandSender::ProcessCommandDataElement(chip::app::CommandDataElement::Parser&)':
/Users/fengbowe/Documents/CHIP/connectedhomeip/examples/all-clusters-app/esp32/build/esp-idf/chip/../../../../../../config/esp32/third_party/connectedhomeip/src/app/CommandSender.cpp:162: undefined reference to `chip::app::DispatchSingleClusterCommand(unsigned short, unsigned char, unsigned char, chip::TLV::TLVReader&, chip::app::Command*)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

 #### Details

OS: Mac 10.15.7 

Repro steps:
(1) mkdir ${HOME}/tools
(2) cd ${HOME}/tools
(3) git clone https://github.com/espressif/esp-idf.git
(4) cd esp-idf
(5) git checkout release/v4.2
(6) git submodule update --init
(7) export IDF_PATH=${HOME}/tools/esp-idf
(8) ./install.sh
(9) . ./export.sh
(10) cd ${HOME}
(11) git clone https://github.com/project-chip/connectedhomeip.git
(12) cd connectedhomeip
(13) source ./scripts/bootstrap.sh
(14) source ./scripts/activate.sh
(15) cd examples/all-clusters-app/esp32
(16) idf.py menuconfig 
`Demo`->`Device Type` -> `ESP32-WROVER-KIT_V4.1`
`Demo`->`Rendezvous Mode`->`BLE`
(17) idf.py build

Full build log:
[build_log.txt](https://github.com/project-chip/connectedhomeip/files/6481612/build_log.txt)

dhrishi, 2021-05-15:
#6774

dhrishi, 2021-05-15:
@yunhanw-google Raised a PR here with @erjiaqing 's suggestions: https://github.com/project-chip/connectedhomeip/pull/6849

pan-apple, 2021-05-15:
I think @fengbowe was able to build this with the following steps (replacing step 16 onwards in the issue description)

```
1. source idf.sh
2. idf make menuconfig
Demo->Device Type -> ESP32-WROVER-KIT_V4.1
Demo->Rendezvous Mode->BLE
3. idf make flash monitor ESPPORT=/dev/tty.SLAB_USBtoUART -j 16
```

dhrishi, 2021-05-15:
@pan-apple Yes, the compilation works fine with automake (make command) but fails with CMake (idf.py). CMake would be the default one going forward and hence is recommended.

bzbarsky-apple, 2021-05-19:
@fengbowe @dhrishi Did #6849 fix this?

fengbowe, 2021-05-19:
@bzbarsky-apple Yes it's fixed now

closed 2021-05-19 by fengbowe