project-chip/connectedhomeip #6851: Source activate.sh error

JourneyYoung, 2021-05-15:
When I try to source activate.sh to build preparation for CHIP project (I use M1 Macbook Pro), I got an error  like this:

```
WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[|]Errors:
  failed to resolve infra/3pp/tools/protoc/mac-arm64@version:3.14.0 (line 17): no such package: infra/3pp/tools/protoc/mac-arm64
  failed to resolve fuchsia/third_party/clang/mac-arm64@git_revision:3dc24bc31edbc01dea085b24a6a6b024d7ae531c (line 19): no such package: fuchsia/third_party/clang/mac-arm64
  failed to resolve infra/3pp/tools/cpython3/mac-arm64@version:3.8.2.chromium.10 (line 21): no such tag
  failed to resolve fuchsia/third_party/qemu/mac-arm64@git_revision:841f14e74f5af7886cf49cfcd4fed264911ae58e (line 31): no such package: fuchsia/third_party/qemu/mac-arm64
/Users/connectedhomeip/.environment/cipd/cipd ensure -ensure-file /Users/connectedhomeip/.environment/cipd/pigweed.ensure -root /Users/connectedhomeip/.environment/cipd/pigweed -log-level warning -cache-dir /Users/.cipd-cache-dir -max-threads 0

None
Traceback (most recent call last):
  File "/Users/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 610, in <module>
    sys.exit(main())
  File "/Users/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 602, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 331, in setup
    result = step(spin)
  File "/Users/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 403, in cipd
    if not cipd_update.update(cipd=cipd_client,
  File "/Users/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 227, in update
    subprocess.check_call(cmd,
  File "/opt/homebrew/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/Users/connectedhomeip/.environment/cipd/pigweed.ensure', '-root', '/Users/connectedhomeip/.environment/cipd/pigweed', '-log-level', 'warning', '-cache-dir', '/Users/.cipd-cache-dir', '-max-threads', '0']' returned non-zero exit status 1.

```
So how can I fix it?  Thank you!

JourneyYoung, 2021-05-18:
Did this project unsupport Macbook Pro on Apple M1 ？

xylophone21, 2021-11-09:
I met the similar issue

lihui@HuideMacBook-Air connectedhomeip % . scripts/activate.sh

  WELCOME TO...

  ▄███▒  ░▓█  ░▓█ ░▓█▓ ▒█████▄
 ██▒ ▀█▒  ▒█   ▒█  ░█▒  ▒█░  █░
 █▓░      ▒██████  ░█▒  ▒█▄▄▄█░
 ▓█   █▒  ▒█   ▒█  ░█░  ▒█▀
 ░▓███▀  ░▓███░▓█▒ ░█░  ▒█

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (18.9s)
  Setting up Python environment.....[\]
ninja: Entering directory `/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out'
[1/92] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_package/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_package/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_package/py
[2/92] touch python/obj/third_party/pigweed/repo/pw_package/py/py._run_pip_install.stamp
[3/92] touch python/obj/third_party/pigweed/repo/pw_package/py/py.install.stamp
[4/92] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_protobuf/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_protobuf/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_protobuf/py
FAILED: python/gen/third_party/pigweed/repo/pw_protobuf/py/py._run_pip_install.pw_pystamp
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_protobuf/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_protobuf/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_protobuf/py
Obtaining file:///Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_protobuf/py
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Checking if build backend supports build_editable: started
  Checking if build backend supports build_editable: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
ERROR: Could not find a version that satisfies the requirement pw-cli (from pw-protobuf) (from versions: none)
ERROR: No matching distribution found for pw-cli
ninja: build stopped: subcommand failed.
['ninja', '-C', '/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 279, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/opt/homebrew/Cellar/python@3.9/3.9.7_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 679, in <module>
    sys.exit(main())
  File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 671, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 414, in setup
    result = step(spin)
  File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 533, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 294, in install
    install_packages(gn_target)
  File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 282, in install_packages
    raise subprocess.CalledProcessError(err.returncode, err.cmd,
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

woody-apple, 2022-01-25:
@JourneyYoung Sorry for the delay, m1 should be working. Is this still happening for you?

@xylophone21 - this looks different, can you re-file is this is still happening?

closed 2022-01-25 by woody-apple

JourneyYoung, 2022-01-25:
您好。你的邮件我已经收到，谢谢！

TheWalkingDead1024, 2022-04-25:
> 您好。你的邮件我已经收到，谢谢！

hello，do you know how to resolve this issue?

JourneyYoung, 2022-04-25:
您好。你的邮件我已经收到，谢谢！

TheWalkingDead1024, 2022-04-25:
> @JourneyYoung Sorry for the delay, m1 should be working. Is this still happening for you?
> 
> @xylophone21 - this looks different, can you re-file is this is still happening?

hi, I have met the same issue. Do you know how to resolve this issue?

TheWalkingDead1024, 2022-04-25:
> @JourneyYoung Sorry for the delay, m1 should be working. Is this still happening for you?
> 
> @xylophone21 - this looks different, can you re-file is this is still happening?

> 您好。你的邮件我已经收到，谢谢！

> I met the similar issue
> 
> lihui@HuideMacBook-Air connectedhomeip % . scripts/activate.sh
> 
> WELCOME TO...
> 
> ▄███▒ ░▓█ ░▓█ ░▓█▓ ▒█████▄ ██▒ ▀█▒ ▒█ ▒█ ░█▒ ▒█░ █░ █▓░ ▒██████ ░█▒ ▒█▄▄▄█░ ▓█ █▒ ▒█ ▒█ ░█░ ▒█▀ ░▓███▀ ░▓███░▓█▒ ░█░ ▒█
> 
> BOOTSTRAP! Bootstrap may take a few minutes; please be patient.
> 
> Downloading and installing packages into local source directory:
> 
> Setting up CIPD package manager...done (18.9s) Setting up Python environment.....[] ninja: Entering directory `/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out' [1/92] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_package/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_package/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_package/py [2/92] touch python/obj/third_party/pigweed/repo/pw_package/py/py._run_pip_install.stamp [3/92] touch python/obj/third_party/pigweed/repo/pw_package/py/py.install.stamp [4/92] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_protobuf/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_protobuf/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_protobuf/py FAILED: python/gen/third_party/pigweed/repo/pw_protobuf/py/py._run_pip_install.pw_pystamp python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_protobuf/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_protobuf/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_protobuf/py Obtaining file:///Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_protobuf/py Installing build dependencies: started Installing build dependencies: finished with status 'done' Checking if build backend supports build_editable: started Checking if build backend supports build_editable: finished with status 'done' Getting requirements to build wheel: started Getting requirements to build wheel: finished with status 'done' Preparing metadata (pyproject.toml): started Preparing metadata (pyproject.toml): finished with status 'done' ERROR: Could not find a version that satisfies the requirement pw-cli (from pw-protobuf) (from versions: none) ERROR: No matching distribution found for pw-cli ninja: build stopped: subcommand failed. ['ninja', '-C', '/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']
> 
> Traceback (most recent call last): File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 279, in install_packages subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs) File "/opt/homebrew/Cellar/python@3.9/3.9.7_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call raise CalledProcessError(retcode, cmd) subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last): File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 679, in sys.exit(main()) File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 671, in main return EnvSetup(**vars(parse())).setup() File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 414, in setup result = step(spin) File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 533, in virtualenv if not virtualenv_setup.install( File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 294, in install install_packages(gn_target) File "/Users/lihui/Code/matter/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 282, in install_packages raise subprocess.CalledProcessError(err.returncode, err.cmd, subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/lihui/Code/matter/connectedhomeip/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

hello, have you resolve the issue? do you know how to fix it?

TheWalkingDead1024, 2022-04-26:
> I met the similar issue
>
> lihui@HuideMacBook-Air connectedhomeip % . scripts/activate.sh
>
> WELCOME TO...
>
> ▄███▒ ░▓█ ░▓█ ░▓█▓ ▒█████▄ ██▒ ▀█▒ ▒█ ▒█ ░█▒ ▒█░ █░ █▓░ ▒██████ ░█▒ ▒█▄▄▄█░ ▓█ █▒ ▒█ ▒█ ░█░ [...]

hi, have you fixed it?