project-chip/connectedhomeip #6855: CommandSender::OnMessageReceived closes the wrong context and causes data loss

bluebin14, 2021-05-16:
### Problem

CommandSender::OnMessageReceived first ensures that the incoming apExchangeContext is equal to its own mpExchangeCtx. Then it calls ProcessCommandMessage and deletes its mpExchangeCtx thinking that it's the same one as at the beginning of the function. In my case many times it isn't, it's already the next incoming message, which due to the deletion gets lost. With CRMP this causes retransmission, while with reliable transport the new message gets lost forever.

### Suggested solutions

1 - quick fix: after calling ProcessCommandMessage check again whether mpExchangeCtx changed and don't close the unprocessed context, e.g.

if (apExchangeContext == mpExchangeCtx)
{
  mpExchangeCtx->Close();
  mpExchangeCtx = nullptr;
  Reset();
}

2 - fix all possible external code to make sure this does not happen, ever.

bluebin14, 2021-05-17:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/CommandSender.cpp

yunhanw-google, 2021-06-02:
@yufengwangca ?

yufengwangca, 2021-06-02:
The real problem here is the CommandSender try to manage multiple exchanges for different command request/response before. 

We have adjusted the logic to allocate a CommandSender for each command request/response, only one exchange is attached to CommandSender now.

yufengwangca, 2021-06-02:
@bluebin14 I don't think this issue still exist with one command one CommandSender design. Pls confirm.

bluebin14, 2021-06-02:
Confirming that the issue is fixed, tested with current master.

closed 2021-06-02 by bluebin14