project-chip/connectedhomeip #6856: Last exchange context of every PASE/CASE exchange is leaked

bluebin14, 2021-05-16:
### Problem

Due to the fact that message 3 of every PASE/CASE exchange is currently designed as imbalanced (it only gets a response if there is an error) the active context allocated by the exchange manager is never released. To see it, add the following to ExchangeManager::OnMessageReceived.

mContextPool.ForEachActiveObject([&](auto * ec) {
    printf("---active %04x %04x 0x%llx 0x%x\n", ec->GetExchangeId(), ec->GetSecureSession().GetAdminId(),
        	ec->GetSecureSession().GetPeerNodeId(), ec->GetSecureSession().GetPeerKeyId());
    return true;
});

Filtered output during and after commissioning:

---active acd9 ffff 0xffffffffffffffff 0x0

---active acd9 ffff 0xffffffffffffffff 0x0

---active acd9 ffff 0xffffffffffffffff 0x0
---active acda 0000 0xbc5c01 0x0

---active acd9 ffff 0xffffffffffffffff 0x0
---active acdb 0000 0xbc5c01 0x0

---active acd9 ffff 0xffffffffffffffff 0x0
---active acdc 0000 0xbc5c01 0x0

---active acd9 ffff 0xffffffffffffffff 0x0
---active acdd 0000 0xbc5c01 0x0

The context with exchangeid acd9 is stuck indefinitely in active state and never released.

### Suggested solution

Fix PASE/CASE so message 3 (or more generally every request) always gets a response, whether there is an error or not. In addition ensure that exchange contexts do not get stuck indefinitely.

yunhanw-google, 2021-05-17:
yep, we also detect this issue in cirque test.

CHIP:EM: Sending Standalone Ack for MsgId:00000006
CHIP:IN: Secure message was encrypted: Msg ID 9
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 742640 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:EM: Flushed pending ack for MsgId:00000006
CHIP:DMG: Client[0] moving to [INIT]
Read Response: 3/3(100.00%) time=0.008ms
CHIP:DMG: ICR moving to [Uninitiali]
CHIP:DMG: Client[0] moving to [UNINIT]
CHIP:DMG: Client[0] moving to [UNINIT]
CHIP:DMG: IM RH moving to [Uninitialized]
chip-im-initiator: ../../../../../src/messaging/ExchangeMgr.cpp:104: CHIP_ERROR chip::Messaging::ExchangeManager::Shutdown(): Assertion `ec.GetReferenceCount() == 0' faile
Not sure how we guarantee the ec are deterministically released after this change
we are continuing to trace down these issues...not sure what is the right mechanism to make cirque back to work

bzbarsky-apple, 2021-05-17:
Note https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/2542

kghost, 2021-05-18:
Thank, I'll try to fix it.

bluebin14, 2021-09-18:
fixed

closed 2021-09-18 by bluebin14