project-chip/connectedhomeip #6863: chip-tool on Mac throws Ble endpoint pool FULL

cjandhyala, 2021-05-16:
With ToT (8a51de25248b6ab506d30e89f0f95b216d06729c) code, chip-tool on Mac throws BLE EndPT full error when trying to provision accessory running on raspi

 ./out/host/chip-tool pairing ble-wifi MYSSID MYPASSWD 87564321 20202021 3840
[1621203866816] [0x989948] CHIP: [DL] _Init
[1621203866816] [0x989948] CHIP: [IN] TransportMgr initialized
[1621203866816] [0x989948] CHIP: [DIS] Init admin pairing table with server storage.
[1621203866816] [0x989948] CHIP: [IN] local node id is 0x000000000001b669
[1621203866816] [0x989948] CHIP: [ZCL] Using ZAP configuration...
[1621203866816] [0x989948] CHIP: [ZCL] deactivate report event
[1621203866816] [0x989948] CHIP: [CTL] Getting operational keys
[1621203866816] [0x989948] CHIP: [CTL] Generating operational certificate for the controller
[1621203866817] [0x989948] CHIP: [CTL] Getting root certificate for the controller from the issuer
[1621203866817] [0x989948] CHIP: [CTL] Generating credentials
[1621203866817] [0x989948] CHIP: [CTL] Loaded credentials successfully
[1621203866817] [0x989957] CHIP: [DL] CHIP task running
[1621203866817] [0x989948] CHIP: [DMG] ICR moving to [Initialize]
[1621203866818] [0x989948] CHIP: [BLE] NewConnection
[1621203866821] [0x989948] CHIP: [BLE] Assigned local session key ID 0
[1621203866821] [0x989948] CHIP: [IN] Message appended to BLE send queue
[1621203866821] [0x989948] CHIP: [BLE] Sent PBKDF param request
[1621203866884] [0x989958] CHIP: [BLE] CBManagerState: ON
[1621203870549] [0x989959] CHIP: [BLE] Connecting to device with discriminator: 3840
[1621203873627] [0x989958] CHIP: [IN] BleConnectionComplete: endPoint 0x10ce8a8f0
[1621203873627] [0x989958] CHIP: [BLE] Ble endpoint pool FULL
[1621203873627] [0x989958] CHIP: [IN] Clearing BLE pending packets.
[1621203873627] [0x989958] CHIP: [IN] BleConnection Error: Ble Error 6002 (0x00001772): No more BLE endpoints
[1621203874433] [0x989958] CHIP: [BLE] subscribe complete, ep = 0x10ce8a8f0
[1621203874482] [0x989958] CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
[1621203874482] [0x989958] CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
[1621203874482] [0x989958] CHIP: [BLE] local and remote recv window size = 3
[1621203874482] [0x989958] CHIP: [IN] BLE EndPoint 0x10ce8a8f0 Connection Complete
[1621203896822] [0x989957] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1621203896822] [0x989957] CHIP: [BLE] PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1621203896822] [0x989957] CHIP: [TOO] Secure Pairing Failed
[1621203896822] [0x989948] CHIP: [CTL] Shutting down the commissioner
[1621203896822] [0x989948] CHIP: [CTL] Shutting down the controller
[1621203896823] [0x989957] CHIP: [TOO] Pairing Failure: CHIP Error 4050 (0x00000FD2): Timeout
[1621203896823] [0x989948] CHIP: [DL] System Layer shutdown
[1621203896823] [0x989948] CHIP: [DL] Inet Layer shutdown
[1621203896823] [0x989948] CHIP: [DL] BLE layer shutdown
[1621203896823] [0x989948] CHIP: [IN] Clearing BLE pending packets.
[1621203896823] [0x989948] CHIP: [BLE] Auto-closing end point's BLE connection.
[1621203896836] [0x989948] CHIP: [DMG] ICR moving to [Uninitiali]

yunhanw-google, 2021-05-17:
[1621203873627] [0x989958] CHIP: [BLE] Ble endpoint pool FULL
[1621203873627] [0x989958] CHIP: [IN] Clearing BLE pending packets.
[1621203873627] [0x989958] CHIP: [IN] BleConnection Error: Ble Error 6002 (0x00001772): No more BLE endpoints

@vivien-apple  please help take a look at this issue, both chip-device-controller are seeing ble endpoint pool full issue in mac, thanks

woody-apple, 2021-05-17:
@bzbarsky-apple @pan-apple to investigate

vivien-apple, 2021-05-19:
I have not been able to reproduce this with the M5 Stack. I think it may be related to `didDiscoverPeripheral` firing twice because of how the Raspi is advertising over mdns.

Pretty sure this is related to some BLE advertisement differences.

jmartinez-silabs, 2021-05-19:
I get the same issue with chip-tool or python controller connecting to a EFR32 platform. It works when the controller is ran on linux

bzbarsky-apple, 2021-05-19:
@cjandhyala if you could provide the specific steps used to compile the chip-tool, that would be pretty useful.

bzbarsky-apple, 2021-05-19:
Also a stack trace to the `Ble endpoint pool FULL` line would be helpful.  And are we getting multiple calls to `BleLayer::OnConnectionComplete` or so?

vivien-apple, 2021-05-22:
> Also a stack trace to the `Ble endpoint pool FULL` line would be helpful.  And are we getting multiple calls to `BleLayer::OnConnectionComplete` or so?

This is likely the issue. I bet that the advertisement is triggering multiple calls to 'didDiscoverPeripheral'

krypton36, 2021-06-21:
I have a fix here: https://github.com/project-chip/connectedhomeip/compare/master...krypton36:macOSBLEFix

I am just getting approval from my manager to submit.

PR #7806 (2021-07-01): Fix Ble endpoint pool FULL on macOS controller
#### Problem
What is being fixed?  Examples:
* Fixes #6863 chip-tool on Mac throws Ble endpoint pool FULL

#### Change overview
- Updated the found variable from local to global.
- Prevents multiple calls to the same service.

#### Testing
How was this tested? (at least one bullet point required)
* Tested pairing BLE accessory with chip-device-ctrl 
* BLE accessory that was tested was RasPi (Linux)
* Controller platform was macOS
* macOS is the only platform accepted
closed 2021-07-01 by bzbarsky-apple