project-chip/connectedhomeip #6877: Fix data corruption/mangling in DeviceCommissioner::SendOperationalCertificate

bluebin14, 2021-05-17:
#### Problem
DeviceCommissioner::SendOperationalCertificate has the task of sending 2 certificates to the device. Instead, it only sends the first one and splits it into 2 portions as a "hack" for ZAP compatibility. This is bad and absolutely not needed since the code is only used with Interaction Model which is not limited to 255 bytes. See OperationalCredentialsCluster::AddOpCert for proof, it calls writer->Put(TLV::ContextTag(argSeqNumber++), noc) which does the right thing regardless of length

 #### Proposed Solution
The following one liner is all you need to do it right:

yunhanw-google, 2021-06-02:
@bluebin14  post a pr?

bluebin14, 2021-06-02:
https://github.com/project-chip/connectedhomeip/pull/7309

PR #7309 (2021-06-02): Fix data corruption/mangling in DeviceCommissioner::SendOperationalCertificate
#### Problem
What is being fixed?
* fixes TODO in DeviceCommissioner::SendOperationalCertificate, which was corrupting the op cert and failing to send the ICA cert
* Fixes #6877

#### Change overview
What's in this PR

DeviceCommissioner::SendOperationalCertificate now correctly sends both the op cert and the ICA cert. Previously it was corrupting them both, thinking that TempZCL protocol needs to be supported which has a 255 byte limitation. TempZCL protocol is not involved here, the code uses Interaction Model exc [... cut]
closed 2021-06-02 by andy31415