project-chip/connectedhomeip #6882: compile lighting-app/nrfconnect example failed

cahty82008, 2021-05-17:
I try to compile the latest commit  [8d7486f](https://github.com/project-chip/connectedhomeip/commit/8d7486f280d9f96d5d4f897779f3bef82f6d506a)  in ubuntu 20.04 LTS, as in the [README](https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/nrfconnect#building) , I use the 'native shell setup', compiled this example but got the following error (west build -b nrf52840dk_nrf52840 -t menuconfig also report the same error):

```
root@ubuntu:# west build -b nrf52840dk_nrf52840
WARNING: ZEPHYR_BASE=~/work_src/ncs/zephyr in the calling environment will be used,
but the zephyr.base config option in ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect is "zephyr"
which implies a different ZEPHYR_BASE=~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr
To disable this warning in the future, execute 'west config --global zephyr.base-prefer env'
-- west build: generating a build system
-- Found Python3: ~/work_src/connectedhomeip/.environment/pigweed-venv/bin/python3 (found version "3.8.5") found components: Interpreter 
Including boilerplate (Zephyr base): ~/work_src/ncs/zephyr/cmake/app/boilerplate.cmake
-- Application: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect
-- Zephyr version: 2.5.99 ~/work_src/ncs/zephyr)
-- Found Python3: ~/work_src/connectedhomeip/.environment/pigweed-venv/bin/python3 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /root/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/opt/gcc-arm-none-eabi-9-2019-q4-major)
-- Found BOARD.dts: ~/work_src/ncs/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/boards/nrf52840dk_nrf52840.overlay
-- Generated zephyr.dts: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/device_extern.h
Parsing ~/work_src/ncs/zephyr/Kconfig
Loaded configuration '~/work_src/ncs/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '~/work_src/connectedhomeip/config/nrfconnect/app/sample-defaults.conf'
Merged configuration 'prj.conf'
Configuration saved to '~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/.config'
Kconfig header saved to '~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- CMake build type: Debug
-- Version: gfd27fc3a7
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "~/work_src/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/west/manifest.py", line 1417, in get_projects
    raise ValueError(unknown, uncloned)
ValueError: (['mbedtls-nrf'], [])
CMake Error at ~/work_src/ncs/nrfxlib/common.cmake:58 (message):
  Failed to find mbedtls, cannot build security libraries
Call Stack (most recent call first):
  ~/work_src/ncs/nrfxlib/nrf_security/CMakeLists.txt:29 (get_mbedtls_dir)

-- Configuring incomplete, errors occurred!
See also "~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/CMakeFiles/CMakeOutput.log".
See also "~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: ~/work_src/connectedhomeip/.environment/cipd/pigweed/bin/cmake -DWEST_PYTHON=~/work_src/connectedhomeip/.environment/pigweed-venv/bin/python3 -B~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build -S~/work_src/connectedhomeip/examples/lighting-app/nrfconnect -GNinja -DBOARD=nrf52840dk_nrf52840

```

cahty82008, 2021-05-18:
i switched my nrf-sdk to tag [v1.5.1](https://github.com/nrfconnect/sdk-nrf/releases/tag/v1.5.1) , now the compile throw another error: 
```
root@ubuntu:# west build -b nrf52840dk_nrf52840
-- west build: generating a build system
-- Found Python3: /root/.espressif/python_env/idf4.2_py3.8_env/bin/python3.8 (found version "3.8.5") found components: Interpreter 
Including boilerplate (Zephyr base): ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/cmake/app/boilerplate.cmake
-- Application: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect
-- Zephyr version: 2.6.0-rc1 (~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr), build: v2.6.0-rc1-123-g878043aff19c
-- Found Python3: /root/.espressif/python_env/idf4.2_py3.8_env/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.11.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Cache files will be written to: /root/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/opt/gcc-arm-none-eabi-9-2019-q4-major)
-- Found BOARD.dts: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/boards/nrf52840dk_nrf52840.overlay
-- Generated zephyr.dts: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/zephyr/include/generated/device_extern.h
Parsing ~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/Kconfig
Loaded configuration '~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration '~/work_src/connectedhomeip/config/nrfconnect/app/sample-defaults.conf'
Merged configuration 'prj.conf'

error: OPENTHREAD_MBEDTLS (defined at subsys/net/l2/openthread/Kconfig:186) is assigned in a
configuration file, but is not directly user-configurable (has no prompt). It gets its value
indirectly from other symbols. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_OPENTHREAD_MBEDTLS.html and/or look up
OPENTHREAD_MBEDTLS in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

CMake Error at zephyr/cmake/kconfig.cmake:264 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  zephyr/cmake/app/boilerplate.cmake:555 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:33 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/root/.espressif/python_env/idf4.2_py3.8_env/bin/python -B~/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build -S~/work_src/connectedhomeip/examples/lighting-app/nrfconnect -GNinja -DBOARD=nrf52840dk_nrf52840

```

Damian-Nordic, 2021-06-02:
Could you try to run `<chip_root>/scripts/setup/nrfconnect/update_ncs.py -u` script to update your NCS to the recommended version? Note that you must have `ZEPHYR_BASE` configure properly before running the script to make it aware of the location of your NCS installation. Then please remove the `build` directory and run the build again.

cahty82008, 2021-06-07:
i updated my source code to aa96ea0, updated the ncs, and i got this error:

```
root@ubuntu:/home/jsh/work_src/connectedhomeip# source /home/jsh/work_src/ncs/zephyr/zephyr-env.sh
root@ubuntu:/home/jsh/work_src/connectedhomeip# ./scripts/setup/nrfconnect/update_ncs.py -u
Updating nRF Connect SDK to recommended revision...
remote: Enumerating objects: 7, done.
remote: Counting objects: 100% (7/7), done.
remote: Compressing objects: 100% (6/6), done.
remote: Total 7 (delta 1), reused 2 (delta 1), pack-reused 0
Unpacking objects: 100% (7/7), 6.62 KiB | 968.00 KiB/s, done.
From https://github.com/nrfconnect/sdk-nrf
   2e8026a43..a92c81b87  master     -> origin/master
HEAD is now at 4e3dac58c manifest: Update sdk-zephyr revision
=== updating zephyr (zephyr):
HEAD is now at a9d3aef654 [nrf fromtree] modules: hal_nordic: nrf_802154: Fix RADIO IRQ ...
=== updating mcuboot (bootloader/mcuboot):
HEAD is now at 3ea42d2 Revert "[nrf temphack] prj.conf: don't use CONFIG_LOG_MODE_MINIMAL yet"
=== updating nrfxlib (nrfxlib):
HEAD is now at fb93618 softdevice_controller: rev cf113d186d33fbcbd82300ab74a5516fc069f5b2
=== updating trusted-firmware-m (modules/tee/tfm):
HEAD is now at 3599bcc1 [nrf mergeup] Merge upstream up to commit 69dc29a48b707dfe732a57e05a329
=== updating tfm-mcuboot (modules/tee/tfm-mcuboot):
HEAD is now at 81d19f0 bootutil: Add cmake build file
=== updating connectedhomeip (modules/lib/connectedhomeip):
HEAD is now at a008cba6 [nrf mergeup] Merge upstream up to commit 1a94b9e72047fa155fe733962d108afb64b4e197
Synchronizing submodule url for 'third_party/nlio/repo'
Synchronizing submodule url for 'third_party/nlassert/repo'
Synchronizing submodule url for 'third_party/pigweed/repo'
=== updating cmock (test/cmock):
M	vendor/unity
HEAD is now at 9d09289 Merge pull request #346 from ThrowTheSwitch/reapply_329
=== updating unity (test/cmock/vendor/unity):
HEAD is now at 74cde08 Merge pull request #540 from numaru/master
=== updating mbedtls-nrf (mbedtls):
HEAD is now at 523f0554b Merge pull request #748 from ARMmbed/mbedtls-2.24.0r0-pr
=== updating nanopb (modules/lib/nanopb):
HEAD is now at df0e92f Publishing nanopb-0.4.2
=== updating Alexa-Gadgets-Embedded-Sample-Code (modules/alexa-embedded):
HEAD is now at face92d Update firmware.options signature_size from 48 to 65
=== updating cddl-gen (modules/lib/cddl-gen):
HEAD is now at 24f1b2b module: leverage external module definition
=== updating cmsis (modules/hal/cmsis):
HEAD is now at c3bd209 DSP: fix compiler warning on inline attribute
=== updating canopennode (modules/lib/canopennode):
HEAD is now at 468d350 Merge pull request #2 from vestas-wind-systems/update_to_v1_3_master
=== updating civetweb (modules/lib/civetweb):
HEAD is now at e6903b80 civetweb: WebSockets support for Zephyr
=== updating fatfs (modules/fs/fatfs):
HEAD is now at 1d1fcc7 fat_fs: Expose _VOLUMES to Kconfig
=== updating hal_nordic (modules/hal/nordic):
HEAD is now at fc301b9 drivers: use new IEEE 802.15.4 subsystem integration scheme
=== updating hal_st (modules/hal/st):
HEAD is now at b52fdbf sensor/stmemsc: Align stmemsc i/f to v1.03
=== updating libmetal (modules/hal/libmetal):
HEAD is now at 9d4ee2c lib: update libmetal to release v2020.10.0
=== updating lvgl (modules/lib/gui/lvgl):
HEAD is now at 31acbaa3 lv_font: Fix compiler warning when compressed fonts are disabled (#1880)
=== updating mbedtls (modules/crypto/mbedtls):
HEAD is now at 24d84ec Bump mbedTLS to 2.16.9
=== updating mcumgr (modules/lib/mcumgr):
HEAD is now at c52ecb7 snapshot: Update mcumgr to commit b92aa0b8c9 from upstream
=== updating net-tools (tools/net-tools):
HEAD is now at 41132e9 docker: Add syslog receiver
=== updating open-amp (modules/lib/open-amp):
HEAD is now at de1b85a lib: update open-amp lib to release v2020.10.0
=== updating loramac-node (modules/lib/loramac-node):
HEAD is now at 2cee5f72 LoRaMac: query minimum tx datarate
=== updating openthread (modules/lib/openthread):
HEAD is now at fd27fc3a7 Merge pull request #38 from kkasperczyk-no/ot_update_pr
[... 48 lines cut ...]
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- CMake build type: Debug
-- Version: gfd27fc3a7
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/root/.espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/west/manifest.py", line 1417, in get_projects
    raise ValueError(unknown, uncloned)
ValueError: (['mbedtls-nrf'], [])
CMake Error at /home/jsh/work_src/ncs/nrfxlib/common.cmake:58 (message):
  Failed to find mbedtls, cannot build security libraries
Call Stack (most recent call first):
  /home/jsh/work_src/ncs/nrfxlib/nrf_security/CMakeLists.txt:29 (get_mbedtls_dir)

-- Configuring incomplete, errors occurred!
See also "/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/CMakeFiles/CMakeOutput.log".
See also "/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/root/.espressif/python_env/idf4.2_py3.8_env/bin/python -B/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build -S/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect -GNinja -DBOARD=nrf52840dk_nrf52840
```

Damian-Nordic, 2021-06-07:
Looks like `west` Python module can't find `mbedtls-nrf` project although the log indicates you have it checked out. Did you remove `build` directory before triggering the build? 

The following command is used to retrieve the project path. Can you run it in the shell directly and paste the result?
```python3 -c "from west.manifest import Manifest; manifest = Manifest.from_file(); print(manifest.get_projects(['mbedtls-nrf'])[0].posixpath)"
/home/damian/src/ncs/mbedtls```

cahty82008, 2021-06-07:
yes, I removed build directory before triggering the build, and ZEPHYR_BASE is configured and exported:
```
root@ubuntu:/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect# echo $ZEPHYR_BASE 
/home/jsh/work_src/ncs/zephyr
```
run the given commandline:
```
root@ubuntu:/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect# /home/jsh/work_src/ncs/mbedtls
bash: /home/jsh/work_src/ncs/mbedtls: Is a directory
```

Damian-Nordic, 2021-06-07:
Sorry, I badly formatted the command and github displayed it incorrectly. Could you run the correct one?

cahty82008, 2021-06-07:
yeah, the python3 command throws an error:
```
root@ubuntu:/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect# python3 -c "from west.manifest import Manifest; manifest = Manifest.from_file(); print(manifest.get_projects(['mbedtls-nrf'])[0].posixpath)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/root/.espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/west/manifest.py", line 1417, in get_projects
    raise ValueError(unknown, uncloned)
ValueError: (['mbedtls-nrf'], [])

```

cahty82008, 2021-06-07:
seems it's the environment problem for normal user and root.
 run the command as  a normal user instead of root, the python3 command will be succeed:
```
jsh@ubuntu:~$ python3 -c "from west.manifest import Manifest; manifest = Manifest.from_file(); print(manifest.get_projects(['mbedtls-nrf'])[0].posixpath)"
/home/jsh/work_src/ncs/mbedtls
```
but compile the example need the root permision, do you have any idea to fix this ?

Damian-Nordic, 2021-06-07:
Why is the root permission needed to compile the example? It shouldn't be needed.

cahty82008, 2021-06-07:
use the normal user to compile will get the permission denied error:
```
jsh@ubuntu:~/work_src/connectedhomeip/examples/lighting-app/nrfconnect$ west  build -b nrf52840dk_nrf52840
WARNING: ZEPHYR_BASE=/home/jsh/work_src/ncs/zephyr in the calling environment will be used,
but the zephyr.base config option in /home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect is "zephyr"
which implies a different ZEPHYR_BASE=/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr
To disable this warning in the future, execute 'west config --global zephyr.base-prefer env'
-- west build: generating a build system
-- Found Python3: /home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/bin/python3 (found version "3.8.5") found components: Interpreter 
Including boilerplate (Zephyr base (cached)): /home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect
-- Zephyr version: 2.6.0-rc1 (/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr), build: v2.6.0-rc1-123-g878043aff19c
-- Found Python3: /home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/bin/python3 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
CMake Error at zephyr/cmake/zephyr_module.cmake:61 (message):
  Traceback (most recent call last):

    File "/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/scripts/zephyr_module.py", line 378, in <module>
      main()
    File "/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/scripts/zephyr_module.py", line 361, in main
      with open(args.kconfig_out, 'w', encoding="utf-8") as fp:

  PermissionError: [Errno 13] Permission denied:
  '/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/Kconfig/Kconfig.modules'

Call Stack (most recent call first):
  zephyr/cmake/app/boilerplate.cmake:183 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:33 (find_package)

-- Configuring incomplete, errors occurred!
CMake Error: Cannot open file for write: /home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build/CMakeCache.txt
CMake Error: : System Error: Permission denied
FATAL ERROR: command exited with status 1: /home/jsh/work_src/connectedhomeip/.environment/cipd/pigweed/bin/cmake -DWEST_PYTHON=/home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/bin/python3 -B/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build -S/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect -GNinja

```

Damian-Nordic, 2021-06-07:
It fails because the previous build was triggered with the root permissions and the build directory contains files owned by root.
Can you `sudo rm -rf build` and start the build again (as a normal user)?

cahty82008, 2021-06-07:
it has permission issue to creat the 'build' directory
```
jsh@ubuntu:~/work_src/connectedhomeip/examples/lighting-app/nrfconnect$ sudo rm -rf build/
jsh@ubuntu:~/work_src/connectedhomeip/examples/lighting-app/nrfconnect$ west  build -b nrf52840dk_nrf52840
WARNING: ZEPHYR_BASE=/home/jsh/work_src/ncs/zephyr in the calling environment will be used,
but the zephyr.base config option in /home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect is "zephyr"
which implies a different ZEPHYR_BASE=/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr
To disable this warning in the future, execute 'west config --global zephyr.base-prefer env'
Traceback (most recent call last):
  File "/home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/bin/west", line 8, in <module>
    sys.exit(main())
  File "/home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/west/app/main.py", line 779, in main
    app.run(argv or sys.argv[1:])
  File "/home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/west/app/main.py", line 106, in run
    self.run_command(argv)
  File "/home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/west/app/main.py", line 338, in run_command
    self.run_extension(args.command, argv)
  File "/home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/west/app/main.py", line 408, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "/home/jsh/work_src/connectedhomeip/.environment/pigweed-venv/lib/python3.8/site-packages/west/commands.py", line 129, in run
    self.do_run(args, unknown)
  File "/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/scripts/west_commands/build.py", line 147, in do_run
    self._setup_build_dir()
  File "/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/zephyr/scripts/west_commands/build.py", line 259, in _setup_build_dir
    os.makedirs(build_dir, exist_ok=False)
  File "/usr/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/home/jsh/work_src/connectedhomeip/examples/lighting-app/nrfconnect/build'

```

cahty82008, 2021-06-07:
maybe i need to recreate the project with a non-root user?

Damian-Nordic, 2021-06-07:
It seems that the CHIP project has been checked out using root or so. You can fix it by changing an owner of the CHIP directory:
`sudo chown -R jsh /home/jsk/work_src/connectedhomeip`.

cahty82008, 2021-06-07:
compile throws the 'mbedtls-nrf' error again, i just realized, the python3 command fails when I run it under 'lighting-app/nrfconnect' directory, but succeed under CHIP directory:
```
jsh@ubuntu:~/work_src/connectedhomeip/examples/lighting-app/nrfconnect$ echo $ZEPHYR_BASE
/home/jsh/work_src/ncs/zephyr
jsh@ubuntu:~/work_src/connectedhomeip$ python3 -c "from west.manifest import Manifest; manifest = Manifest.from_file(); print(manifest.get_projects(['mbedtls-nrf'])[0].posixpath)"
/home/jsh/work_src/ncs/mbedtls
jsh@ubuntu:~/work_src/connectedhomeip$ cd examples/lighting-app/nrfconnect/
jsh@ubuntu:~/work_src/connectedhomeip/examples/lighting-app/nrfconnect$ python3 -c "from west.manifest import Manifest; manifest = Manifest.from_file(); print(manifest.get_projects(['mbedtls-nrf'])[0].posixpath)"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jsh/.espressif/python_env/idf4.2_py3.8_env/lib/python3.8/site-packages/west/manifest.py", line 1417, in get_projects
    raise ValueError(unknown, uncloned)
ValueError: (['mbedtls-nrf'], [])

```

cahty82008, 2021-06-07:
@Damian-Nordic  thanks a lot for your helpful support, maybe the configuration in lighting-app/nrfconnect/.west/config is not correct, i delete the .west directory then it works.
```
jsh@ubuntu:~/work_src/connectedhomeip/examples/lighting-app/nrfconnect$ cat .westbak/config 
[manifest]
path = zephyr
file = west.yml

[zephyr]
base = zephyr
```

Damian-Nordic, 2021-06-07:
Yeah, looks like the manifest.path configuration was set incorrectly. So it works correctly now, right?

cahty82008, 2021-06-07:
yeah, the example compiled successfully, thank you again

closed 2021-06-07 by cahty82008