project-chip/connectedhomeip #6917: Lock guard chip event loop and all exposed API of chip.

kghost, 2021-05-18:
#### Problem
We have encountered lots of racing problem. It is required to lock guard all chip API which can be called from outside.

This will be a huge change, create an issue to track it.

Follow chip component will be exposed:
 * device controller and device API.
 * messaging layer API
 * channel API
 * protocols which provides public API
   * BDX

apps should be considered outside chip, so lock is not required.

kghost, 2021-05-18:
#3746 Save issue for device controller

woody-apple, 2022-01-25:
We're in progress of reviewing older issues. Is this still an issue given how old the SDK this was tested against? Please re-open if so. Thanks! (And please add detail about what changes are required)

closed 2022-01-25 by woody-apple