project-chip/connectedhomeip #6928: chip-tool core dump

chrisbeach59, 2021-05-18:
After sending a command chip-tool core dumps
Git commit: 6bfdd05783f4fb60148fa469d29f2bcf16b7f50b
Chip built with chip_bypass_rendezvous=true (due to issues with rendezvous failing)

Frequency of occurrence is nearly consistent
Core file from systemd-coredump attached
lz4 file is gziped so it coule be attached.

[core.gz](https://github.com/project-chip/connectedhomeip/files/6501350/core.gz)
[
Output from:chip-tool
CHIP:ZCL: APS frame processing success!
CHIP:ZCL: RX len 9, ep 1, clus 0x201
CHIP:ZCL:  FC 8 seq 0 cmd 1 payload[
CHIP:ZCL: 12 00 00 29 D0 07
CHIP:ZCL: ]
CHIP:ZCL: ReadAttributesResponse:
CHIP:ZCL:   ClusterId: 0x0201
CHIP:ZCL:   attributeId: 0x0012
CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
CHIP:ZCL:   attributeType: 0x29
CHIP:ZCL:   value: 2000
CHIP:TOO: Int16s attribute Response: 2000
CHIP:ZCL: Data model processing success!
CHIP:EM: Sending Standalone Ack for MsgId:00000001
CHIP:IN: Secure message was encrypted: Msg ID 2
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000bc5c01 at utc time: 4279060 msecCHIP:CTL: Shutting down the controller
CHIP:IN: Sending secure msg on generic transport
Segmentation fault (core dumped)

Output from chip-all-clusters-app
CHIP:ZCL: OTA READ: ep:1 cid:201 attid:12 msk:40 mfcode: 0
CHIP:ZCL: READ: clus 201, attr 12, dataLen: 2, OK
CHIP:ZCL: Measured APS frame size 11
CHIP:ZCL: Successfully encoded 11 bytes
CHIP:IN: Secure message was encrypted: Msg ID 1
CHIP:IN: Sending msg from 0x0000000000bc5c01 to 0x000000000001b669 at utc time: 4279059 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:ZCL: T   0:TX (0x558985c3628e) Ucast 0x00x558985c362ae
CHIP:ZCL: TX buffer: [
CHIP:ZCL: 08 00 01 12 00 00 29 D0 07
CHIP:ZCL: ]
CHIP:ZCL: Data model processing success!
CHIP:EM: Crit-err 4003 when sending CHIP MsgId:00000001, send tries: 0

kpschoedel, 2021-05-21:
I'm also seeing this, but only intermittently.

kpschoedel, 2021-05-21:
Looks like a shutdown race. Good:

```
CHIP:EM: Sending Standalone Ack for MsgId:0000001A
CHIP:IN: Secure message was encrypted: Msg ID 2
CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000BC5C01 at utc time: 1634721623 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:CTL: Shutting down the controller
```

Bad:
```
CHIP:EM: Sending Standalone Ack for MsgId:0000001B
CHIP:CTL: Shutting down the controllerCHIP:IN:
Secure message was encrypted: Msg ID 2
CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000BC5C01 at utc time: 1634722771 msec
CHIP:IN: Sending secure msg on generic transport

Thread 8 "chip-tool" received signal SIGSEGV, Segmentation fault.
```

The message `CHIP:CTL: Shutting down the controller` is from `DeviceController::Shutdown()` which as its first step calls `SecureSessionMgr::Shutdown()` which sets `mTransportMgr = nullptr` causing the crash.

Perhaps fixing #6931 will also fix this.