project-chip/connectedhomeip #6937: lighting-app example running over ble is unsuccessful due to PASE handshake failure

nageshshamnur, 2021-05-18:
#### Problem
**expected behavior:** Connect command in ChipDeviceController running on laptop should be able to connect to Lighting App running on RPi4B
**actual behavior:** Connect command in ChipDeviceController is timeout and unable to connect to Lighting App running on RPi4B.
**system configuration:**
    **Topology:** Intel Corei7 laptop running Ubuntu 20.04.2 LTS  <-------------------> RPi4B running Ubuntu 21.04
**Branch:** Master
**Note:** Both laptop and RPi using the default bluetooth radio (without any dongles). hciconfig on both devices is having hci0.
```
Laptop BD Address: 6C:6A:77:48:40:08
RPi4B BD Address: B8:27:EB:91:23:92
```
**steps to reproduce:**
RPi4B running Lighting App as suggested in the page https://github.com/project-chip/connectedhomeip/tree/master/examples/lighting-app/linux Logs as below:
```
root@ubuntu:/home/ubuntu/connectedhomeip/examples/lighting-app/linux# ./out/debug/chip-lighting-app --ble-device 0
CHIP:DL: Device Configuration:
CHIP:DL:   Serial Number: TEST_SN
CHIP:DL:   Vendor Id: 9050 (0x235A)
CHIP:DL:   Product Id: 65279 (0xFEFF)
CHIP:DL:   Product Revision: 1
CHIP:DL:   Setup Pin Code: 20202021
CHIP:DL:   Setup Discriminator: 3840 (0xF00)
CHIP:DL:   Manufacturing Date: (not set)
CHIP:DL:   Device Type: 257 (0x101)
CHIP:SVR: SetupQRCode: [CH:YNJV793700A50324800]
CHIP:SVR: Copy/paste the below URL in a browser to see the QR Code:
CHIP:SVR: https://dhrishi.github.io/connectedhomeip/qrcode.html?data=CH%3AYNJV793700A50324800
CHIP:SVR: Manual pairing code: [34970112332]
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:ZCL: OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pairing table.
CHIP:DIS: Set the admin pairing table delegate
CHIP:ZCL: OpCreds: Call to writeAdminsIntoFabricsListAttribute
CHIP:ZCL: OpCreds: Stored 0 admins in fabrics list attribute.
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:ZCL: Unknown cluster ID: 40
CHIP:DIS: Init admin pairing table with server storage.
CHIP:IN: TransportMgr initialized
CHIP:DL: writing settings to file (/tmp/chip_counters.ini.tmp)
CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
CHIP:DL: NVS set: chip-counters/GlobalMCTR = 12000 (0x2EE0)
CHIP:IN: local node id is 0x0000000000bc5c01
CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:BLE: Assigned local session key ID 0
CHIP:BLE: Waiting for PBKDF param request
CHIP:DIS: Start dns-sd server
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DIS: Replying to DNS-SD service listing request
CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:DL: wpa_supplicant: _IsWiFiStationProvisioned: interface not connected
CHIP:DIS: Using wifi MAC for hostname
CHIP:DL: rotatingDeviceId: 00007C5F6E176CD40F68685D100A1CF8A98B
CHIP:DIS: Advertise commission parameter vendorID=74586 productID=130815 discriminator=3840/00
CHIP:DIS: CHIP minimal mDNS configured as 'Commissionable node device'.
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bluez mainloop starting Thread
[... 92 lines cut ...]
2021-05-18 20:39:30,461 ChipBLEMgr   INFO     Pairing State   = 0
2021-05-18 20:39:30,461 ChipBLEMgr   INFO     Discriminator   = 3840
2021-05-18 20:39:30,462 ChipBLEMgr   INFO     Vendor Id       = 9050
2021-05-18 20:39:30,462 ChipBLEMgr   INFO     Product Id      = 65279
2021-05-18 20:39:30,464 ChipBLEMgr   INFO     Adv UUID        = 0000fff6-0000-1000-8000-00805f9b34fb
2021-05-18 20:39:30,465 ChipBLEMgr   INFO     Adv Data        = 00000f5a23fffe
2021-05-18 20:39:30,465 ChipBLEMgr   INFO     
2021-05-18 20:39:39,280 ChipBLEMgr   INFO     scanning stopped
chip-device-ctrl > connect -ble 3840 20202021
Device is assigned with nodeid = 963381
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: DC:A6:32:F2:CB:46
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
Failed to establish secure session to device: 4050
CHIP:BLE: PASESession timed out while waiting for a response from the peer. Expected message type was 33
CHIP Error 4050 (0x00000FD2): Timeout
Failed to establish secure session to device: 4050
chip-device-ctrl > exit
```
Note: As soon as new connection requested is received at lighting-app immediately connection disconnection is happening and PASE failure and connection timeout at chip-dev-ctrl is happening much later.

#### Proposed Solution
NA

hainahouwei, 2021-05-19:
I facing the seam problem too. Have you get the solution?

hainahouwei, 2021-05-19:
And do you konw what is PASE ?

nageshshamnur, 2021-05-19:
> I facing the seam problem too. Have you get the solution?

not yet solved. waiting for the solution.

nageshshamnur, 2021-06-24:
@Damian-Nordic : Can you help me with this issue? Thanks.

fdelacourt-wyze, 2021-06-24:
I"m having the same issue with the TE3 image. :-/
The master branch does not seem to show that problem.

nageshshamnur, 2021-06-25:
> I"m having the same issue with the TE3 image. :-/
> The master branch does not seem to show that problem.

last i tested in mid may on master and i was facing this issue. Will check on confirm if this issue still persists in master now. Thanks for the info.

closed 2022-01-25 by woody-apple