project-chip/connectedhomeip #6947: TestUDP is failing for some developers

mspang, 2021-05-18:
We've had multiple reports of TestUDP failing for some developers (on Linux).

Also, one of the reports went away after rebooting.

kghost, 2021-09-14:
I'm having the same problem when in the office, but can't reproduce when at home. I guess it is related to the network environment, Maybe decided by whether ipv6 is assigned by AP.

mlepage-google, 2021-09-14:
Happening for me too, at home, on Linux.

mspang, 2021-09-14:
Is it a deterministic fail in these cases or does it pass when you retry?

mlepage-google, 2021-09-14:
It seems to always fail when I run from scratch, but yes it passed on subsequent runs. (Probably because of less contention for the port.)

@kghost made a PR which seems to fix it (not sure it was linked to this issue)
https://github.com/project-chip/connectedhomeip/pull/9692

mspang, 2021-09-14:
Ok, just a port reuse issue - that's what I thought.

closed 2021-09-14 by woody-apple