project-chip/connectedhomeip #6951: CHIP TOOL APP Commissioning  Error

ChengqiangShe, 2021-05-19:
commit ：a95b6221acd72ffa7fc90684e7e14f0947cea5db

We use the CHIP Tool APP to configure the network, and the device will report an error. Do you know the reason?

@jmartinez-silabs

00> <detail> [DL] Start BLE advertissement
00> <detail> [DL] bleAdv Timeout : Start slow advertissment
00> <info  > [DL] Connect Event for handle : 1
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 9)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [BLE] local and remote recv window sizes = 3
00> <info  > [BLE] selected BTP version 3
00> <info  > [BLE] using BTP fragment sizes rx 128 / tx 20.
00> <info  > [DL] CHIPoBLE unsubscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
00> <error > [IN] Failed to establish BLE connection: Ble Error 6000004 (0x005B8D84): BLE central unsubscribed
00> <detail> [IN] Clearing BLE pending packets.
00> <info  > [BLE] Releasing end point's BLE connection back to application.
00> <info  > [DL] CHIPoBLE unsubscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
00> <error > [BLE] no endpoint for unsub recvd
00> <info  > [DL] Disconnect Event for handle : 1
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4115)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
00> <detail> [DL] Start BLE advertissement
00> <detail> [DL] bleAdv Timeout : Start slow advertissment
00> <info  > [DL] Connect Event for handle : 1
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 9)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [BLE] local and remote recv window sizes = 3
00> <info  > [BLE] selected BTP version 3
00> <info  > [BLE] using BTP fragment sizes rx 128 / tx 20.
00> <info  > [DL] Disconnect Event for handle : 1
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4115)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
00> <error > [IN] Failed to establish BLE connection: Ble Error 6000011 (0x005B8D8B): BLE transport protocol fired abort
00> <detail> [IN] Clearing BLE pending packets.
00> <info  > [BLE] Releasing end point's BLE connection back to application.
00> <detail> [DL] Start BLE advertissement
00> <info  > [DL] Connect Event for handle : 1
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 9)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [BLE] local and remote recv window sizes = 3
00> <info  > [BLE] selected BTP version 3
00> <info  > [BLE] using BTP fragment sizes rx 128 / tx 20.
00> <info  > [DL] CHIPoBLE unsubscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
00> <error > [IN] Failed to establish BLE connection: Ble Error 6000004 (0x005B8D84): BLE central unsubscribed
00> <detail> [IN] Clearing BLE pending packets.
00> <info  > [BLE] Releasing end point's BLE connection back to application.
00> <info  > [DL] CHIPoBLE unsubscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
00> <error > [BLE] no endpoint for unsub recvd
00> <info  > [DL] Disconnect Event for handle : 1
00> <info  > [DL] BLE GATT connection closed (con 1, reason 4115)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEConnectionError
00> <detail> [DL] Start BLE advertissement

ChengqiangShe, 2021-05-20:
device log：
00> <efr32 > ==================================================
00> <efr32 > chip-efr32-lighting-example starting
00> <efr32 > ==================================================
00> <efr32 > Init CHIP Stack
00> <detail> [-] Initializing key value store
00> <detail> [-] KeyValueStore init complete: active keys 0, deleted keys 0, sectors 4, logical sector size 8192 bytes
00> <info  > [DL] Setting device name to : "EFR32_LIGHT"
00> <efr32 > Initializing OpenThread stack
00> <info  > [DL] OpenThread started: OK
00> <info  > [DL] Setting OpenThread device type to ROUTER
00> <efr32 > Starting Platform Manager Event Loop
00> <efr32 > Starting OpenThread task
00> <efr32 > Starting App Task
00> <efr32 > Starting FreeRTOS scheduler
00> <info  > [DL] Bluetooth stack booted: v3.1.1-b214
00> 
00> <detail> [DL] Thread task running
00> <detail> [DL] CHIP task running
00> <info  > [ZCL] Using ZAP configuration...
00> <detail> [DL] Start BLE advertissement
00> <info  > [ZCL] deactivate report event
00> <info  > [ZCL] OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pair<detail> [DL] bleAdv Timeout : Start slow advertissment
00> <info  > [DL] Connect Event for handle : 1
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 9)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [BLE] local and remote recv window sizes = 3
00> <info  > [BLE] selected BTP version 3
00> <info  > [BLE] using BTP fragment sizes rx 128 / tx 20.
00> <info  > [DL] CHIPoBLE subscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLESubscribe
00> <detail> [IN] BLE EndPoint 0x2000f100 Connection Complete
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32777
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 51)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [EM] Received message of type 32 and protocolId 0
00> <detail> [EM] ec id: 39025, Delegate: 0x20011108
00> <detail> [BLE] Received PBKDF param request
00> <detail> [BLE] Sent PBKDF param response
00> <error > [DL] Long dispatch time: 1284 ms
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 86)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [EM] Received message of type 34 and protocolId 0
[... 76 lines cut ...]
Failed to establish secure session to device: 4050
CHIP:BLE: PASESession timed out while waiting for a response from the peer. Expected message type was 33
CHIP Error 4050 (0x00000FD2): Timeout
Failed to establish secure session to device: 4050
chip-device-ctrl > 
chip-device-ctrl > connect -ble 3840 73141520 5551
Device is assigned with nodeid = 5551
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: 75:5C:ED:20:AB:91
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:BLE: subscribe complete, ep = 0xffffa98389b8
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
CHIP:EM: Received message of type 33 and protocolId 0
CHIP:EM: Received message of type 47 and protocolId 0
CHIP:BLE: Received error (255) during pairing process
Failed to establish secure session to device: 255
Error 255 (0x000000FF)

@jmartinez-silabs
Hi jmartinez，
I re-use the chip-device-ctrl tool to configure the network, the above error will appear, the log is as above, do you know how I should solve this problem？

hainahouwei, 2021-05-20:
I also met the same problem in commit 5b9f9dc66e627241659f57936cbcb9455254a289  which is the newest release tag "test_event_2_2012_04_22".
And I try to use the chip-tool to configure the network ,follow the document https://www.silabs.com/documents/public/training/wireless/chip-connected-home-over-ip-lab.pdf  met the same erro like this "PASESession timed out while waiting for a response from the peer. Expected message type was 33".   Can you please give me any advise to solve this problem？  @jmartinez-silabs

jmartinez-silabs, 2021-05-20:
I am not able to reproduce the error you are seeing on current master and I know it was succesfull during the test_event 2

Can you give me the following information about your setup pls.

- On what platform is the Controller built and running. What commit are you using ?
- What is efr32 board used to build and run the example. What commit are you using ?

hainahouwei, 2021-05-20:
1. I built and run the Controller on the ubuntu 20.04 .
2. I built and run the lighting-app on the efr32 board  BRD4161A.
3. Both of  the Controller  and lighting-app are base on commit 5b9f9dc66e627241659f57936cbcb9455254a289 .
@jmartinez-silabs

ChengqiangShe, 2021-05-21:
1. commit ：a95b6221acd72ffa7fc90684e7e14f0947cea5db
2. lighting-app on the efr32 board BRD4180A.  efr32mg21A020F1024
3.  I changed "SL_HEAP_SIZE=(12 * 1024)" to "SL_HEAP_SIZE=(8 * 1024)", otherwise the compilation will report insufficient RAM error
@jmartinez-silabs

jmartinez-silabs, 2021-05-21:
@ChengqiangShe pull top of the tree, I have a pr tondeal with the RAM build issue on mg21. I am guessing that by reducing the heap to 8k the encryption fails somewhere in mbedtls functions due to not enough heap remaining.

ChengqiangShe, 2021-05-21:
chip-device-ctrl ：log

chip-device-ctrl > connect -ble 3840 73141520 1238
Device is assigned with nodeid = 1238
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: 7B:73:10:9A:E6:F9
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:BLE: subscribe complete, ep = 0xffffb00d2c30
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
CHIP:EM: Received message of type 33 and protocolId 0
CHIP:EM: Received message of type 35 and protocolId 0
CHIP:IN: Sending msg from 0x000000000001B669 to 0x00000000000004D6 at utc time: 171939964 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39020
CHIP:ZCL: DefaultResponse:
CHIP:ZCL:   Transaction: 0xffffb00d1358
CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
CHIP:CTL: Device failed to receive the CSR request Response: 0x01
Failed to establish secure session to device: 4172
CHIP Error 4172 (0x0000104C): Internal error

device（efr32mg21A020F1024）log：
00> <efr32 > ==================================================
00> <efr32 > chip-efr32-lighting-example starting
00> <efr32 > ==================================================
00> <efr32 > Init CHIP Stack
00> <detail> [-] Initializing key value store
00> <detail> [-] KeyValueStore init complete: active keys 0, deleted keys 0, sectors 4, logical sector size 8192 bytes
00> <info  > [DL] Setting device name to : "EFR32_LIGHT"
00> <efr32 > Initializing OpenThread stack
00> <info  > [DL] OpenThread started: OK
00> <info  > [DL] Setting OpenThread device type to ROUTER
00> <efr32 > Starting Platform Manager Event Loop
00> <efr32 > Starting OpenThread task
00> <efr32 > Starting App Task
00> <efr32 > Starting FreeRTOS scheduler
00> <info  > [DL] Bluetooth stack booted: v3.1.1-b214
00> 
00> <detail> [DL] Thread task running
00> <detail> [DL] CHIP task running
00> <info  > [ZCL] Using ZAP configuration...
00> <detail> [DL] Start BLE advertissement
00> <info  > [ZCL] deactivate report event
00> <info  > [ZCL] OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pair<info  > [DL] Connect Event for handle : 1
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 9)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [BLE] local and remote recv window sizes = 3
00> <info  > [BLE] selected BTP version 3
00> <info  > [BLE] using BTP fragment sizes rx 128 / tx 20.
00> <info  > [DL] CHIPoBLE subscribe received
00> <info  > [DL] _OnPlatformEvent kCHIPoBLESubscribe
00> <detail> [IN] BLE EndPoint 0x2000f100 Connection Complete
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32777
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 51)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [EM] Received message of type 32 and protocolId 0
00> <detail> [EM] ec id: 39019, Delegate: 0x20011108
00> <detail> [BLE] Received PBKDF param request
00> <detail> [BLE] Sent PBKDF param response
[... 95 lines cut ...]
00> <info  > [SVR] Stored value in server storage
00> <info  > [DIS] Admin (0) persisted to storage. Calling OnAdminPersistedToStorage.
00> <info  > [ZCL] OpCreds: Admin 0 was persisted to storage. FabricId lX, NodeId lX, VendorId lX
00> <info  > [ZCL] OpCreds: Call to writeAdminsIntoFabricsListAttribute
00> <info  > [ZCL] OpCreds: Skipping over unitialized admin with fabricId lX, nodeId lX vendorId B669
00> <info  > [ZCL] OpCreds: Stored 0 admins in fabrics list attribute.
00> <info  > [EM] Received message of type 8 and protocolId 327680
00> <detail> [EM] ec id: 39020, Delegate: 0x200139d8
00> <detail> [DMG] ICR moving to [Initialize]
00> <detail> [ZCL] Received Cluster Command: Cluster=3e Command=hx Endpoint=hx
00> <info  > [ZCL] OpCreds: commissioner has requested an OpCSR
00> <info  > [ZCL] OpCreds: Finding admin with fabricId  lX.
00> <info  > [DIS] Looking at index 0 with fabricID lu nodeID lu vendorId 112233 to see if it matches fabricId lu nodeId lu vendorId 0.
00> <info  > [DIS] Found a match!
00> <info  > [ZCL] OpCreds: NewCertificateSigningRequest returned 0
00> <detail> [DMG] ICR moving to [AddCommand]
00> <detail> [IN] Secure message was encrypted: Msg ID 1
00> <info  > [IN] Sending msg from 0x00000000000004d6 to 0x000000000001b669 at utc time: ld msec
00> <info  > [IN] Sending secure msg on generic transport
00> <efr32 > TASK OVERFLOW
00> <efr32 > CHIP
00> 
00> FREERTOS ASSERT ( (volatile void *) NULL )

Device and chip-device-ctrl I used commit fa887331e7044fea9a59777f56ad37abcbbf5b2e, the above error occurred when pairing.@jmartinez-silabs

jmartinez-silabs, 2021-05-21:
Yes like said use top of the tree the commit you use is before the PR to fix ram issues 

https://github.com/project-chip/connectedhomeip/commit/8dc938d63504e0e44d204eca505cc893fbc4d690

hainahouwei, 2021-05-21:
I have successed ctl the light with the commit 8dc938d .
Thank you very  much! @jmartinez-silabs 

`chip-device-ctrl > zcl OnOff  Toggle 221443 0 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000036103 at utc time: 6032640 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:IN: Secure transport received message destined to node ID (112233)
CHIP:EM: Received message of type 9 and protocolId 327680
SetCommandIndexStatus commandHandle=140619291309136 commandIndex=1
CHIP:ZCL: DefaultResponse:
Received command status response:
Container: 
    ProtocolId = 5
    ProtocolCode = 0
    EndpointId = 1
    ClusterId = 6
    CommandId = 2
    CommandIndex = 1
CHIP:ZCL:   Transaction: 0x7fe47ae83450
CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
chip-device-ctrl > zcl OnOff  Off 221443 0 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000036103 at utc time: 6055053 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:IN: Secure transport received message destined to node ID (112233)
CHIP:EM: Received message of type 9 and protocolId 327680
SetCommandIndexStatus commandHandle=140619291309136 commandIndex=1
CHIP:ZCL: DefaultResponse:
Received command status response:
Container: 
    ProtocolId = 5
    ProtocolCode = 0
    EndpointId = 1
    ClusterId = 6
    CommandId = 0
    CommandIndex = 1
CHIP:ZCL:   Transaction: 0x7fe47ae83450
CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
chip-device-ctrl > zcl OnOff  On 221443 0 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000036103 at utc time: 6062827 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:IN: Secure transport received message destined to node ID (112233)
CHIP:EM: Received message of type 9 and protocolId 327680
SetCommandIndexStatus commandHandle=140619291309136 commandIndex=1
CHIP:ZCL: DefaultResponse:
Received command status response:
Container: 
    ProtocolId = 5
    ProtocolCode = 0
    EndpointId = 1
    ClusterId = 6
    CommandId = 1
    CommandIndex = 1
CHIP:ZCL:   Transaction: 0x7fe47ae83450
CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
chip-device-ctrl > `

ChengqiangShe, 2021-05-21:
chip-device-ctrl ：log

chip-device-ctrl > connect -ble 3840 73141520 1267
Device is assigned with nodeid = 1267
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: New device scanned: 68:FF:91:D5:06:A5
CHIP:BLE: Device discriminator match. Attempting to connect.
CHIP:BLE: Scan complete notification without an active scan.
CHIP:BLE: subscribe complete, ep = 0xffffbc44abe0
CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
CHIP:BLE: local and remote recv window size = 3
CHIP:EM: Received message of type 33 and protocolId 0
CHIP:EM: Received message of type 35 and protocolId 0
CHIP:IN: Sending msg from 0x000000000001b669 to 0x00000000000004f3 at utc time: 190290973 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39018
CHIP:ZCL: DefaultResponse:
CHIP:ZCL:   Transaction: 0xffffbc449308
CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
CHIP:CTL: Device failed to receive the CSR request Response: 0x01
Failed to establish secure session to device: 4172
CHIP Error 4172 (0x0000104C): Internal error
chip-device-ctrl > 
chip-device-ctrl > CHIP:IN: Secure transport received message destined to node ID (112233)
CHIP:IN: Setting fabricID 4F3 on admin.
CHIP:IN: Since admin was modified, persisting changes to KVS
CHIP:EM: Received message of type 9 and protocolId 327680
CHIP:EM: OnMessageReceived failed, err = CHIP Error 4112 (0x00001010): Unsolicited msg with originator bit clear

device（efr32mg21A020F1024）log：
00> <efr32 > ==================================================
00> <efr32 > chip-efr32-lighting-example starting
00> <efr32 > ==================================================
00> <efr32 > Init CHIP Stack
00> <detail> [-] Initializing key value store
00> <detail> [-] KeyValueStore init complete: active keys 4, deleted keys 0, sectors 4, logical sector size 8192 bytes
00> <info  > [DL] Setting device name to : "EFR32_LIGHT"
00> <efr32 > Initializing OpenThread stack
00> <info  > [DL] OpenThread started: OK
00> <info  > [DL] Setting OpenThread device type to ROUTER
00> <efr32 > Starting Platform Manager Event Loop
00> <efr32 > Starting OpenThread task
00> <efr32 > Starting App Task
00> <efr32 > Starting FreeRTOS scheduler
00> <info  > [DL] Bluetooth stack booted: v3.1.1-b214
00> 
00> <detail> [DL] Thread task running
00> <detail> [DL] CHIP task running
00> <info  > [ZCL] Using ZAP configuration...
00> <detail> [DL] Start BLE advertissement
00> <info  > [ZCL] deactivate report event
00> <info  > [ZCL] OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pairing table.
00> <info  > [DIS] Set the admin pairing table delegate
00> <info  > [ZCL] OpCreds: Call to writeAdminsIntoFabricsListAttribute
00> <info  > [ZCL] OpCreds: Stored 0 admins in fabrics list attribute.
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <efr32 > Unknown cluster ID: 40
00> <info  > [DIS] Init admin pairing table with server storage.
00> <detail> [IN] TransportMgr initialized
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32778
[... 291 lines cut ...]
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm

Device and chip-device-ctrl I used commit 8dc938d63504e0e44d204eca505cc893fbc4d690, the above error occurred when pairing.@jmartinez-silabs

hainahouwei, 2021-07-24:
>  pull top of the tree, I have a pr tondeal with the RAM build issue on mg21. I am guessing that by reducing the heap to 8k the encryption fails somewhere in mbedtls functions due to not enough heap remaining.
@jmartinez-silabs  I pull top of the tree (commit 762a204b04049dac01c7ecf8f695ddd58f085c29 ) show the RAM build issue on mg21  again.

closed 2022-01-25 by woody-apple