project-chip/connectedhomeip #6958: AddWiFiNetwork command over NetworkCommisioning cluster fails for python controller.

sweetymhaiske, 2021-05-19:
>  #### Problem
Using python controller, AddWiFiNetwork command over NetworkCommisioning cluster fails for both commissioning using WiFi and BLE.

**Chip tool command for commissioning using WiFi**
```
chip-device-ctrl > connect -ip 192.168.4.1 20202021 456456
chip-device-ctrl > zcl NetworkCommissioning AddWiFiNetwork 456456 0 0 ssid=str:Wce*** credentials=str:123456789 breadcrumb=0 timeoutMs=1000
```
**Chip tool command for commissioning using BLE**
```
chip-device-ctrl > connect -ble 3840 20202021 144838
chip-device-ctrl > zcl NetworkCommissioning AddWiFiNetwork 144838 0 0 ssid=str:Wce*** credentials=str:123456789 
```
**Device error log for both WiFi and BLE**
```
GAP procedure initiated: stop advertising.
I (55004) chip[DL]: CHIPoBLE advertising stopped
I (55014) app-devicecallbacks: Current free heap: 116888
E (88564) chip[IN]: Secure transport received message destined to node ID (729325)
E (88564) chip[CR]: mbedTLS error: CCM - Authenticated decryption failed
E (88564) chip[IN]: Secure transport received message, but failed to decode it, discarding
```

dhrishi, 2021-05-20:
@woody-apple @andreilitvin Can you guys please take a look.

sweetymhaiske, 2021-05-20:
Closing this issue, as it no more occurs with the latest commit on the master branch. Thanks.

closed 2021-05-20 by sweetymhaiske