project-chip/connectedhomeip #7005: Python controller fails to receive ZCL command response

jmartinez-silabs, 2021-05-20:
The control fails to receive the response of a command send on the thread network.
Command was received, executed by the device and a reply was sent back. Controller times out on a failure to receive the response

Controller is on a raspi running Ubuntu 20.10 . The device is a EFR32 MG12 running the lighting example
running the ToT at the moment of the test commit 8dc938d.

I did the thread commissioning on BLE
```
connect -ble 3840 73141520 1
zcl NetworkCommissioning AddThreadNetwork 1 0 0 operationalDataset=hex:0e080000000000000000000300000f35060004001fffe00208dead00beef00cafe0708fdb44ab5910ed1c8051000112233445566778899aabbccddeedf030e4f70656e54687265616444656d6f010212340410538c2a622249a0595e78169834896a2f0c0402a0fff8 breadcrumb=0 timeoutMs=1000
zcl NetworkCommissioning EnableNetwork 1 0 0 networkID=hex:dead00beef00cafe breadcrumb=0 timeoutMs=1000
close-ble
resolve 0 1
```

I send the command zcl OnOff Toggle 1 0 0
```
chip-device-ctrl > zcl OnOff Toggle 1 0 0
CHIP:DMG: ICR moving to [Initialize]
CHIP:DMG: ICR moving to [AddCommand]
CHIP:IN: Secure message was encrypted: Msg ID 5
CHIP:IN: Sending msg from 0x000000000001b669 to 0x0000000000000001 at utc time: 7112272 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
CHIP:DMG: ICR moving to [   Sending]
CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39025
CHIP:ZCL: DefaultResponse:
CHIP:ZCL:   Transaction: 0xffff8eda2fb8
CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
Failed to receive command response: None
CHIP:ZCL: emberAfDefaultResponseCallback: Missing success callback
CHIP:ZCL: emberAfDefaultResponseCallback: Missing failure callback
CHIP:DMG: ICR moving to [Uninitiali]
```

The device received both commands and send a response to the right Destination (IPv6 of the host)
```
00> <detail> [DL] Thread packet RCVD: UDP, len 116
00> <detail> [DL]     src  FD11:33::DEA6:32FF:FE9B:6711, port 11097
00> <detail> [DL]     dest FD11:22::22FA:6888:B8D9:3911, port 11097
00> <error > [IN] Secure transport received message destined to node ID (lu)
00> <info  > [EM] Received message of type 8 and protocolId 327680
00> <detail> [EM] ec id: 39024, Delegate: 0x200151c8
00> <detail> [DMG] ICR moving to [Initialize]
00> <detail> [ZCL] Received Cluster Command: Cluster=6 Command=hx Endpoint=hx
00> <info  > [ZCL] On/Off set value: 0 2
00> <info  > [ZCL] Toggle on/off from 0 to 1
00> <efr32 > Turning light ON
00> <efr32 > Unknown cluster ID: 8
00> <detail> [DMG] ICR moving to [AddCommand]
00> <detail> [IN] Secure message was encrypted: Msg ID 6
00> <info  > [IN] Sending msg from 0x0000000000000001 to 0x000000000001b669 at utc time: ld msec
00> <info  > [IN] Sending secure msg on generic transport
00> <detail> [DL] Thread packet SENT: UDP, len 122
00> <detail> [DL]     src  FD11:22::22FA:6888:B8D9:3911, port 11097
00> <detail> [DL]     dest FD11:33::DEA6:32FF:FE9B:6711, port 11097
00> <info  > [IN] Secure msg send status No Error
00> <detail> [DMG] ICR moving to [   Sending]
00> <detail> [DMG] ICR moving to [Uninitiali]
00> <efr32 > Light ON
```

Verified that the packet is delivered to host ip with tcpdump
```
pi@raspberrypi:~ $ sudo tcpdump -i wpan0 udp port 11097 
listening on wpan0, link-type LINUX_SLL (Linux cooked), capture size 262144 bytes
11:21:51.074289 IP6 fd11:33::dea6:32ff:fe9b:6711.11097 > fd11:22::22fa:6888:b8d9:3911.11097: UDP, length 68
11:21:51.238420 IP6 fd11:22::22fa:6888:b8d9:3911.11097 > fd11:33::dea6:32ff:fe9b:6711.11097: UDP, length 74
11:22:05.004434 IP6 fd11:33::dea6:32ff:fe9b:6711.11097 > fd11:22::22fa:6888:b8d9:3911.11097: UDP, length 68
11:22:05.087452 IP6 fd11:22::22fa:6888:b8d9:3911.11097 > fd11:33::dea6:32ff:fe9b:6711.11097: UDP, length 74
```

OT-BR syslog also show that it processed both the Receive and transmit
```
May 20 11:21:51 raspberrypi otbr-agent[425]: [INFO]-MAC-----: Sent IPv6 UDP msg, len:116, chksum:2db1, to:0x4800, sec:yes, prio:low
May 20 11:21:51 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     src:[fd11:33:0:0:dea6:32ff:fe9b:6711]:11097
May 20 11:21:51 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     dst:[fd11:22:0:0:22fa:6888:b8d9:3911]:11097
May 20 11:21:51 raspberrypi otbr-agent[425]: [INFO]-MAC-----: Received IPv6 UDP msg, len:122, chksum:09e1, from:0x4800, sec:yes, prio:normal, rss:-75.0
May 20 11:21:51 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     src:[fd11:22:0:0:22fa:6888:b8d9:3911]:11097
May 20 11:21:51 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     dst:[fd11:33:0:0:dea6:32ff:fe9b:6711]:11097
May 20 11:21:51 raspberrypi otbr-agent[425]: [INFO]-PLAT----: processReceive: OK
...
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-PLAT----: processTransmit: OK
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-MAC-----: Sent IPv6 UDP msg, len:116, chksum:0641, to:0x4800, sec:yes, prio:low
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     src:[fd11:33:0:0:dea6:32ff:fe9b:6711]:11097
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     dst:[fd11:22:0:0:22fa:6888:b8d9:3911]:11097
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-MAC-----: Received IPv6 UDP msg, len:122, chksum:02d2, from:0x4800, sec:yes, prio:normal, rss:-75.0
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     src:[fd11:22:0:0:22fa:6888:b8d9:3911]:11097
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-MAC-----:     dst:[fd11:33:0:0:dea6:32ff:fe9b:6711]:11097
May 20 11:22:05 raspberrypi otbr-agent[425]: [INFO]-PLAT----: processReceive: OK
```

jmartinez-silabs, 2021-06-03:
did a tcp dump on the controller and the command response does reach back wlan0 of the controller
```
ubuntu@ubuntuservercontroller:~$ sudo tcpdump -i wlan0 udp port 11097
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), capture size 262144 bytes
16:47:24.580208 IP6 ubuntuservercontroller.11097 > fd11:22::e3ca:35fc:63a5:9229.11097: UDP, length 68
16:47:24.656524 IP6 fd11:22::e3ca:35fc:63a5:9229.11097 > ubuntuservercontroller.11097: UDP, length 74
```

jmartinez-silabs, 2021-06-03:
I got it to work a few times. once the session works, it keeps working until i close the controller. Same thing on the other end, when it fails it keeps failing for the whole session. I still didn’t find where would the packet drops. When it doesn’t work, `UDP::OnUdpReceive` is never triggered.

I was surprised that even when it would works, the trace that I added here in `src/inet/InetLayer.cpp`  never happens
```
    case kInetEvent_UDPDataReceived:
        ChipLogProgress(Inet, "UDP DATA RECEIVED");
        static_cast<UDPEndPoint &>(aTarget).HandleDataReceived(
            System::PacketBufferHandle::Adopt(reinterpret_cast<chip::System::PacketBuffer *>(aArgument)));
        break;
```

I also added a callback when registering `mUDPEndPoint->Listen`  for a error case onUdpReceiveError but that also never triggers. Not sure where I should look upstream for those UDP message dropped

yufengwangca, 2021-06-21:
@yufengwangca 

If it keeps failing for the whole session, then it is not the race issue we discussed over the slack.  I am surprised you could not see the log within the Inet, could you.

1. Reprint the log with ChipLogError?
 case kInetEvent_UDPDataReceived:
        ChipLogError(Inet, "UDP DATA RECEIVED");

2. Instrument log also in UDP.cpp :OnUdpReceive() since the previous log is Lwip only, just confirm if the packet is received from another path

jmartinez-silabs, 2021-06-21:
@yufengwangca I will rety with ChipLogError on both. At the time off my previous post, I did add a trace in UDP.cpp :OnUdpReceive() and that trace would never get printed either when it wasn't work but it was when working. Thank you

yufengwangca, 2021-06-21:
@jmartinez-silabs 

In the case of working, you said you did see the log from UDP.cpp :OnUdpReceive() , could you confirm if you also see the log from src/inet/InetLayer.cpp? Just want to confirm if the Lwip is active or not.

yufengwangca, 2021-06-22:
I think this might be the root cause https://github.com/project-chip/connectedhomeip/issues/7804

Receiving message is actually driven by the CHIP event loop, but there are two event loop started by the python controller.

yufengwangca, 2021-06-28:
https://github.com/project-chip/connectedhomeip/pull/7930 should fix this issue, please verify.

jmartinez-silabs, 2021-06-29:
Thank you @yufengwangca . I can close this issue now.

closed 2021-06-29 by jmartinez-silabs