project-chip/connectedhomeip #7015: Python controller to have APIs generate setup code, QR code

cjandhyala, 2021-05-20:
Python controller to have APIs generate setup code, QR code, this is needed to generated setupCode/QRcode with various parameters for certification testing of controllers to have them verify all invalid combinations. Right now we are using setup code/QRcode generated in the sample apps. but for certification testing we need to generate different combinations of these codes.

erjiaqing, 2021-05-26:
there is a existing qrcodetool under src , maybe we can port it to python script

yufengwangca, 2021-06-17:
Need to clarify, if we want to add the following two commands on device controller?
> setup-payload generate-manual
34970112332
> setup-payload generate-qr
"VP:vendorpayload%MT:W0GU2OTB00KA0648G00"

Those codes are from controller itself, if we want to get the manual or qr code from the paired node, we first need to connect to those device and issue command like:
setup-payload generate-manual <node-id>
setup-payload generate-qr <node-id>

But we don't have any protocol or cluster defined to acquire those info from the devices, I don't think it is doable until we add this in the spec.

cjandhyala, 2021-06-21:
@yufengwangca ew don't need to read manual/QR codes from the device(accessory). Instead if the API "setup-payload generate-manual" and "generate-qr" can take the params specified in the spec, then we can generate different set of manual and qr codes

[image]

[image]

PR #7964 (2021-06-28): [TE4] Add command to generate setup code, QR code with various parame…
…ters

#### Problem
What is being fixed?  Examples:
Python controller misses command to generate setup code, QR code, this is needed to generated setupCode/QRcode with various parameters for certification testing of controllers to have them verify all invalid combinations. Right now we are using setup code/QRcode generated in the sample apps. but for certification testing we need to generate different combinations of these codes.

* Fixes #7015
#### Change overview
Add command to generate setup code, QR code with various pa [... cut]
closed 2021-06-28 by woody-apple