project-chip/connectedhomeip #7038: Android CHIPTool pairing fialed

cahty82008, 2021-05-21:
#### Problem
 I'm at the latest commit [82a8544](https://github.com/project-chip/connectedhomeip/commit/82a8544c3a7df2e59e16f30fb7393df22a654a87)

After tap the PROVISION CHIP DEVICE WITH THREAD button and scan the commissioning QR code, i got the 'pairing failed' notifications :
Scaning for BLE device 3840 -> Connecting to ChipLock -> Pairing -> **Pairing failed**

Damian-Nordic, 2021-06-02:
@cahty82008 I recently did a couple of fixes to the Android CHIPTool and nRF Connect platform to make the commissioning work again. Could you try with the latest master please?
Note that sending CHIP commands still doesn't work in the Android app, but there's ongoing effort to bring it back to life.

cahty82008, 2021-06-08:
@Damian-Nordic  I tried the latest commit, it's not working for me.I got the pairing failed error. But actually I set up a border router in a Raspberry Pi instead of in a ubuntu 20.

I'm confused, The [tutorial](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/nrfconnect_android_commissioning.md)   metioned, "The accessory device joins a **CHIP-enabled** Thread network", but seems like the tutorial created a normal openthread network. I wonder if it's the reason for pairing failed.

Damian-Nordic, 2021-06-08:
That's ok. "CHIP-enabled Thread network" means a normal Thread network whose nodes happen to be CHIP nodes as well. The tutorial will be updated soon, to switch to the RPi guide. 

However, I can't explain why the commissioning fails in your case. I just tested it yesterday and everything worked for me using nRF Connect Lock Example. I suggest you collect logs from the Android CHIPTool and try to find some error messages related to the commissioning. Or just paste all the logs here. Also, make sure that you rebuild both apps using the same revision. Typically, when the commissioning fails, it's because of the incompatibility between the controller and the device.

cahty82008, 2021-06-09:
thank you for the help, after a step by step complete recompile, the pairing function works properly

closed 2021-06-09 by cahty82008