project-chip/connectedhomeip #7047: Descriptor Cluster takes 1k RAM and 2k Text along

yufengwangca, 2021-05-21:
#### Problem
After I enable Descriptor Cluster on an example app. It incurs 1k RAM and 2k TEXT increase, and all of them from the auto-generated code for Descriptor Cluster. Here is the break down from my ESP32 lock example:

Total sizes:
DRAM .data size: 16176 bytes
DRAM .bss size: 45720 bytes ---> +1024
Used static DRAM: 61896 bytes ( 62684 available, 49.7% used) --> +1024

Used static IRAM: 116655 bytes ( 14417 available, 89.0% used)

  Flash code:  701251 bytes --> +940
Flash rodata:  155800 bytes --> +1168
Total image size:~ 989882 bytes (.bin may be padded larger)

Symbols within the archive: libmain.a (Not all symbols may be reported)

Symbols from section: .dram0.bss

attributeData(2606) ---> +1018

Section total: 15889 ---> +1018

Symbols from section: .dram0.data
Section total: 15

Symbols from section: .flash.rodata
_ZL17generatedDefaults(2539) ---> +1016
_ZL19generatedAttributes(1248) ---> +60
_ZL17generatedClusters(240) ---> +20
_Z41emberAfPluginDescriptorServerInitCallbackv.str1.1(71) --> +71
CSWTCH$123(4) ---> +4

Section total: 10572 --> +1171

Symbols from section: .flash.text
_Z15emberAfCopyListtP24EmberAfAttributeMetPhS1_i(1710) ---> +208
_Z26emberAfClusterInitCallbackht(256) ---> +16
_Z29emberAfAttributeValueListSizettPKh(172) ---> +36
_Z41emberAfPluginDescriptorServerInitCallbackv(142) --> +142
_Z20emberAfGetNthClusterhhb(100) --> +100
_Z19emberAfClusterCounthb(91) --> +91
_Z26writeClientServerAttributehb(84) --> +84
_Z11emberAfInitPN4chip9Messaging15ExchangeManagerE(82) ---> +2
_Z19writePartsAttributeh(78) --> +78
_Z20writeDeviceAttributeh(60) --> +60
_Z14writeAttributehtPhi(46) --> +46
_Z22emberAfContainsClusterht(25) --> +25
_Z20writeClientAttributeh(18) --> +18
_Z20writeServerAttributeh(18) --> +18
_Z36emberAfDescriptorClusterInitCallbackh(5) --> +5

Section total: 28758 ---> +929

Currently, we can not specify the List type directly with size in attribute, we use ARRAY to specify list type, not sure what is the default size of ARRAY type.

<attribute side="server" code="0x0000" define="DEVICE_LIST" type="ARRAY" entryType="DeviceType" length="254" writable="false" optional="false">device list</attribute>

All of items in Descriptor Cluster are of type ARRAY.

yufengwangca, 2021-05-21:
@vivien-apple @tima-q

bzbarsky-apple, 2021-10-22:
https://github.com/project-chip/connectedhomeip/pull/10793 reduced the .text use for descriptor by 500-1000 bytes.