project-chip/connectedhomeip #7049: Bridge App lack of the implementations related to update to the set of Bridged Devices occurs

Kevin552270, 2021-05-22:
#### Problem
As the section "9.12.6. Changes to the set of Bridged Devices" described, Bridge node should expose/update the bridged endpoints accordingly when there is add/remove operation happened from their own tools, maybe the mobile app.

[https://github.com/project-chip/connectedhomeip/tree/master/examples/bridge-app/linux], there is no related implementation to solve this problem, right now, the bridged endpoints list under the Bridge node is fixed, and not dynamic change according to the real operation. 
 >

 #### Proposed Solution
I will suggest adding below features in example codse:
 1. on the Descriptor cluster of endpoint 0:
   ◦ update the PartsList attribute (add/remove entries from this list)
   ◦ send a PartsListChanged event to indicate the changes to PartsList
2. update the exposed endpoints and their descriptors according to the new set of Bridged Devices

Kevin552270, 2021-05-22:
@yufengwangca Hi Yufeng, hope this request can be considered by your team. thanks

yufengwangca, 2021-05-25:
will take a look.

woody-apple, 2022-02-07:
@Kevin552270 can you confirm if this is complete?

closed 2022-02-07 by woody-apple

Kevin552270, 2022-02-07:
> [External email: Use caution with links and attachments]

Hello Justin,

It has been completed, thanks for confirming.

Br,
Kevin

Internal