project-chip/connectedhomeip #7057: failed to compile. ambiguous call to `writer->Put`

Kxuan, 2021-05-24:
#### Problem

**Step to reproduce**
Build `examples/all-cluster-app/esp32`

The error mesage shows:
<details>
  <summary>Click to expand!</summary>

```
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/__/third_party/connectedhomeip/src/app/clusters/door-lock-server/door-lock-server-logging.cpp.obj
/home/xuan/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../third_party/connectedhomeip/examples/all-clusters-app/all-clusters-common -I../third_party/connectedhomeip/third_party/nlio/repo/include -I../third_party/connectedhomeip/src -I../main/include -I../third_party/connectedhomeip/examples/common/screen-framework/include -I/data/git/esp-idf/components/newlib/platform_include -I/data/git/esp-idf/components/freertos/include -I/data/git/esp-idf/components/freertos/port/riscv/include -I/data/git/esp-idf/components/esp_hw_support/include -I/data/git/esp-idf/components/esp_hw_support/port/esp32c3/. -I/data/git/esp-idf/components/esp_hw_support/port/esp32c3/private_include -I/data/git/esp-idf/components/heap/include -I/data/git/esp-idf/components/log/include -I/data/git/esp-idf/components/lwip/include/apps -I/data/git/esp-idf/components/lwip/include/apps/sntp -I/data/git/esp-idf/components/lwip/lwip/src/include -I/data/git/esp-idf/components/lwip/port/esp32/include -I/data/git/esp-idf/components/lwip/port/esp32/include/arch -I/data/git/esp-idf/components/soc/include -I/data/git/esp-idf/components/soc/esp32c3/. -I/data/git/esp-idf/components/soc/esp32c3/include -I/data/git/esp-idf/components/hal/esp32c3/include -I/data/git/esp-idf/components/hal/include -I/data/git/esp-idf/components/esp_rom/include -I/data/git/esp-idf/components/esp_rom/esp32c3 -I/data/git/esp-idf/components/esp_common/include -I/data/git/esp-idf/components/esp_system/include -I/data/git/esp-idf/components/esp32c3/include -I/data/git/esp-idf/components/driver/include -I/data/git/esp-idf/components/driver/esp32c3/include -I/data/git/esp-idf/components/esp_ringbuf/include -I/data/git/esp-idf/components/efuse/include -I/data/git/esp-idf/components/efuse/esp32c3/include -I/data/git/esp-idf/components/riscv/include -I/data/git/esp-idf/components/espcoredump/include -I/data/git/esp-idf/components/esp_timer/include -I/data/git/esp-idf/components/esp_ipc/include -I/data/git/esp-idf/components/esp_pm/include -I/data/git/esp-idf/components/vfs/include -I/data/git/esp-idf/components/esp_wifi/include -I/data/git/esp-idf/components/esp_wifi/esp32c3/include -I/data/git/esp-idf/components/esp_event/include -I/data/git/esp-idf/components/esp_netif/include -I/data/git/esp-idf/components/esp_eth/include -I/data/git/esp-idf/components/tcpip_adapter/include -I/data/git/esp-idf/components/mbedtls/port/include -I/data/git/esp-idf/components/mbedtls/mbedtls/include -I/data/git/esp-idf/components/mbedtls/esp_crt_bundle/include -I/data/git/esp-idf/components/app_update/include -I/data/git/esp-idf/components/spi_flash/include -I/data/git/esp-idf/components/bootloader_support/include -I/data/git/esp-idf/components/nvs_flash/include -I/data/git/esp-idf/components/pthread/include -I/data/git/esp-idf/components/esp_gdbstub/include -I/data/git/esp-idf/components/esp_gdbstub/riscv -I/data/git/esp-idf/components/esp_gdbstub/esp32c3 -I/data/git/esp-idf/components/wpa_supplicant/include -I/data/git/esp-idf/components/wpa_supplicant/port/include -I/data/git/esp-idf/components/wpa_supplicant/include/esp_supplicant -I/data/git/esp-idf/components/app_trace/include -I/data/git/esp-idf/components/asio/asio/asio/include -I/data/git/esp-idf/components/asio/port/include -I/data/git/esp-idf/components/bt/common/osi/include -I/data/git/esp-idf/components/bt/include/esp32c3/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include -I/data/git/esp-idf/components/bt/host/nimble/port/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/include -I/data/git/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include -I/data/git/esp-idf/components/bt/host/nimble/esp-hci/include -I/data/git/esp-idf/components/cbor/port/include -I/data/git/esp-idf/components/unity/include -I/data/git/esp-idf/components/unity/unity/src -I/data/git/esp-idf/components/cmock/CMock/src -I/data/git/esp-idf/components/coap/port/include -I/data/git/esp-idf/components/coap/port/include/coap -I/data/git/esp-idf/components/coap/libcoap/include -I/data/git/esp-idf/components/coap/libcoap/include/coap2 -I/data/git/esp-idf/components/console -I/data/git/esp-idf/components/nghttp/port/include -I/data/git/esp-idf/components/nghttp/nghttp2/lib/includes -I/data/git/esp-idf/components/esp-tls -I/data/git/esp-idf/components/esp-tls/esp-tls-crypto -I/data/git/esp-idf/components/esp_adc_cal/include -I/data/git/esp-idf/components/esp_hid/include -I/data/git/esp-idf/components/tcp_transport/include -I/data/git/esp-idf/components/esp_http_client/include -I/data/git/esp-idf/components/esp_http_server/include -I/data/git/esp-idf/components/esp_https_ota/include -I/data/git/esp-idf/components/protobuf-c/protobuf-c -I/data/git/esp-idf/components/protocomm/include/common -I/data/git/esp-idf/components/protocomm/include/security -I/data/git/esp-idf/components/protocomm/include/transports -I/data/git/esp-idf/components/mdns/include -I/data/git/esp-idf/components/esp_local_ctrl/include -I/data/git/esp-idf/components/sdmmc/include -I/data/git/esp-idf/components/esp_serial_slave_link/include -I/data/git/esp-idf/components/esp_websocket_client/include -I/data/git/esp-idf/components/expat/expat/expat/lib -I/data/git/esp-idf/components/expat/port/include -I/data/git/esp-idf/components/wear_levelling/include -I/data/git/esp-idf/components/fatfs/diskio -I/data/git/esp-idf/components/fatfs/vfs -I/data/git/esp-idf/components/fatfs/src -I/data/git/esp-idf/components/freemodbus/common/include -I/data/git/esp-idf/components/idf_test/include -I/data/git/esp-idf/components/idf_test/include/esp32c3 -I/data/git/esp-idf/components/jsmn/include -I/data/git/esp-idf/components/json/cJSON -I/data/git/esp-idf/components/libsodium/libsodium/src/libsodium/include -I/data/git/esp-idf/components/libsodium/port_include -I/data/git/esp-idf/components/mqtt/esp-mqtt/include -I/data/git/esp-idf/components/openssl/include -I/data/git/esp-idf/components/spiffs/include -I/data/git/esp-idf/components/wifi_provisioning/include -I/data/git/connectedhomeip/src/platform/ESP32 -I/data/git/connectedhomeip/src/include -I/data/git/connectedhomeip/src/lib -I/data/git/connectedhomeip/src -I/data/git/connectedhomeip/third_party/nlassert/repo/include -I/data/git/connectedhomeip/third_party/nlio/repo/include -Iesp-idf/chip/src/include -Iesp-idf/chip/include -Iesp-idf/chip/gen/include -I../third_party/connectedhomeip/config/esp32/components/esp32_mbedtls -I/data/git/connectedhomeip/examples/common/QRCode/repo/c -march=rv32imc -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Wno-error=format= -nostartfiles -Wno-format -Og -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu++11 -fno-exceptions -fno-rtti -std=gnu++14 -Os -DLWIP_IPV6_SCOPES=0 -DCHIP_HAVE_CONFIG_H -D_GNU_SOURCE -DIDF_VER= [... cut ...]
3. The compiler tries to cast from `EmberAfStatus` to other compatible int type, but it find so many candidates, and it doesn't know which one should be take, so it fails again.

 #### Proposed Solution
I think there are more solutions we can choice:

1. explicit specify the underlying integer types for `EmberAfStatus` and other types which may be written to TLV. (But it need C++17)
2. explicit static_cast EmberAfStatus to `int32_t` when we call `chip::TLV::TLVWriter::Put`.
3. update the riscv32-esp toolchains, and define `int32_t` from `int` instead of `long int`.

Kxuan, 2021-05-24:
It seems this problem is introduced by 80120a20e6a33636c453b072acf22cf77f974f92. @erjiaqing @vivien-apple 

And also related #6345

chshu, 2021-05-26:
Since the original type is `enum`, the explicit `static_cast` sounds a reasonable solution.

As discussed in [esp-idf#6906](https://github.com/espressif/esp-idf/issues/6906#issuecomment-823077404), type alias on riscv32-esp is from the upstream RISC-V toolchain.

bzbarsky-apple, 2021-06-14:
Is this fixed by #7266?

Kxuan, 2021-06-15:
Fixed

closed 2021-06-15 by Kxuan