project-chip/connectedhomeip #7058: Openthread submodule needs to be updated

doru91, 2021-05-24:
#### Problem

[third_party/openthread/repo](https://github.com/project-chip/connectedhomeip/tree/master/third_party/openthread) is referencing an old commit from the OpenThread repo. The Openthread repo referenced by this old commit contains both the openthread stack and the vendor examples.

Meanwhile, the Openthread repo was refactored such that it only contains the openthread stack functionality at this moment. Each vendor has a separate repo with its examples that references the openthread repo. Example: [ot-nxp](https://github.com/openthread/ot-nxp) contains the NXP platform files for the K32W061 platform. This repo has a submodule to [openthread repo](https://github.com/openthread/openthread). 

If a vendor needs to update its openthread platform files then the vendor examples repo has to be modified. 

 #### Proposed Solution
Matter repo needs to be modified such that it references the vendor examples repo instead of the openthread repo.

I've started the work for doing the above change for the ot-nxp repo here: https://github.com/doru91/connectedhomeip/tree/feature/ot_repo_refactoring . Basically, Matter repo references ot-nxp as a submodule, then ot-nxp references openthread repo as submodule. But I have an issue with the GN Build system:

```
doru@ubuntu:~/Desktop/public_repos/connectedhomeip/examples/lock-app/k32w$ gn gen out/debug --args="k32w_sdk_root=\"${K32W061_SDK_ROOT}\" is_debug=false chip_crypto=\"mbedtls\" chip_with_se05x=0 host=\"host_k32w\""
ERROR at //third_party/connectedhomeip/third_party/openthread/ot-nxp/openthread/etc/gn/openthread.gni:38:28: Duplicate build argument declaration.
  openthread_config_deps = []
                           ^
Here you're declaring an argument that was already declared elsewhere.
You can only declare each argument once in the entire build so there is one
canonical place for documentation and the default value. Either move this
argument to the build config file (for visibility everywhere) or to a .gni file
that you "import" from the files where you need it (preferred).
See //third_party/connectedhomeip/third_party/openthread/repo/etc/gn/openthread.gni:38:28: Previous declaration.
  openthread_config_deps = []
                           ^
See also "gn help buildargs" for more on how build arguments work.
See //third_party/connectedhomeip/third_party/openthread/ot-nxp/openthread/src/core/BUILD.gn:28:1: whence it was imported.
import("../../etc/gn/openthread.gni")
^-----------------------------------
See //third_party/connectedhomeip/third_party/openthread/platforms/BUILD.gn:69:5: which caused the file to be included.
    "${openthread_repo_root}/src/core:libopenthread_core_headers"
```

GN Build system complains that some symbols are defined in multiple places: both in //third_party/connectedhomeip/third_party/openthread/ot-nxp/openthread and also in  //third_party/connectedhomeip/third_party/openthread/repo . Do you know how I could force the build system to exclude  //third_party/connectedhomeip/third_party/openthread/repo from build?

mspang, 2021-05-24:
This structure makes it nigh impossible to write a portable application that uses OpenThread, right?

mspang, 2021-05-24:
Or rather, I think we should bump the common openthread dependency and update all the platforms.

If this isn't possible, it's a serious problem. I do not think we are in a place where we can have chip-nxp, chip-efr, etc at this time. It would be a nightmare to keep everything building with such a separation.

doru91, 2021-05-25:
@mspang , maybe I wasn't clear, we don't need chip-nxp, chip-efr, etc. We only need different repos for the vendor examples.

So at this moment, inside connectedhomeip/third_party/openthread we have:
- "repo" - a submodule to an older commit of the openthread repo that contains both the openthread stack and _all_ the vendor examples;
- platform/efr32 - build rules for the efr32 examples;
- platform/k32w - build rules for the k32w examples;
- platform/qpg6100 - build rules for the Qorvo examples
- platform/BUILD.gn - build rules for the openthread stack using "repo" files for the openthread stack.

Due to the OpenThread refactorization, my initial proposal was that  connectedhomeip/third_party/openthread to have the following structure:
- "repo" - remove it;
- add ot-nxp repo as a submodule;
- add ot-efr32 repo as a submodule;
- add ot-qorvo as a submodule;
- platform/efr32 - build rules for efr32 using ot-efr;
- platform/k32w - build rules for k32w using ot-nxp;
- platform/qpg6100 - build rules for qorvo6100 using ot-qorvo;
- platform/BUILD.gn - build rules for the openthread stack using "repo" files from ot-nxp/openthread or from ot-efr32/openthread or from ot-qorvo/openthread depending on the platform.
So this is not possible? Instead of using the openthread core submodule of each vendor repo should we use a single openthread repo?

mspang, 2021-05-25:
Yes we should have a single version. Multiple versions does solve some problems but it creates others. Multiple exact versions constraints are going to be unsatifisable unless they are the same.

If the application wants to test against some version and ot-nxp wants a 2nd, it's not going to work.

More generally, and kind of diamond dependency is not going work unless there is enough backwards compatibility.

Here, the application is the CHIP examples and so CHIP should decide the version of OpenThread. If we can't get this working, then there is not enough backwards compatibility for people to actually use OpenThread.

jmartinez-silabs, 2021-05-25:
@mspang @doru91 I would need to validate, but I think not all sur all platforms (ot-nxp, ot-efr, ...) use the same openthread commit hash as submodule. This might be troublesome either way

doru91, 2021-05-26:
I,ve submitted https://github.com/project-chip/connectedhomeip/pull/7139 for addressing this issue. Help from Silabs and Qorvo is needed as the PR also touches their platforms.

doru91, 2021-06-03:
I'll close this issue as the PR encapsulating the changes was merged.

closed 2021-06-03 by doru91