project-chip/connectedhomeip #7078: BLE connection failure between Android chip tool and EFR32 Lock App

saurabhst, 2021-05-25:
#### Problem
BLE connection failure with the latest android chip tool built from master and EFR32 lock app because of which Thread commissioning using Android CHIP -tool fails. 

EFR32 side logs 
```
`<efr32 > ==================================================
<efr32 > chip-efr32-lock-example starting
<efr32 > ==================================================
<efr32 > Init CHIP Stack
<detail> [-] Initializing key value store
<detail> [-] KeyValueStore init complete: active keys 2, deleted keys 2, sectors 4, logical sector size 8192 bytes
<info  > [DL] Setting device name to : "EFR32_LOCK"
<efr32 > Initializing OpenThread stack
<info  > [DL] OpenThread started: OK
<info  > [DL] Setting OpenThread device type to ROUTER
<efr32 > Starting Platform Manager Event Loop
<efr32 > Starting OpenThread task
<efr32 > Starting App Task
<efr32 > Starting FreeRTOS scheduler
<info  > [DL] Bluetooth stack booted: v3.1.1-b214
<detail> [DL] Thread task running
<detail> [DL] CHIP task running
<info  > [ZCL] Using ZAP configuration...
<detail> [DL] Start BLE advertissement
<info  > [ZCL] deactivate report event
<info  > [ZCL] OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pairing table.
<detail> [DIS] Set the admin pairing table delegate
<info  > [ZCL] OpCreds: Call to writeAdminsIntoFabricsListAttribute
<info  > [ZCL] OpCreds: Stored 0 admins in fabrics list attribute.
<info  > [ZCL] OpCreds: Failed to write admin count 0 in fabrics list
<detail> [DIS] Init admin pairing table with server storage
<detail> [IN] TransportMgr initialized
<info  > [DL] _OnPlatformEvent default:  event->Type = 32778
<detail> [DL] OpenThread State Changed (Flags: 0x00038200)
<detail> [DL]    Network Name: OpenThread
<detail> [DL]    PAN Id: 0xFFFF
<detail> [DL]    Extended PAN Id: 0xDEAD00BEEF00CAFE
<detail> [DL]    Channel: 11
<detail> [DL]    Mesh Prefix: FDDE:AD00:BEEF::/64
<info  > [IN] local node id is 0x0000000000BC5C01
<detail> [BLE] Assigned local session key ID 0
<detail> [BLE] Waiting for PBKDF param request
<info  > [DIS] Start dns-sd server
<info  > [SVR] Server Listening...
<efr32 > Current Firmware Version: 0.1ALPHA
<info  > [DL] Device Configuration:
<info  > [DL]   Serial Number: TEST_SN
<info  > [DL]   Vendor Id: 9050 (0x235A)
<info  > [DL]   Product Id: 21323 (0x534B)
<info  > [DL]   Product Revision: 1
<info  > [DL]   Setup Pin Code: 73141520
<info  > [DL]   Setup Discriminator: 3840 (0xF00)
<info  > [DL]   Manufacturing Date: (not set)
<info  > [DL]   Device Type: 65535 (0xFFFF)
<info  > [SVR] SetupQRCode: [CH:W0GU245600C0NF0ET00]
<info  > [SVR] Copy/paste the below URL in a browser to see the QR Code:
<info  > [SVR] https://dhrishi.github.io/connectedhomeip/qrcode.html?data=CH%3AW0GU245600C0NF0ET00
<info  > [SVR] Manual pairing code: [35249644644]
<efr32 > App Task started
<detail> [DL] bleAdv Timeout : Start slow advertissment
<info  > [DL] Connect Event for handle : 1
<info  > [DL] Char Write Req, char : 26
<detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 9)
<info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
<info  > [BLE] local and remote recv window sizes = 3
<info  > [BLE] selected BTP version 3
<info  > [BLE] using BTP fragment sizes rx 128 / tx 20.
<info  > [DL] CHIPoBLE unsubscribe received
<info  > [DL] _OnPlatformEvent kCHIPoBLEUnsubscribe
<error > [IN] Failed to establish BLE connection: Ble Error 6000004 (0x005B8D84): BLE central unsubscribed
<detail> [IN] Clearing BLE pending packets.
<info  > [BLE] Releasing end point's BLE connection back to application.
```

 #### Proposed Solution
BLE connection should be successful between Android CHIP-tool as well as efr32 lock app.

saurabhst, 2021-05-25:
After having discussion with @Damian-Nordic it seems the current spec requires support for BLE GATT notifications instead of indications and Android  CHIP Tool sets the CCCD descriptor to receive notification . Its seems like EFR32 platform couldn't handle that write to the CCCD properly and interpreted that as an "unsubscribe" event. Here are the 2 relevant PRs which other platforms including EFR32 must implement to replace indications with GATT notifications -
1) https://github.com/project-chip/connectedhomeip/pull/6955
2)https://github.com/project-chip/connectedhomeip/pull/7039

jmartinez-silabs, 2021-06-03:
@mkardous-silabs

cjszz2001, 2021-06-09:
@mkardous-silabs, are you gonna fix this issue on EFR32 lock app?

fengli, 2021-08-04:
Just a side note here, I had a very similar issue with lighting app but for linux (I was running ubuntu server on Raspberry PI). more details #8774 

@austinh0 suspect that this might be the same issue, but for linux (bluez).

closed 2021-08-04 by jmartinez-silabs