project-chip/connectedhomeip #7108: Message counter enforcement needs to be different for encrypted and unencrypted message counters

bzbarsky-apple, 2021-05-25:
#### Problem
In the spec, the message counter value enforcement rules are different for encrypted and unencrypted messages.  We don't seem to be implementing that, if I look at `PeerMessageCounter::Verify`.

 #### Proposed Solution
Implement something more aligned with the spec.  But note https://github.com/project-chip/connectedhomeip/pull/7106#discussion_r639153813 where the spec for the unencrypted counter has various problems.

@kghost

kghost, 2021-05-26:
I guess the trouble comes from that whether we should persistent the local counter value. There are 2 questions to answer:

1. Should I persist my local counter value
2. Should peer persist my local counter value, which is his peer counter value

 * For global unsecure and secure counter
   * Local: global counter must be monotonic, so I must persist my local counter value.
   * Peer: global counter uses trust-first policy, so peer shouldn't persist the value

 * For secure session counter
   * Local: It depends on how CASE resumption works, I'm waiting for the spec.
   * Peer: It is required for CASE resumption to renegotiate the value, so peer shouldn't persist the value

bzbarsky-apple, 2021-05-26:
> Local: global counter must be monotonic

It's not monotonic in the spec for the global unsecure counter ...

But also, global secure and global unsecure have different behavior around wraparound in the spec, no?

bzbarsky-apple, 2022-02-23:
Need to sort out https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/4974 to fix this.

closed 2022-03-02 by woody-apple