project-chip/connectedhomeip #7120: resolve command using link local address on Mac

cjandhyala, 2021-05-26:
resolve command using link local address on Mac

chip-device-ctrl > resolve 0 12344321
[1622004214972] [0x4b066c] CHIP: [DL] Mdns: OnGetAddrInfo hostname:246F288FD758.local.
[1622004214972] [0x4b066c] CHIP: [DIS] Node ID resolved for 0x0000000000BC5C01
[1622004214974] [0x4b066c] CHIP: [CTL] SyncSetKeyValue on PairedDevicebc5c01
Node address has been updated
Current address: fe80::266f:28ff:fe8f:d758:11097
chip-device-ctrl > zcl OnOff Toggle 12344321 1 0
[1622004231856] [0x4affc6] CHIP: [DMG] ICR moving to [Initialize]
[1622004231858] [0x4affc6] CHIP: [DMG] ICR moving to [AddCommand]
[1622004231858] [0x4affc6] CHIP: [IN] Secure message was encrypted: Msg ID 3
[1622004231858] [0x4affc6] CHIP: [IN] Sending msg from 0x000000000001B669 to 0x0000000000BC5C01 at utc time: 130133227 msec
[1622004231858] [0x4affc6] CHIP: [IN] Sending secure msg on generic transport
[1622004231858] [0x4affc6] CHIP: [IN] Secure msg send status OS Error 2065 (0x00000811): No route to host
[1622004231858] [0x4affc6] CHIP: [-] OS Error 2065 (0x00000811): No route to host at ../../../../../src/app/CommandSender.cpp:75

Damian-Nordic, 2021-05-26:
@franck-apple The controller works fine. It's up to the device to decide which address the service is available on. Why is link-local wrong?

cjandhyala, 2021-05-26:
@Damian-Nordic , I have M5 provisioned on to WiFi. Isn't the error in the zcl  "Secure msg send status OS Error 2065 (0x00000811): No route to host" caused bcoz of controller using link local address.

bzbarsky-apple, 2021-05-28:
The "No route to host" bit is probably https://github.com/project-chip/connectedhomeip/issues/7232 ...  If we get the interface right, link-local should not be a problem.

And, importantly, link-local needs to work for all the people who don't have ipv6 dhcp set up on their routers.

bluebin14, 2021-06-01:
Mdns::OnGetAddrInfo gets called multiple times if kDNSServiceFlagsMoreComing is set in flags, then you can choose which IP and interface to use. The code however only looks at the first one which can be random and may not be the best choice.

closed 2021-06-04 by bzbarsky-apple

bzbarsky-apple, 2022-02-11:
> Mdns::OnGetAddrInfo gets called multiple times if kDNSServiceFlagsMoreComing is set in flags, then you can choose which IP and interface to use. The code however only looks at the first one which can be random and may not be the best choice.

Will try to fix this as I fix https://github.com/project-chip/connectedhomeip/issues/13274

PR #7388 (2021-06-04): Fix mdns resolution on darwin
#### Problem
There were two issues:
1. The address type argument to Resolve() was being ignored.
2. The interface id returned by resolution was not being propagated to the OnNodeIdResolved consumer.

#### Change overview
Propagate through the address type to the point where we can use it, and propagate the interface id out to our consumer.

#### Testing
1. Tested the address type bits by modifying `CHIPDeviceController.cpp` to only request IPv4 addresses, then testing pairing between all-clusters-app on an m5Stack and chip-tool on Mac using:
    ```
    ./out/debu [... cut]