project-chip/connectedhomeip #7124: ExchangeContext double free in chip-tool

gjc13, 2021-05-26:
#### Problem

Currently using chip-tool to pair a device via BLE will cause the process to crash.

Logs (Extra log lines added for debugging purpose): 
```
$ ./out/debug/chip-tool pairing ble-wifi ChipTE chiptest123 112233 11223344 291                                                              
......
CHIP:ZCL: DefaultResponse:
CHIP:ZCL:   Transaction: 0x560e43961db8
CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
CHIP:CTL: Device confirmed that it has received the root certificate
CHIP:CTL: Operational credentials provisioned on device 0x560e43ed1dc0
CHIP:TOO: Secure Pairing Success
CHIP:CTL: CHIPDevice::Reset
CHIP:TOO: Pairing Success
CHIP:TOO: Add network for device 12344321

CHIP:DMG: ICR moving to [Initialize]
CHIP:DMG: ICR moving to [AddCommand]
CHIP:CTL: LoadSecureSessionParametersIfNeeded for 0x560e43ed1dc0, state 0
CHIP:IN: Connection from 'BLE' expired
CHIP:CTL: OnConnectionExpired was called for unknown device, ignoring it.
CHIP:EM: Close exchange context 0x560e46f8bab8
Release 0x560e46f8bacc
Release 0x560e46f8ba10
CHIP:IN: New pairing for device 0x0000000000bc5c01, key 0!!
CHIP:CTL: mState OnNewConnection SecureConnected this 0x560e43ed1dc0
Retain 0x560e46f8ba10
Retain 0x560e46f8ba74
CHIP:IN: Secure message was encrypted: Msg ID 1
CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000BC5C01 at utc time: 1297987467 msec
CHIP:IN: Sending secure msg on generic transport
CHIP:IN: Secure msg send status No Error
Release 0x560e46f8ba74
CHIP:DMG: ICR moving to [   Sending]
CHIP:DMG: ICR moving to [Uninitiali]
ExchangeContext::Release 0x560e46f8bab8
Release 0x560e46f8bacc
[1]    1793885 abort (core dumped)  ./out/debug/chip-tool pairing ble-wifi ChipTE chiptest123 112233 11223344 291
```的

 #### Root cause

This is caused by `Device::Reset()` called when handling the message, the stack backtrace is:

```gdb
......
#0  chip::Controller::Device::Reset() (this=0x5555557ad7c0) at ../../third_party/connectedhomeip/src/controller/CHIPDevice.h:312             
#1  0x00005555556588c8 in chip::Controller::DeviceController::ReleaseDevice(chip::Controller::Device*)                                                                                                                                                                                    
    (this=0x555555793d98, device=0x555555793dc0) at ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:594            
#2  0x000055555565b9cb in chip::Controller::DeviceCommissioner::ReleaseDevice(chip::Controller::Device*)                                     
    (this=0x555555793d98, device=0x555555793dc0) at ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1390           
#3  0x000055555565891f in chip::Controller::DeviceController::ReleaseDevice(unsigned short) (this=0x555555793d98, index=0)                   
    at ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:601                                                         
#4  0x000055555565a556 in chip::Controller::DeviceCommissioner::RendezvousCleanup(int) (this=0x555555793d98, status=0)                       
    at ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1060                                                        
#5  0x000055555565b824 in chip::Controller::DeviceCommissioner::OnOperationalCredentialsProvisioningCompletion(chip::Controller::Device*)    
    (this=0x555555793d98, device=0x555555793dc0) at ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1368           
#6  0x000055555565b653 in chip::Controller::DeviceCommissioner::OnRootCertSuccessResponse(void*) (context=0x555555793d98)                    
    at ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1325                                                        
#7  0x0000555555631fc7 in IMDefaultResponseCallback(chip::app::Command const*, EmberAfStatus)                                                
    (commandObj=0x5555556ffdb8 <chip::app::sInteractionModelEngine+1112>, status=EMBER_ZCL_STATUS_SUCCESS)                                   
    at ../../third_party/connectedhomeip/src/controller/data_model/gen/CHIPClientCallbacks.cpp:303                                           
#8  0x000055555565bba9 in chip::Controller::DeviceControllerInteractionModelDelegate::CommandResponseProcessed(chip::app::CommandSender const*) (this=0x55555884e830, apCommandSender=0x5555556ffdb8 <chip::app::sInteractionModelEngine+1112>)
    at ../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1504                                                        
#9  0x00005555556bdce8 in chip::app::CommandSender::OnMessageReceived(chip::Messaging::ExchangeContext*, chip::PacketHeader const&, chip::PayloadHeader const&, chip::System::PacketBufferHandle&&)
    (this=0x5555556ffdb8 <chip::app::sInteractionModelEngine+1112>, apExchangeContext=0x55555884dab8, aPacketHeader=..., aPayloadHeader=..., aPayload=...) at ../../third_party/connectedhomeip/src/app/CommandSender.cpp:108
#10 0x000055555566acf2 in chip::Messaging::ExchangeContext::HandleMessage(chip::PacketHeader const&, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&)
    (this=0x55555884dab8, packetHeader=..., payloadHeader=..., peerAddress=..., msgBuf=...)                                                  
[... 6 lines cut ...]
#13 0x000055555566c535 in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SecureSessionHandle, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&, chip::SecureSessionMgr*)
    (this=0x55555884d8f0, packetHeader=..., payloadHeader=..., session=..., source=..., msgBuf=..., msgLayer=0x55555884ce60)                 
    at ../../third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:207                                                                   
#14 0x0000555555691f15 in chip::SecureSessionMgr::SecureMessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0x55555884ce60, packetHeader=..., peerAddress=..., msg=...)
    at ../../third_party/connectedhomeip/src/transport/SecureSessionMgr.cpp:473                                                              
#15 0x0000555555691553 in chip::SecureSessionMgr::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0x55555884ce60, peerAddress=..., msg=...) at ../../third_party/connectedhomeip/src/transport/SecureSessionMgr.cpp:297
#16 0x0000555555692db5 in chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0x5555588486d0, peerAddress=..., msg=...) at ../../third_party/connectedhomeip/src/transport/TransportMgrBase.cpp:57
#17 0x00005555556b4443 in chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0x5555588486f8, source=..., buffer=...) at ../../third_party/connectedhomeip/src/transport/raw/Base.h:92
#18 0x00005555556b4e9a in chip::Transport::BLEBase::OnEndPointMessageReceived(chip::Ble::BLEEndPoint*, chip::System::PacketBufferHandle&&)   
    (this=0x5555588486f8, endPoint=0x5555557014a0 <chip::Ble::BleEndPointPool::Get(unsigned long) const::sEndPointPool>, buffer=...)         
```

Then in  `PairingCommand::OnPairingComplete`, we asked the reset device to re-establish the session to send network commission cluster messages. This caused the connection to expire in `SecureSessionMgr::NewPairing`, which in turn causes the `ExchangeContext` to be released.

However, the `ExchangeContext::HandleMessage` isn't aware of these stories and calls `Release` at the end of the function. This leads to a double free.

 #### Proposed solutions

`chip-device-ctrl` is not affected by this issue since it sends pairing and network commissioning packets separately. However, it's natural to setup the network in the pairing callback just as `chip-tool` does. So I suggest to fix the message handling logic.

Possible solutions can be:
* Only release the exchange context if it's not released in `ExchangeContext::HandleMessage`. However we don't really have an api to check the validity of an ExchangeContext.
* Don't release the ExchangeContext automatically when connection expires and leave it to users. This on another hand can lead to ExchangeContext leak.

Since I'm no expert in message dispatching, I'd leave the fix to more capable engineers in this field.

kghost, 2021-05-26:
Working on it #7125

pan-apple, 2021-05-26:
Most likely, #7114 will fix this issue as well.

yufengwangca, 2021-05-26:
Yes, #7114 should address this issue. We should not close the exchange on connection close.

yufengwangca, 2021-05-27:
close this ticket with #7114

closed 2021-05-27 by yufengwangca