project-chip/connectedhomeip #7127: [How] How to understand and learn Matter?

InfiniteYuan, 2021-05-26:
As we all know, Matter is one of the most promising protocols in the IOT field in the future. So how should developers understand and learn Matter? And how to use it?

Will Matter officially have relevant tutorials and videos? This should be very necessary information for developers. When does Matter officially plan to release related tutorials and videos?

Thank you and look forward to your reply.

doru91, 2021-05-26:
A good place to start would be reproducing some of the demos available from the vendors: https://github.com/project-chip/connectedhomeip/tree/master/docs/guides

Also, the specification would be a very good starting point once it's open-source.

ShadowJonathan, 2022-07-25:
@stale no.

Also for anyone wishing to get some more info on Matter, i recommend [this site](https://matter-smarthome.de/en/), which tries to independently follow Matter and its developments.

ShadowJonathan, 2023-01-21:
@stale still no

closed 2023-02-04 by stale[bot]

ShadowJonathan, 2023-02-04:
This is NOT a stale issue