project-chip/connectedhomeip #7129: Build Android CHIP Tool App Failed

Lemon-7, 2021-05-26:
Facing error when I'm trying to build android CHIP tool app by running task "Build Android App" in vscode

Below are the logs which I see at the end. Can someone help me to solve this issue?

//*********** Logs **************//
+ ninja -C out/android_arm64 src/setup_payload/java src/controller/java default
ninja: Entering directory `out/android_arm64'
ninja: no work to do.
+ rsync -a out/android_arm64/lib/CHIPController.jar out/android_arm64/lib/SetupPayloadParser.jar src/android/CHIPTool/app/libs
scripts/examples/android_app.sh: line 43: rsync: command not found
The terminal process "/bin/bash '-c', 'TARGET_CPU=arm64 scripts/examples/android_app.sh'" terminated with exit code: 127.
//***********************//

austinh0, 2021-05-26:
`android_app.sh` works for me close to HEAD. Can you install `rsync`?

Lemon-7, 2021-05-27:
> `android_app.sh` works for me close to HEAD. Can you install `rsync`?

I have installed rsync, but it isn't work

austinh0, 2021-05-27:
Is rsync on your PATH? Does it make a difference if you run `TARGET_CPU=arm64 ./scripts/examples/android_app.sh` from terminal instead of using the VSCode task?

closed 2022-01-25 by woody-apple