project-chip/connectedhomeip #7135: tv-app gen/ files are modified by darwin helper.js changes

holbrookt, 2021-05-26:
#### Problem
When adding OtaSoftwareUpdateServer Cluster to `src/darwin/Framework/CHIP/templates/helper.js` inside `function asExpectedEndpointForCluster(clusterName)`, files inside `tv-app/tv-common/gen/` are modified after running `zap_regen_all.py`.

##### Steps to Reproduce:
1. checkout master
2. modify `helper.js` as such:
```
@@ -38,6 +38,7 @@ function asExpectedEndpointForCluster(clusterName)
   case 'NetworkCommissioning':
   case 'OperationalCredentials':
   case 'TrustedRootCertificates':
+  case 'OtaSoftwareUpdateServer':
     return 0;
   }
```
3. run `zap_regen_all.py`
4. Observe that `tv-app/` files have been modified. Observe that tv-app build fails

 #### Proposed Solution
Not sure. But it seems to me that changes to darwin `helper.js` should not affect `tv-app`.

holbrookt, 2021-06-10:
I believe this bug may be more widespread than is described in the initial comment. I'm noticing new behavior as of today:
* if I checkout master and run `./scripts/tools/zap_regen_all.py`, `tv-app` files are changed.
* I can run `./scripts/tools/zap_regen_all.py` multiple times, and `tv-app` files are changed every time

This is causing CI to fail such as in #6476

bzbarsky-apple, 2021-06-24:
So I think I know what might be going on with tv-app and ZAP not generating things deterministically for it.  It has the Server side of the "Content Launcher" cluster enabled on three endpoints:

1.  On endpoint 1, where none of the checkboxes are checked under "Commands".
1.  On endpoint 3, where the two "In" checkboxes are checked for the two Client -> Server commands.
1.  On endpoint 4, configured like endpoint 3.

Now the template involved is doing:
```
{{#all_user_clusters}}
{{#if (user_cluster_has_enabled_command name side)}}
```
and afaict `user_cluster_has_enabled_command` does `let endpointCommands = await queryZcl.exportCommandDetailsFromAllEndpointTypesAndClusters` and then walks the list to see which are enabled.  But `exportCommandDetailsFromAllEndpointTypesAndClusters` does an SQL query ending with `GROUP BY COMMAND.NAME, COMMAND.COMMAND_ID` so it will only pick up the incoming/outgoing values for one of the three instances of the LaunchContent command in this case.  And maybe it gets the one that's enabled, and maybe it gets the one that's not enabled....

@tecimovic @vivien-apple  where should this get fixed?  I can obviously change the .zap file to enable the commands on all endpoints, but that would just work around the underlying problem...  We should not be doing any grouping across things that might differ between the entries of the list in a meaningful way, it feels like.  That means at least `ENDPOINT_TYPE_COMMAND.INCOMING` and `ENDPOINT_TYPE_COMMAND.OUTGOING` though maybe also other things?

bzbarsky-apple, 2021-06-24:
For what it's worth, https://github.com/project-chip/connectedhomeip/pull/7867 will just enable these commands on endpoint 1 to work around this.

woody-apple, 2022-01-25:
We're in progress of reviewing older issues. Is this still an issue given how old the SDK this was tested against? Please re-open if so. Thanks!

bzbarsky-apple, 2022-01-26:
This is the thing that #13871 is hopefully going to fix....

yufengwangca, 2022-02-02:
@holbrookt could you check if this issue still exist?

bzbarsky-apple, 2022-02-02:
We put in place a workaround for tv-app, so it won't be reproducible there right now....

closed 2022-02-02 by bzbarsky-apple