project-chip/connectedhomeip #7141: Improve CHIP_EVENT_ID_FOUND in EventIterator

yunhanw-google, 2021-05-26:
This makes me wonder why we return something that is not 'CHIP_ERROR_' prefixed as a CHIP_ERROR.

it looks like we are trying to have two return values: real error and some status of found or not. This seems slightly counter-intuitive and maybe we could design our API differently.

Saying "Error of event id was found" sounds wrong, so it is not a CHIP_ERROR.

_Originally posted by @andy31415 in https://github.com/project-chip/connectedhomeip/pull/7110#discussion_r640033187_

kpschoedel, 2021-07-14:
There are several others like this. At least: CHIP_END_OF_TLV, CHIP_EVENT_ID_FOUND, CHIP_END_OF_INPUT, and CHIP_ERROR_SENTINEL (formerly code used CHIP_ERROR_MAX). Maybe we should just rename CHIP_ERROR to CHIP_STATUS?

yunhanw-google, 2021-07-14:
Agree, CHIP_STATUS looks good, @kpschoedel are you updating those? maybe assign this to you for overall update? thanks

closed 2022-10-18 by stale[bot]