project-chip/connectedhomeip #7188: chip-device-ctrl always crashes in DeviceController::Shutdown()

bluebin14, 2021-05-27:
#### Problem
chip-device-ctrl always crashes when the exit command is issued, with segmentation fault on the following place in src/controller/CHIPDeviceController .cpp DeviceController::Shutdown()

chip::Platform::Delete(mDefaultIMDelegate);

At that point mDefaultIMDelegate is a random uninitialized pointer.

 #### Proposed Solution
Ensure that mDefaultIMDelegate pointer is properly initialized e.g. in CHIPDeviceController.h

DeviceControllerInteractionModelDelegate * mDefaultIMDelegate = nullptr;

Might want to check the other uninitialized members too.

bluebin14, 2021-05-27:
@vivien-apple @bzbarsky-apple

bzbarsky-apple, 2021-05-27:
Looks like fallout from #5945?  If `params.imDelegate` is set in `DeviceController::Init` then we will end up in this state.  @erjiaqing 

```
DeviceControllerInteractionModelDelegate * mDefaultIMDelegate = nullptr;
```
Makes sense to me.  Want to post a PR @bluebin14 ?

bluebin14, 2021-05-28:
PR: https://github.com/project-chip/connectedhomeip/pull/7230

PR #7230 (2021-06-02): Fix crash caused by uninitialized variable
#### Problem
What is being fixed?
* Fix crash in DeviceController::Shutdown()
* Fixes #7188 chip-device-ctrl always crashes in DeviceController::Shutdown()

#### Change overview
This PR fixes crash caused by uninitialized member variable. The crash happened in DeviceController::Shutdown().

#### Testing
How was this tested? (at least one bullet point required)
* launch chip-device-ctrl
* commission accessory e.g. connect -ble 3840 12345678 12344321
* optionally enter any other commands
* issue exit command

Result without the fix: crash with segmentat [... cut]
closed 2021-06-02 by woody-apple