project-chip/connectedhomeip #7197: [mac] build failure in shell

turon, 2021-05-27:
#### Problem

Top-of-tree fails to build on mac:
```
./gn_build.sh

...

.--------------------------------
-- Build: GN configure
'--------------------------------
ERROR at //examples/platform/linux/AppMain.cpp:28:11: Include not allowed.
#include <lib/shell/Engine.h>
          ^-----------------
It is not in any dependency of
  //examples/platform/linux:app-main
The include file is in the target(s):
  //src/lib/shell:shell_core
which should somehow be reachable.
```

It works if I roll back to before this commit:
```
commit c8f286f6d7dcbc1c69925edd3ca5bef714824e27
Author: Jiacheng Guo <guojiacheng@espressif.com>
Date:   Thu May 27 22:45:48 2021 +0800
    [shell] enable shell as an optional module in apps (#7080)
    This PR allows users to build shell into the devices for debugging and
    testing purpose.
```

It seems the Mac CI is not building for all configurations.

turon, 2021-05-27:
Duplicate of #7199.
Also #7200 split out as well.

turon, 2021-05-27:
Also #7200 split out as well.

woody-apple, 2022-01-25:
Looks complete?

closed 2022-01-25 by woody-apple