project-chip/connectedhomeip #7235: Something goes wrong with memory and CASE sessions

bzbarsky-apple, 2021-05-28:
#### Problem
Steps to reproduce:

1. Pull ToT (rev `34bccf5d90edd8c1c465f59c565cd13a0b058d62` for me right now).
2. Build all-clusters-app for m5stack:
    ```
    % cd examples/all-clusters-app/esp32
    % source idf.sh
    % idf make erase_flash flash monitor -j16 ESPPORT=/dev/tty.SLAB_USBtoUART
    ```
3. In another terminal:
    ```
    % ./gn_build.sh
    % ./out/debug/standalone/chip-tool pairing ble-wifi SSID PASSWD 0 12345678 3840
    % ./out/debug/standalone/chip-tool onoff toggle 1
    % ./out/debug/standalone/chip-tool onoff toggle 1
    % ./out/debug/standalone/chip-tool onoff toggle 1
    % ./out/debug/standalone/chip-tool onoff toggle 1
    ```

I see a bunch of asserts/crashes/backtraces on the m5stack.  See attached log, search for "backtrace".  They generally seem to be happening during heap operations from crypto code.

 #### Proposed Solution
Figure out what's going on (are we running out of heap?) and fix it.

@pan-apple

pan-apple, 2021-05-28:
It seems we are running out of heap on ESP32 platform. Likely a memory leak somewhere. 

```
    /* Note: this check must be done while holding the lock as both
       malloc & realloc may temporarily shrink the free_bytes value
       before they split a large block. This can result in false negatives,
       especially if the heap is unfragmented.
    */
    if (heap->free_bytes < size) {
        MULTI_HEAP_UNLOCK(heap->lock);
        return NULL;
    }
```

bzbarsky-apple, 2021-05-28:
I just realized all-clusters-app actually logs its free heap amounts.  Just those lines from the log:

```
(1002) app-devicecallbacks: Current free heap: 108376
(1022) app-devicecallbacks: Current free heap: 108376
```
<details>
<summary>A bunch of output to be expanded</summary>

```
(1102) app-devicecallbacks: Current free heap: 108376
(1132) app-devicecallbacks: Current free heap: 108376
(1392) app-devicecallbacks: Current free heap: 105548
(1422) app-devicecallbacks: Current free heap: 89628
(1442) app-devicecallbacks: Current free heap: 89632
(1462) app-devicecallbacks: Current free heap: 89624
(1492) app-devicecallbacks: Current free heap: 88988
(1592) app-devicecallbacks: Current free heap: 85760
(1612) app-devicecallbacks: Current free heap: 85760
(1652) app-devicecallbacks: Current free heap: 84660
(1682) app-devicecallbacks: Current free heap: 84660
(1702) app-devicecallbacks: Current free heap: 84660
(1722) app-devicecallbacks: Current free heap: 84660
(1752) app-devicecallbacks: Current free heap: 84660
(1772) app-devicecallbacks: Current free heap: 84660
(1802) app-devicecallbacks: Current free heap: 84660
(1832) app-devicecallbacks: Current free heap: 84656
(1862) app-devicecallbacks: Current free heap: 84656
(1892) app-devicecallbacks: Current free heap: 84656
(1912) app-devicecallbacks: Current free heap: 84656
(1982) app-devicecallbacks: Current free heap: 83400
(2012) app-devicecallbacks: Current free heap: 83268
(5552) app-devicecallbacks: Current free heap: 77864
(9622) app-devicecallbacks: Current free heap: 76384
(12902) app-devicecallbacks: Current free heap: 65120
(13072) app-devicecallbacks: Current free heap: 68048
(13092) app-devicecallbacks: Current free heap: 68048
(16992) app-devicecallbacks: Current free heap: 70500
(17012) app-devicecallbacks: Current free heap: 70500
(18852) app-devicecallbacks: Current free heap: 70272
(18932) app-devicecallbacks: Current free heap: 68448
(18952) app-devicecallbacks: Current free heap: 69812
(19022) app-devicecallbacks: Current free heap: 69508
(19092) app-devicecallbacks: Current free heap: 69724
(28602) app-devicecallbacks: Current free heap: 60768
(28622) app-devicecallbacks: Current free heap: 60768
(28652) app-devicecallbacks: Current free heap: 60768
(36952) app-devicecallbacks: Current free heap: 62588
(36972) app-devicecallbacks: Current free heap: 62588
(42622) app-devicecallbacks: Current free heap: 56880
(42642) app-devicecallbacks: Current free heap: 56880
(42672) app-devicecallbacks: Current free heap: 56880
(45312) app-devicecallbacks: Current free heap: 57752
(45342) app-devicecallbacks: Current free heap: 57752
(50832) app-devicecallbacks: Current free heap: 52048
(50852) app-devicecallbacks: Current free heap: 52048
(50872) app-devicecallbacks: Current free heap: 52040
(54992) app-devicecallbacks: Current free heap: 52904
(55022) app-devicecallbacks: Current free heap: 52912
(59622) app-devicecallbacks: Current free heap: 47204
(59652) app-devicecallbacks: Current free heap: 47204
(59672) app-devicecallbacks: Current free heap: 47204
(62792) app-devicecallbacks: Current free heap: 48080
(62812) app-devicecallbacks: Current free heap: 48080
(71522) app-devicecallbacks: Current free heap: 42400
(71542) app-devicecallbacks: Current free heap: 42400
(71572) app-devicecallbacks: Current free heap: 42400
(80582) app-devicecallbacks: Current free heap: 43280
(80602) app-devicecallbacks: Current free heap: 43280
(92112) app-devicecallbacks: Current free heap: 37568
(92142) app-devicecallbacks: Current free heap: 37568
(92162) app-devicecallbacks: Current free heap: 37568
```

</details>

```
(103412) app-devicecallbacks: Current free heap: 38448
(103442) app-devicecallbacks: Current free heap: 36748
(116622) app-devicecallbacks: Current free heap: 32732
(116642) app-devicecallbacks: Current free heap: 32732
(116672) app-devicecallbacks: Current free heap: 32732
```

woody-apple, 2022-01-25:
Looks resolved?

closed 2022-01-25 by woody-apple

bzbarsky-apple, 2022-01-26:
Yes, haven't hit this in a while.