project-chip/connectedhomeip #7281: BIGENDIAN_CPU is not defined in the build system for big-endian builds

jrhees-cae, 2021-06-01:
#### Problem
The zcl code in CHIP relies on the BIGENDIAN_CPU define to affect endianness behavior. However, BIGENDIAN_CPU is not defined by default on gnu compilers for bigendian targets.

 #### Proposed Solution
Possible solutions:
1. Change the references to BIGENDIAN_CPU in the zcl code and zap templates to more universal endian identifier
2. Add a global define for BIGNEDIAN_CPU to the root BUILD.gn

gerickson, 2021-06-01:
Would definitely vote for the ZCL/ZAP code to use the same constructs found in the underlying CHIP/Matter [nlio package](https://github.com/project-chip/connectedhomeip/blob/master/.gitmodules#L9).

bzbarsky-apple, 2021-11-04:
@mspang What's the best option here?  I'm happy to change the ZCL code to use whatever define is most convenient....

woody-apple, 2022-11-02:
Issue Scrub: Keeping this open, however needs someone to champion this work, moving to Future.

closed 2023-09-18 by mergify[bot]