project-chip/connectedhomeip #7285: Do not send ack for unsolicited messages with no unsolicited message handler

pan-apple, 2021-06-01:
#### Problem
As per specifications, the exchange manager should silently drop the unsolicited messages that do not have an unsolicited message handler registered. This includes not sending an ack for such messages (in case R flag is set on the received message).

The current code is creating a temporary exchange and sending an ack back. The code should be updated to  match the spec.

The last row of the following table applies to this issue. The table is copied from spec at commit: https://github.com/CHIP-Specifications/connectedhomeip-spec/commit/3d47b6efc0c90c083181280a8d1dec22794eb23b

[image]

 #### Proposed Solution
Update code to not ack unsolicited messages if there in no registered unsolicited message handler.

A unit test should be added that ensures that the ack is not received even after all the retries are exhausted for such messages.

The current tests should be updated to register an unsolicited message handler for the message being sent.
Update tests: `CheckResendApplicationMessage` and `CheckCloseExchangeAndResendApplicationMessage`.

turon, 2021-06-02:
Yes, to clarify, the current spec (and this table) currently states (via row 2 and 3) that ack response should go back regardless of whether a protocol handler is registered.  The scenario seems highly unlikely (except possibly during development) for a valid active session to exist (with shared credentials and counter sync) but the protocol id is not registered on the receiver.

The motivation to ack in all cases was:
1) to be a good citizen to the peer node and cut short the retransmissions it needs to do. 
2) to enforce layer separation between MRP and layer that performs Protocol Id processing

However, I'm supportive of an "always drop unregistered protocol id" policy.  That would mean adjusting the spec so row 2 and 3 to have `Y` in column 1.  The only pause would be motivation (2) above.  Thoughts?

pan-apple, 2021-06-02:
I am mainly concerned about CRMP for CASE and PASE session establishment exchanges. Those messages are not encrypted. So anyone on the network can send a node SigmaR1 message (or PBKDFParamRequest) and it’ll still send an ack back.

In other cases, the message has to first be decrypted before it reaches CRMP/ExchangeManager. So the sender has to have a valid key to encrypt and send the message. Otherwise, it'll fail decryption and get dropped anyway.

turon, 2021-06-02:
Yes, I agree the unencrypted channel behavior makes the choice to use an "always drop unregistered" policy an easy one.

bzbarsky-apple, 2021-06-03:
> Yes, to clarify, the current spec (and this table) currently states (via row 2 and 3) that ack response should go back regardless of whether a protocol handler is registered. 

@turon I don't think it does.  It says that an ack response should go back for a "duplicate message".  For non-duplicates it does not say you send an ack.

Note that there are open spec issues about this table being incredibly confusing and the general fact that behavior here is badly underspecified.  The fact that SDK developers can't tell what the right behavior is from the spec without having a personal conversation with a spec editor (and often not even then) is a symptom of that...

turon, 2021-06-03:
@bzbarsky-apple Yes, agree with your points here.  Minimally, the middle column header needs to be amended to "duplicate secure message that needs MRP ack" to clarify that duplicate unencrypted messages don't send back acks as readily.  Agree that the table needs normative text to spell out the various cases.

Also, given the initial ack is only sent when a protocol handler is registered, specifying that the same coupling occurs for duplicates simplifies things.

bzbarsky-apple, 2022-05-19:
After the changes in https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5217 I believe the SDK matches the spec.

closed 2022-05-25 by woody-apple