project-chip/connectedhomeip #7287: Set up tests for Android CHIPTool

austinh0, 2021-06-01:
#### Problem
Android CHIPTool has no testing infrastructure.

 #### Proposed Solution
Set up Robolectric tests.

turon, 2022-02-01:
This is an older issue.  
Android tests are required for v1.
They likely exist now in a form sufficient to close this, but I haven't validated.
Removing tag `v1_triage_split_4`

austinh0, 2022-02-01:
I still need to work on this. Closing #7286 as duplicate.

utzcoz, 2022-08-27:
For Android tests, we can use [Robolectric](https://github.com/robolectric/robolectric) as local test infra to run many Android related tests on JVM without Emulator and real devices, and we can use [android-test](https://github.com/android/android-test) with Espresso API to test behaviors on Emulator. All of them can be run GitHub Actions for CI purpose. We can run Robolectric on JVM with a normal JVM and Android building environment, and run Emulator tests with [android-emulator-runner](https://github.com/ReactiveCircus/android-emulator-runner). Hi @austinh0 , what is the current progress of this feature?

yufengwangca, 2023-01-17:
Currently, Matter Java APIs is tested with java-matter-controller in GitHub CI, I think we are talking about testing Matter Java API, not Android ChipTool example app, right?

yufengwangca, 2023-01-17:
We have Matter Java APIs tested with java-matter-controller in GitHub CI. Android CHIPTool need to run either in emulator or real device, so it is not good candidate to test Java API in GitHub CI.

closed 2023-01-17 by yufengwangca