project-chip/connectedhomeip #7315: Resolve failed to update node address

jmartinez-silabs, 2021-06-02:
#### Problem

The command resolve on fails to update the node address with repeating error 
```
resolve 0 2
CHIP:DIS: Attempt to mDNS broadcast to an unreachable destination.
Failed to update node address: 4072
Failed to update node address: 4072
Failed to update node address: 4072
Failed to update node address: 4072
chip-device-ctrl > Failed to update node address: 4072
Failed to update node address: 4072
Node address has been updated
Node address has been updated
Node address has been updated
Node address has been updated
```

chip._tcp mDNs is present so controller should be able to resolve the address

```
ubuntu@ubuntuservercontroller:~/connectedhomeip$ avahi-browse -rt _chip._tcp
+  wlan0 IPv6 0000000000000000-0000000000000002             _chip._tcp           local
+  wlan0 IPv4 0000000000000000-0000000000000002             _chip._tcp           local
=  wlan0 IPv6 0000000000000000-0000000000000002             _chip._tcp           local
   hostname = [7E07D373D6F3B902.local]
   address = [fd11:22::f5f6:669:37a3:662b]
   port = [11097]
   txt = ["CRA=300" "CRI=5000"]
=  wlan0 IPv4 0000000000000000-0000000000000002             _chip._tcp           local
   hostname = [7E07D373D6F3B902.local]
   address = [fd11:22::f5f6:669:37a3:662b]
   port = [11097]
   txt = ["CRA=300" "CRI=5000"]
```
[image]

Setup
- Test_event_3 branch code
- Python controller chip-device-ctrl on raspi ubuntu
- Lock or Light example on EFR32 platform (with both MG12 or MG21)

I attached the
[EFR32_LOG.txt](https://github.com/project-chip/connectedhomeip/files/6584683/EFR32_LOG.txt)
[chip-device-ctrl_log.txt](https://github.com/project-chip/connectedhomeip/files/6584687/chip-device-ctrl_log.txt)

 logs of the efr device and controller

andy31415, 2021-06-02:
Do we know if this is using minimal mdns or platform mdns?

jmartinez-silabs, 2021-06-02:
Adding topology info of my setup

1st raspberry pi 4 running with a EFR32 RCP running the OT BR built from ot-br-posix (commit 8f23e77966a36ed67eb99b9be14e6189806977b6 )
With a thread network formed
```
sudo ot-ctl dataset active
Active Timestamp: 0
Channel: 15
Channel Mask: 0x07fff800
Ext PAN ID: 1111111122222222
Mesh Local Prefix: fd0c:62d:8065:76d9::/64
Master Key: 00112233445566778899aabbccddeeff
Network Name: OpenThreadDemo
PAN ID: 0x1234
PSKc: 61e1206d2c2b46e079eb775f41fc7219
Security Policy: 672 onrcb
Done
```

2nd raspberry pi with ubuntu 20.10 
controller built with `scripts/build_python.sh` so mdns is by default at minimal

command sent
```
connect -ble 3840 73141520 2

zcl NetworkCommissioning AddThreadNetwork 2 0 0 operationalDataset=hex:0e080000000000000000000300000f35060004001fffe0020811111111222222220708fd0c062d806576d9051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f01021234041061e1206d2c2b46e079eb775f41fc72190c0402a0fff8 breadcrumb=0 timeoutMs=1000

zcl NetworkCommissioning EnableNetwork 2 0 0 networkID=hex:1111111122222222 breadcrumb=0 timeoutMs=1000

Close-le

resolve 0 2
```
DUT : EFR32 MG21 lighting app built and flash with
```
./scripts/examples/gn_efr32_example.sh examples/lighting-app/efr32/ out/lighting_app BRD4180A
python3 out/lighting_app/BRD4180A/chip-efr32-lighting-example.flash.py
```

jmartinez-silabs, 2021-06-02:
New info, a colleague was succesful with a setup similar to mine, the two differences are

ot-br-posix commit hash 72815d01885f2ecbcfbd24d6d8ae72606c07a398
python controller built with chip_mdns= "platform"
```
cd src/controller/python/
rm -rf out
source ../../../scripts/activate.sh 
gn gen out/debug --args='chip_mdns="platform"'
ninja -C out/debug python
```

I will try that setup also

andy31415, 2021-06-02:
mdns platform vs minimal should will make a huge difference.
I would expect if we want minmdns to run on raspberry, we may want to shut down avahi (although I am not sure if that would be a problem or not running them in parallel - they would both try to listen on the same UDP port otherwise)

jmartinez-silabs, 2021-06-02:
Resolve works using 'chip_mdns="platform"'

closed 2021-06-02 by jmartinez-silabs