project-chip/connectedhomeip #7320: [K32W] Failure during the BLE Rendezvous session

doru91, 2021-06-02:
SPAKE2+ session is successful. Then the commissioner sends a CSR Request to the DUT, it looks like the DUT replies with a CSR Request Response but the chip-device-ctrl looks to timeout the message. 

Head is at 

```
commit f6af7b9c9d86174f2e894d54e90dbf9cef250ad0 (upstream/master, upstream/master)
Author: Song Guo <songguo@google.com>
Date:   Wed Jun 2 13:15:21 2021 +0800
```

    [python] Add flag to disable BLE for controller (#7179)

I checked with other vendors and everything is ok on their side. I expect that there is a problem on the DUT side. Any help on how to debug this would be appreciated.

Logs on the device side:
```
[Info]Welcome to NXP ELock Demo App
[Info]OpenThread started: OK
[Info]Setting OpenThread device type to SLEEPY END DEVICE
[Info]OpenThread polling interval set to 1000ms
[Info]Using ZAP configuration...
[Info]deactivate report event
[Info]OpCreds: Initiating OpCreds cluster by writing fabrics list from admin pairing table.
[Info]Set the admin pairing table delegate
[Info]OpCreds: Call to writeAdminsIntoFabricsListAttribute
[Info]OpCreds: Stored 0 admins in fabrics list attribute.
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Unknown cluster ID: 40
[Info]Init admin pairing table with server storage
[Info]TransportMgr initialized
[Info]local node id is 0x0000000000BC5C01
[Info]Start dns-sd server
[Info]Server Listening...
[Info]SetupQRCode: [CH:Y3.13G3600A50324800]
[Info]Copy/paste the below URL in a browser to see the QR Code:
[Info]https://dhrishi.github.io/connectedhomeip/qrcode.html?data=CH%3AY3.13G3600A50324800
[Info]Manual pairing code: [34970112332]
[Info]Current Firmware Version: 04-2020-te2
[Info]Thread task running
[Info]CHIP task running
[Info]OpenThread State Changed (Flags: 0x00038200)
[Info]   Network Name: OpenThread
[Info]   PAN Id: 0xFFFF
[Info]   Extended PAN Id: 0xDEAD00BEEF00CAFE
[Info]   Channel: 11
[Info]   Mesh Prefix: FDDE:AD00:BEEF::/64
[Info]   Master Key: 4D69D05232A8447494BA06C0AF1F0F5D
[Info]Assigned local session key ID 0
[Info]Waiting for PBKDF param request
[Info]Started BLE Advertising!
[Info]Started Advertising.
[Info]Started NFC Tag Emulation!
[Info]BLE is connected with device: 0.

[Info]Started Advertising.
[Info]NFC Tag emulation is already started!
[Info]BLE MTU size has been changed to 247.
[Info]local and remote recv window sizes = 3
[Info]selected BTP version 3
[Info]using BTP fragment sizes rx 128 / tx 20.
[Info]BLE EndPoint 0x400ea50 Connection Complete
[Info]Long dispatch time: 300 ms
[Info]Received message of type 32 and protocolId 0 on exchange 39017
[... 254 lines cut ...]
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: HandlePlatformSpecificBLEEvent 16387
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:DL: Indication received, conn = 0xffff8c038140
CHIP:DL: HandlePlatformSpecificBLEEvent 16389
CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 0
```

pan-apple, 2021-06-02:
It doesn’t seem that the controller is getting the OpCSR response from the DUT. I can see the DUT did send a message. Can you add more logs in this function to make sure that it did the correct response?

`emberAfOperationalCredentialsClusterOpCSRRequestCallback()`

The function has logs for error scenarios, but not sure if those are not getting printed for some reason.

For reference, we should see some of the following logs on the controller side when it receives the OpCSR.
```
[1622647452470] [0x1701a3] CHIP: [EM] Received message of type 9 and protocolId 327680 on exchange 44250
[1622647452470] [0x1701a3] CHIP: [DMG] InvokeCommand =
[1622647452470] [0x1701a3] CHIP: [DMG] {
[1622647452470] [0x1701a3] CHIP: [DMG]  CommandList =
[1622647452470] [0x1701a3] CHIP: [DMG]  [
[1622647452470] [0x1701a3] CHIP: [DMG]          CommandDataElement =
[1622647452470] [0x1701a3] CHIP: [DMG]          {
[1622647452470] [0x1701a3] CHIP: [DMG]                  CommandPath =
[1622647452470] [0x1701a3] CHIP: [DMG]                  {
[1622647452470] [0x1701a3] CHIP: [DMG]                          EndpointId = 0x0,
[1622647452470] [0x1701a3] CHIP: [DMG]                          ClusterId = 0x3e,
[1622647452470] [0x1701a3] CHIP: [DMG]                          CommandId = 0x5,
[1622647452470] [0x1701a3] CHIP: [DMG]                  },
[1622647452470] [0x1701a3] CHIP: [DMG] 
[1622647452470] [0x1701a3] CHIP: [DMG]                  CommandData = 
[1622647452470] [0x1701a3] CHIP: [DMG]                  {
[1622647452470] [0x1701a3] CHIP: [DMG]                          0x0 = [
[1622647452470] [0x1701a3] CHIP: [DMG]                                  0x30, 0x81, 0xcb, 0x30, 0x70, 0x2, 0x1, 0x0, 0x30, 0xe, 0x31, 0xc, 0x30, 0xa, 0x6, 0x3, 0x55, 0x4, 0xa, 0xc, 0x3, 0x43, 0x53, 0x52, 0x30, 0x59, 0x30, 0x13, 0x6, 0x7, 0x2a, 0x86, 0x48, 0xce, 0x3d, 0x2, 0x1, 0x6, 
[1622647452470] [0x1701a3] CHIP: [DMG]                  ]
[1622647452470] [0x1701a3] CHIP: [DMG]                          0x1 = [
[1622647452470] [0x1701a3] CHIP: [DMG]                                  0x14, 0x5c, 0xbd, 0xd, 0xce, 0xc0, 0xa8, 0x91, 0x9b, 0xd6, 0x73, 0xc4, 0x84, 0xc, 0x2a, 0x57, 0x23, 0xda, 0xa, 0x4, 0xd6, 0x18, 0x1c, 0x10, 0x9e, 0x19, 0xa7, 0xee, 0x60, 0xbd, 0x3e, 0x11, 
[1622647452470] [0x1701a3] CHIP: [DMG]                  ]
[1622647452470] [0x1701a3] CHIP: [DMG]                          0x2 = [
[1622647452470] [0x1701a3] CHIP: [DMG] 
[1622647452470] [0x1701a3] CHIP: [DMG]                  ]
[1622647452470] [0x1701a3] CHIP: [DMG]                          0x3 = [
[1622647452470] [0x1701a3] CHIP: [DMG] 
[1622647452470] [0x1701a3] CHIP: [DMG]                  ]
[1622647452470] [0x1701a3] CHIP: [DMG]                          0x4 = [
[1622647452470] [0x1701a3] CHIP: [DMG] 
[1622647452470] [0x1701a3] CHIP: [DMG]                  ]
[1622647452470] [0x1701a3] CHIP: [DMG]                          0x5 = [
[1622647452470] [0x1701a3] CHIP: [DMG] 
[1622647452470] [0x1701a3] CHIP: [DMG]                  ]
[1622647452470] [0x1701a3] CHIP: [DMG]                  },
[1622647452470] [0x1701a3] CHIP: [DMG]          },
[1622647452470] [0x1701a3] CHIP: [DMG] 
[1622647452470] [0x1701a3] CHIP: [DMG]  ],
[1622647452470] [0x1701a3] CHIP: [DMG] 
[1622647452470] [0x1701a3] CHIP: [DMG] }
[1622647452470] [0x1701a3] CHIP: [ZCL] Received Cluster Command: Cluster=3e Command=5 Endpoint=0
[1622647452470] [0x1701a3] CHIP: [ZCL] OpCSRResponse:
[1622647452470] [0x1701a3] CHIP: [ZCL]   CSR: 206
[1622647452470] [0x1701a3] CHIP: [ZCL]   CSRNonce: 32
[1622647452470] [0x1701a3] CHIP: [ZCL]   VendorReserved1: 0
[1622647452470] [0x1701a3] CHIP: [ZCL]   VendorReserved2: 0
[1622647452470] [0x1701a3] CHIP: [ZCL]   VendorReserved3: 0
[1622647452470] [0x1701a3] CHIP: [ZCL]   Signature: 0
[1622647452470] [0x1701a3] CHIP: [CTL] Received certificate signing request from the device
```

doru91, 2021-06-03:
Thanks @pan-apple for helping me. 

I step into emberAfOperationalCredentialsClusterOpCSRRequestCallback and it returns with success.

Then I took a closer look at the logs on the chip-device-ctrl-side:

> CHIP:CTL: Sending OpCSR request to 0x2eeabb08 device
> CHIP:DMG: ICR moving to [Initialize]
> CHIP:DMG: ICR moving to [AddCommand]
> CHIP:IN: Secure message was encrypted: Msg ID 0
> CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000000001 at utc time: 857971535 msec
> CHIP:IN: Sending secure msg on generic transport
> CHIP:IN: Secure msg send status No Error
> CHIP:DMG: ICR moving to [   Sending]
> CHIP:CTL: Sent OpCSR request, waiting for the CSR
> CHIP:DL: HandlePlatformSpecificBLEEvent 16387
> CHIP:DL: HandlePlatformSpecificBLEEvent 16387
> CHIP:DL: Indication received, conn = 0xffff80038120
> CHIP:DL: HandlePlatformSpecificBLEEvent 16389
> CHIP:DL: HandlePlatformSpecificBLEEvent 16387
> CHIP:DL: Indication received, conn = 0xffff80038120
> CHIP:DL: HandlePlatformSpecificBLEEvent 16389
> CHIP:DL: HandlePlatformSpecificBLEEvent 16387
> CHIP:DL: Indication received, conn = 0xffff80038120
> CHIP:DL: HandlePlatformSpecificBLEEvent 16389
> CHIP:DL: Indication received, conn = 0xffff80038120
> CHIP:DL: HandlePlatformSpecificBLEEvent 16389
> CHIP:DL: HandlePlatformSpecificBLEEvent 16387
> CHIP:DL: Indication received, conn = 0xffff80038120
> CHIP:DL: HandlePlatformSpecificBLEEvent 16389
> CHIP:DL: Indication received, conn = 0xffff80038120
> CHIP:DL: HandlePlatformSpecificBLEEvent 16389
> CHIP:DL: HandlePlatformSpecificBLEEvent 16387
> CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39018
> CHIP:ZCL: DefaultResponse:
> CHIP:ZCL:   Transaction: 0xffff91977378
> CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
> CHIP:CTL: Device failed to receive the CSR request Response: 0x01

So there are some indications that are being received from the device side. Then I looked into the device side, and these indications are being sent immediately after the CSR Op Rsp is generated, so they should contain DER bytes. As long as the SPAKE2+ session was successful, I don't expect that BLE transport lost any byte. Here are the relevant logs on the device side (first I print the DER bytes inside emberAfOperationalCredentialsClusterOpCSRRequestCallback then I print the BLE bytes in BLEManagerImpl.cpp):

```
[Info]Received message of type 8 and protocolId 327680 on exchange 39018
[Info]ec id: 39018, eived Cluster Command: Cluster=3e Command=hx Endpoint=hx
[Info]OpCreds: commissioner has requested an OpCSR
[Info]OpCreds: Finding admin with fabricId 0x000000000001B669.
[Info]Checking ind:0 [fabricId 0x000000000001B669 deId 0x000000000 0]
[Info]Found a match!
[Info]OpCreds: NewCertificateSigningRequest returned 0
[Info]ICR moving to [AddCommand]
[Info]csr[0]=30
[Info]csr[1]=81
[Info]csr[2]=ca
[Info]csr[3]=30
[Info]csr[4]=70
[Info]csr[5]=2
[Info]csr[6]=1
[Info]csr[7]=0
[Info]csr[8]=30
[Info]csr[9]=e
[Info]csr[10]=31
[Info]csr[11]=c
[Info]csr[12]=30
[Info]csr[13]=a
[Info]csr[14]=6
[Info]csr[15]=3
[Info]csr[16]=55
[Info]csr[17]=4
[Info]csr[18]=a
[Info]csr[19]=c
[... 622 lines cut ...]
[Info]Long dispatch time: 410 ms
[Info]Trying to send indication. Len: 13
[Info]data[0]=e
[Info]data[1]=12
[Info]data[2]=21
[Info]data[3]=52
[Info]data[4]=0
[Info]data[5]=40
[Info]data[6]=92
[Info]data[7]=85
[Info]data[8]=87
[Info]data[9]=ec
[Info]data[10]=25
[Info]data[11]=8
[Info]data[12]=a8
[Info]BLE Event - Sent :-)
[Info]Long dispatch time: 210 ms
[Info]Attribute write request(device: 0,handle: 14).
[Info]Write request/command received forCHIPoBLE RX characteristic (con 0, len 3)
[Info]Trying to send indication. Len: 3
[Info]data[0]=8
[Info]data[1]=14
[Info]data[2]=23
[Info]BLE Event - Sent :-)
```

doru91, 2021-06-03:
After several hours of debugging I found a fix:

```
diff --git a/src/messaging/ExchangeContext.cpp b/src/messaging/ExchangeContext.cpp
index 5aa546cda69..ae54556138f 100644
--- a/src/messaging/ExchangeContext.cpp
+++ b/src/messaging/ExchangeContext.cpp
@@ -73,7 +73,9 @@ void ExchangeContext::SetResponseExpected(bool inResponseExpected)

 void ExchangeContext::SetResponseTimeout(Timeout timeout)
 {
-    mResponseTimeout = timeout;
+    ChipLogError(ExchangeManager, "Set timeout %d", timeout);
+
+    mResponseTimeout = timeout * 2;
 }
```

The initial value for the mResponseTimeout is 30000 (30s?), but the timeout error occurs after 20s - I don't understand why. Doubling the value fixed the Rendezvous session.

pan-apple, 2021-06-03:
That's really weird. Does doubling it change it to 40s? It would indicate some issue with the timer itself.
Also, 20s should have been long enough to receive the CSR response.

doru91, 2021-06-04:
With the default value, the timer always expires in the middle of the CSR response reception - this happens after ~16-17s after the "connect -ble" is issued. However, it's unclear to me if the timer is armed for the whole rendezvous session or only for the CSR response.

doru91, 2021-06-08:
This timer issue seems to be related to https://github.com/project-chip/connectedhomeip/issues/7005: the embedded device sent the required packets but the chip-device-ctrl times out.

pan-apple, 2021-06-08:
@doru91 , I think this function is setting up the response timer

```
CHIP_ERROR CommandSender::SendCommandRequest(NodeId aNodeId, Transport::AdminId aAdminId, SecureSessionHandle * secureSession)
```

specifically, here
```
    mpExchangeCtx->SetResponseTimeout(kImMessageTimeoutMsec);
```

Looks like the default timeout is 6 seconds.
```
uint32_t kImMessageTimeoutMsec  = 6000;
```

pan-apple, 2021-06-09:
@doru91, I have submitted this issue (https://github.com/project-chip/connectedhomeip/issues/7489) to have a proper fix for the timeouts. For time being, you could probably increase `kImMessageTimeoutMsec` to 12 seconds. If you plan to go that route, please mention the issue 7489 in the code comment as a TODO.

doru91, 2021-06-09:
Perfect, @pan-apple . Submitted https://github.com/project-chip/connectedhomeip/pull/7490

woody-apple, 2022-01-25:
Is this resolved?

closed 2022-01-25 by woody-apple