project-chip/connectedhomeip #7340: BLE connection gets timed-out while running "connect -ble 3840 20202021 12344321" command on controller Raspi with Accessory Raspi.

ramakrishnanandyala, 2021-06-02:
Hi

I'm new to this connected home Ip Project and I'm starting with testing TE2. While running Ble connect command on controller raspi ble connection gets timed out. 

My ubuntu version is 20.04

Please find the logs below.

========== CONTROLLER LOGS ========
ubuntu@ubuntu:~/connectedhomeip/src/controller/python$ chip-device-ctrl 
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 0x000000000001b669
CHIP:ZCL: Using ZAP configuration...
CHIP:ZCL: deactivate report event
CHIP:DL: CHIP task running
CHIP:DL: Platform main loop started.
Chip Device Controller Shell
 
chip-device-ctrl > connect -ble 3840 20202021 12344321
Device is assigned with nodeid = 12344321
CHIP:BLE: Assigned local session key ID 0
CHIP:DL: TRACE: Bluez mainloop starting Thread
CHIP:EM: SessionEstablishmentExchangeDispatch::SendMessageImpl  mTransportMgr 0x16a2c870
CHIP:IN: Message appended to BLE send queue
CHIP:BLE: Sent PBKDF param request
CHIP:DL: TRACE: Bus acquired for name C-2aef
CHIP:DL: PlatformBlueZInit init success
CHIP:BLE: BLE removing known devices.
CHIP:BLE: BLE initiating scan.
CHIP:BLE: Device 6E:1E:52:15:9E:96 does not look like a CHIP device.
CHIP:BLE: Device D0:B5:5C:9A:11:8A does not look like a CHIP device.
CHIP:BLE: Device D8:F8:83:7A:87:B6 does not look like a CHIP device.
CHIP:BLE: Device 41:BF:75:F9:2C:BC does not look like a CHIP device.
CHIP:BLE: Device F0:EF:86:D4:01:12 does not look like a CHIP device.
CHIP:BLE: Device 65:D9:E7:B4:6A:24 does not look like a CHIP device.
CHIP:BLE: Device F9:26:E7:8D:68:8E does not look like a CHIP device.
CHIP:BLE: Device 43:E0:D4:C3:F1:4E does not look like a CHIP device.
CHIP:BLE: Device F4:BC:DA:3A:5D:F7 does not look like a CHIP device.
CHIP:BLE: Device 34:F6:4B:73:04:7D does not look like a CHIP device.
CHIP:BLE: Device 90:CC:DF:AC:D9:B9 does not look like a CHIP device.
CHIP:BLE: Device F6:63:15:F5:BF:A1 does not look like a CHIP device.
CHIP:BLE: Device 79:13:26:2D:2C:81 does not look like a CHIP device.
CHIP:BLE: Device F0:EF:86:EA:D4:BC does not look like a CHIP device.
CHIP:BLE: Device 5A:D4:C7:17:45:CF does not look like a CHIP device.
CHIP:BLE: Device 24:0A:C4:EA:51:26 does not look like a CHIP device.
CHIP:BLE: Device 79:EE:D1:FF:A6:77 does not look like a CHIP device.
CHIP:BLE: Device C3:E6:38:14:96:B4 does not look like a CHIP device.
CHIP:BLE: Device D3:A8:C0:61:3B:0E does not look like a CHIP device.
CHIP:BLE: Device C1:28:B2:81:B9:67 does not look like a CHIP device.
CHIP:BLE: Device CD:9A:4F:9C:11:B2 does not look like a CHIP device.
CHIP:BLE: Device 5E:E0:5D:A5:9C:D9 does not look like a CHIP device.
CHIP:BLE: Device 64:DC:F9:EC:AF:FD does not look like a CHIP device.
CHIP:BLE: Device F3:0F:9B:92:43:30 does not look like a CHIP device.
CHIP:BLE: Device FE:41:0D:BB:04:B7 does not look like a CHIP device.
CHIP:BLE: Device 6C:AE:01:CF:4B:08 does not look like a CHIP device.
CHIP:BLE: Device F8:64:09:DE:3A:A6 does not look like a CHIP device.
CHIP:BLE: Device C8:B2:9B:95:E0:E2 does not look like a CHIP device.
CHIP:BLE: Device 43:A8:52:42:A3:B2 does not look like a CHIP device.
CHIP:BLE: Device 4E:A0:13:99:13:C0 does not look like a CHIP device.
CHIP:BLE: Device FF:88:20:D2:C8:DB does not look like a CHIP device.
CHIP:BLE: Device B0:E4:D5:DF:11:DA does not look like a CHIP device.
CHIP:BLE: Device 76:B4:15:2F:8B:B4 does not look like a CHIP device.
CHIP:BLE: Device E5:BD:1B:A7:72:6C does not look like a CHIP device.
CHIP:BLE: Device B0:E4:D5:DC:41:6E does not look like a CHIP device.
CHIP:BLE: Device F7:77:E8:73:78:F1 does not look like a CHIP device.
CHIP:BLE: Device 42:F9:1B:44:3B:67 does not look like a CHIP device.
CHIP:BLE: Device 1C:F2:9A:E4:F5:C1 does not look like a CHIP device.
CHIP:BLE: Device A4:C1:38:3B:D0:81 does not look like a CHIP device.
CHIP:BLE: Device D4:7F:CA:43:DA:1A does not look like a CHIP device.
CHIP:BLE: Device EB:90:73:FE:2F:B8 does not look like a CHIP device.
[... 270 lines cut ...]
CHIP:DL: wpa_supplicant: connected to wpa_supplicant proxy
CHIP:DL: wpa_supplicant: can't find interface wlan0: GDBus.Error:fi.w1.wpa_supplicant1.InterfaceUnknown: wpa_supplicant knows nothing about this interface.
CHIP:DL: wpa_supplicant: try to create interface wlan0
CHIP:DL: wpa_supplicant: WiFi interface: /fi/w1/wpa_supplicant1/Interfaces/0
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent 16392
CHIP:DL: Create adv object at /chipoble/07f4/advertising
CHIP:DL: SET service data to {'0xFFF6': <[byte 0x00, 0x00, 0x0f, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent 16393
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: wpa_supplicant: connected to wpa_supplicant interface proxy
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent 16394
CHIP:DL: New BLE connection 0xffff780476c0, device DC:A6:32:BC:1B:05, path /org/bluez/hci0/dev_DC_A6_32_BC_1B_05
CHIP:DL: BluezCharacteristicAcquireWrite is called, conn: 0xffff780476c0
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 517
CHIP:DL: FAIL: short read in BluezCharacteristicWriteFD (0)
CHIP:DL: CHIPoBLE subscribe received
CHIP:BLE: no endpoint for sub recvd
CHIP:DL: HandlePlatformSpecificBLEEvent 32777
CHIP:DL: Receive kCHIPoBLEConnectionEstablished
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
 
(process:2036): GLib-CRITICAL **: 18:41:04.594: Source ID 52 was not found when attempting to remove it

ramakrishnanandyala, 2021-06-02:
@erjiaqing, @Damian-Nordic 

Can any of you help me with this

yunhanw-google, 2021-06-03:
Please use Ubuntu 20.10 (Raspberry Pi 3/4) 64-bit for tesing, in 20.04, the bluez is too old

ramakrishnanandyala, 2021-06-03:
I don't see ubuntu 20.10 to install, I see only 20.04 works for Raspi-4
https://ubuntu.com/download/raspberry-pi

erjiaqing, 2021-06-03:
20.10 is reaching EOL, however, you can still find it on http://cdimage.ubuntu.com/releases/20.10/release/

closed 2022-01-25 by woody-apple