project-chip/connectedhomeip #7357: Fabric label should probably not be stored as "null-terminated uint8_t*"

bzbarsky-apple, 2021-06-03:
#### Problem
PR #6670 has a fabric label that is being stored as a `uint8_t*` with null-termination to indicate the length.

 #### Proposed Solution
We should consider storing the length explicitly; either by storing it as a Pascal-style (ZCL-style) string with length as prefix or by storing the length separately.  Especially because I don't see any checks for there not being a null in the middle of the value anywhere.

In general, we need to figure out what our at-rest representation of CHAR_STRING should be.

@mrjerryjohns @vivien-apple @shana-apple

mrjerryjohns, 2021-06-03:
Is the concern that byte value 0x00 is a permissible, legal value in a UTF-8 representation?

We could store it as a Span, but I'm concerned that we have very poor support for printing out UTF-8 spans in our printing libraries.

In typical C++, this would have been solved using `cout` and std::string. I'm not sure what our strategy with CHIP would be...

bzbarsky-apple, 2021-06-03:
If we think it's a UTF-8 string, we should probably store it as char*...  because printing out null-terminated uint8_t* is also a pain in practice: have to have casts all over.  And even finding the length needs a cast right now.

Another option is to just add a CharString class, have it expose a sane API, and then we can implement however on the inside.  Would need to think about the "how to print it" bit in that case... Maybe it could expose some sort of "get a null-terminated char* that the CharString class will manage lifetime for" getter and then it's an internal problem how we manage that lifetime.

woody-apple, 2022-01-25:
@bzbarsky-apple still an issue?

closed 2022-01-25 by woody-apple

bzbarsky-apple, 2022-01-26:
This is fixed now.