project-chip/connectedhomeip #7387: CHIP is crashing during certification test runs

pan-apple, 2021-06-04:
#### Problem
A new set of tests are being written for CHIP certification. The test currently performs pairing between two devices and sends commands. The devices are running as two processes in Ubuntu/Darwin OS environment.

The following are the stack traces of the crashes.

```* thread #2, queue = 'com.zigbee.chip.framework.controller.workqueue', stop reason = breakpoint 2.1
  * frame #0: 0x0000000100160000 chip-tool`chip::Transport::AdminPairingInfo::StoreIntoKVS(this=0x0000000108162e78, kvs=0x00007ffeefbff158) at AdminPairingTable.cpp:93:11
    frame #1: 0x0000000100160fe3 chip-tool`chip::Transport::AdminPairingTable::Store(this=0x0000000108162e78, id=0) at AdminPairingTable.cpp:391:18
    frame #2: 0x0000000100164861 chip-tool`chip::SecureSessionMgr::SecureMessageDispatch(this=0x0000000102809200, packetHeader=0x00007000027bf878, peerAddress=0x00007000027bf9c0, msg=0x00007000027bfcc0) at SecureSessionMgr.cpp:461:18
    frame #3: 0x0000000100163fe1 chip-tool`chip::SecureSessionMgr::OnMessageReceived(this=0x0000000102809200, peerAddress=0x00007000027bf9c0, msg=0x00007000027bfcc0) at SecureSessionMgr.cpp:301:9
    frame #4: 0x0000000100166a17 chip-tool`chip::TransportMgrBase::HandleMessageReceived(this=0x0000000100c04290, peerAddress=0x00007000027bf9c0, msg=0x00007000027bfcc0) at TransportMgrBase.cpp:57:28
    frame #5: 0x00000001001758cd chip-tool`chip::Transport::Base::HandleMessageReceived(this=0x0000000100c042b8, source=0x00007000027bf9c0, buffer=0x00007000027bfcc0) at Base.h:92:20
    frame #6: 0x00000001001756cd chip-tool`chip::Transport::UDP::OnUdpReceive(endPoint=0x00000001001aae18, buffer=0x00007000027bfcc0, pktInfo=0x00007000027bfba0) at UDP.cpp:110:10
    frame #7: 0x000000010012a309 chip-tool`invocation function for block in chip::Inet::IPEndPointBasis::HandlePendingIO(.block_descriptor=0x00007000027bfb70) at IPEndPointBasis.cpp:1202:17
    frame #8: 0x00007fff203087e6 libdispatch.dylib`_dispatch_client_callout + 8
    frame #9: 0x00007fff203155ae libdispatch.dylib`_dispatch_lane_barrier_sync_invoke_and_complete + 60
    frame #10: 0x000000010012a083 chip-tool`chip::Inet::IPEndPointBasis::HandlePendingIO(this=0x00000001001aae18, aPort=11098) at IPEndPointBasis.cpp:1201:13
    frame #11: 0x000000010012d994 chip-tool`chip::Inet::UDPEndPoint::HandlePendingIO(this=0x00000001001aae18) at UDPEndPoint.cpp:910:26
    frame #12: 0x000000010012b4fa chip-tool`chip::Inet::InetLayer::HandleSelectResult(this=0x00000001001abbc8, selectRes=7, readfds=0x00000001001abc98, writefds=0x00000001001abd18, exceptfds=0x00000001001abd98) at InetLayer.cpp:1252:28
    frame #13: 0x0000000100149f17 chip-tool`chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::SysProcess(this=0x00000001001abc88) at GenericPlatformManagerImpl_POSIX.cpp:210:19
    frame #14: 0x0000000100149b16 chip-tool`chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop(this=0x00000001001abc88) at GenericPlatformManagerImpl_POSIX.cpp:228:9
    frame #15: 0x000000010014a3ce chip-tool`chip::DeviceLayer::PlatformManager::RunEventLoop(this=0x00000001001abc80) at PlatformManager.h:233:37
    frame #16: 0x000000010014a14b chip-tool`chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::EventLoopTaskMain(arg=0x00000001001abc88) at GenericPlatformManagerImpl_POSIX.cpp:239:78
    frame #17: 0x00007fff204b2954 libsystem_pthread.dylib`_pthread_start + 224
    frame #18: 0x00007fff204ae4a7 libsystem_pthread.dylib`thread_start + 15
```

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
#1  0x0000fffff756150c in __GI_abort () at abort.c:79
#2  0x0000fffff77587ec in __gnu_cxx::__verbose_terminate_handler() () at /lib/aarch64-linux-gnu/libstdc++.so.6
#3  0x0000fffff77551dc in  () at /lib/aarch64-linux-gnu/libstdc++.so.6
#4  0x0000fffff7755240 in  () at /lib/aarch64-linux-gnu/libstdc++.so.6
#5  0x0000fffff7756250 in __cxa_deleted_virtual () at /lib/aarch64-linux-gnu/libstdc++.so.6
#6  0x0000aaaaaabd5020 in chip::Messaging::ExchangeMessageDispatch::OnMessageReceived(chip::PayloadHeader const&, unsigned int, chip::Transport::PeerAddress const&, chip::Messaging::ReliableMessageContext*)
    (this=0xaaaaaade2c40, payloadHeader=..., messageId=0, peerAddress=..., reliableMessageContext=0xaaaaae14c278) at ../../../examples/chip-tool/third_party/connectedhomeip/src/messaging/ExchangeMessageDispatch.cpp:107
#7  0x0000aaaaaabd4460 in chip::Messaging::ExchangeContext::HandleMessage(chip::PacketHeader const&, chip::PayloadHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&)
    (this=0xaaaaae14c278, packetHeader=..., payloadHeader=..., peerAddress=..., msgBuf=...) at ../../../examples/chip-tool/third_party/connectedhomeip/src/messaging/ExchangeContext.cpp:382
#8  0x0000aaaaaabd65a0 in operator()<chip::Messaging::ExchangeContext>(chip::Messaging::ExchangeContext*) const (__closure=0xfffff4ec8080, ec=0xaaaaae14c278) at ../../../examples/chip-tool/third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:222
#9  0x0000aaaaaabd66dc in chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8>::ForEachActiveObject<chip::Messaging::ExchangeManager::OnMessageReceived(const chip::PacketHeader&, const chip::PayloadHeader&, chip::SecureSessionHandle, const chip::Transport::PeerAddress&, chip::System::PacketBufferHandle&&, chip::SecureSessionMgr*)::<lambda(auto:3*)> >(struct {...}) (this=0xaaaaae14c248, f=...) at ../../../examples/chip-tool/third_party/connectedhomeip/src/lib/support/Pool.h:140
#10 0x0000aaaaaabd5db8 in chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SecureSessionHandle, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&, chip::SecureSessionMgr*)
    (this=0xaaaaae14c110, packetHeader=..., payloadHeader=..., session=..., source=..., msgBuf=..., msgLayer=0xaaaaae14b680) at ../../../examples/chip-tool/third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:211
#11 0x0000aaaaaabfc7f8 in chip::SecureSessionMgr::MessageDispatch(chip::PacketHeader const&, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xaaaaae14b680, packetHeader=..., peerAddress=..., msg=...)
    at ../../../examples/chip-tool/third_party/connectedhomeip/src/transport/SecureSessionMgr.cpp:316
#12 0x0000aaaaaabfc714 in chip::SecureSessionMgr::OnMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xaaaaae14b680, peerAddress=..., msg=...)
    at ../../../examples/chip-tool/third_party/connectedhomeip/src/transport/SecureSessionMgr.cpp:305
#13 0x0000aaaaaabfdd44 in chip::TransportMgrBase::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xaaaaae146c00, peerAddress=..., msg=...)
    at ../../../examples/chip-tool/third_party/connectedhomeip/src/transport/TransportMgrBase.cpp:57
#14 0x0000aaaaaac20c74 in chip::Transport::Base::HandleMessageReceived(chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&) (this=0xaaaaae146c90, source=..., buffer=...)
    at ../../../examples/chip-tool/third_party/connectedhomeip/src/transport/raw/Base.h:92
#15 0x0000aaaaaac21134 in chip::Transport::UDP::OnUdpReceive(chip::Inet::IPEndPointBasis*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*) (endPoint=0xaaaaaac82798 <chip::Inet::UDPEndPoint::sPool>, buffer=..., pktInfo=0xfffff4ec8490)
```

https://github.com/project-chip/connectedhomeip/pull/7381 CI observed these crashes. 

 #### Proposed Solution
Analyze and fix the problems.

pan-apple, 2021-06-04:
With the latest update to the test PR, it's crashing with the following stack trace
```
    frame #3: 0x0000000100140047 chip-tool`chip::Messaging::ReliableMessageMgr::AddToRetransTable(this=0x0000000101810e28, rc=0x0000000101810f68, rEntry=0x000070000e71f138) at ReliableMessageMgr.cpp:263:5
    frame #4: 0x000000010013c8f0 chip-tool`chip::Messaging::ExchangeMessageDispatch::SendMessage(this=0x0000000108124cb8, session=(mPeerNodeId = 18446744073709551615, mPeerKeyId = 0, mAdmin = 65535), exchangeId=44249, isInitiator=true, reliableMessageContext=0x0000000101810f68, isReliableTransmission=true, protocol=(mVendorId = Common, mProtocolId = 0), type='/', message=0x000070000e71f328) at ExchangeMessageDispatch.cpp:78:9
    frame #5: 0x000000010013b777 chip-tool`chip::Messaging::ExchangeContext::SendMessage(this=0x0000000101810f68, protocolId=(mVendorId = Common, mProtocolId = 0), msgType='/', msgBuf=0x000070000e71f328, sendFlags=0x000070000e71f308) at ExchangeContext.cpp:124:22
    frame #6: 0x000000010017db38 chip-tool`int chip::Messaging::ExchangeContext::SendMessage<chip::Protocols::SecureChannel::MsgType, void>(this=0x0000000101810f68, msgType=PASE_Spake2pError, msgPayload=0x000070000e71f328, sendFlags=0x000070000e71f308) at ExchangeContext.h:108:16
    frame #7: 0x000000010017e1c4 chip-tool`chip::PASESession::SendErrorMsg(this=0x00000001081244a0, errorCode=kInvalidKeyConfirmation) at PASESession.cpp:719:5
    frame #8: 0x000000010017f3d1 chip-tool`chip::PASESession::HandleMsg2_and_SendMsg3(this=0x00000001081244a0, msg=0x000070000e71fcc0) at PASESession.cpp:658:9
    frame #9: 0x000000010017fa6c chip-tool`chip::PASESession::OnMessageReceived(this=0x00000001081244a0, exchange=0x0000000101810f68, packetHeader=0x000070000e71f878, payloadHeader=0x000070000e71f808, msg=0x000070000e71fcc0) at PASESession.cpp:808:15
    frame #10: 0x000000010013c47c chip-tool`chip::Messaging::ExchangeContext::HandleMessage(this=0x0000000101810f68, packetHeader=0x000070000e71f878, payloadHeader=0x000070000e71f808, peerAddress=0x000070000e71f9c0, msgBuf=0x000070000e71fcc0) at ExchangeContext.cpp:400:20
    frame #11: 0x000000010013eb9d chip-tool`auto chip::Messaging::ExchangeManager::OnMessageReceived(this=0x000070000e71f630, ec=0x0000000101810f68)::$_1::operator()<chip::Messaging::ExchangeContext>(chip::Messaging::ExchangeContext*) const at ExchangeMgr.cpp:222:17
    frame #12: 0x000000010013dd5c chip-tool`bool chip::BitMapObjectPool<chip::Messaging::ExchangeContext, 8ul>::ForEachActiveObject<chip::Messaging::ExchangeManager::OnMessageReceived(this=0x0000000101810f38, f=(anonymous class) @ 0x000070000e71f630)::$_1>(chip::Messaging::ExchangeManager::OnMessageReceived(chip::PacketHeader const&, chip::PayloadHeader const&, chip::SecureSessionHandle, chip::Transport::PeerAddress const&, chip::System::PacketBufferHandle&&, chip::SecureSessionMgr*)::$_1) at Pool.h:140:26
    frame #13: 0x000000010013d847 chip-tool`chip::Messaging::ExchangeManager::OnMessageReceived(this=0x0000000101810e00, packetHeader=0x000070000e71f878, payloadHeader=0x000070000e71f808, session=(mPeerNodeId = 18446744073709551615, mPeerKeyId = 0, mAdmin = 65535), source=0x000070000e71f9c0, msgBuf=0x000070000e71fcc0, msgLayer=0x0000000101816c00) at ExchangeMgr.cpp:211:18
    frame #14: 0x0000000100164ab8 chip-tool`chip::SecureSessionMgr::MessageDispatch(this=0x0000000101816c00, packetHeader=0x000070000e71f878, peerAddress=0x000070000e71f9c0, msg=0x000070000e71fcc0) at SecureSessionMgr.cpp:316:14
    frame #15: 0x000000010016400b chip-tool`chip::SecureSessionMgr::OnMessageReceived(this=0x0000000101816c00, peerAddress=0x000070000e71f9c0, msg=0x000070000e71fcc0) at SecureSessionMgr.cpp:305:9
    frame #16: 0x0000000100166a17 chip-tool`chip::TransportMgrBase::HandleMessageReceived(this=0x0000000100a228e0, peerAddress=0x000070000e71f9c0, msg=0x000070000e71fcc0) at TransportMgrBase.cpp:57:28
    frame #17: 0x00000001001758cd chip-tool`chip::Transport::Base::HandleMessageReceived(this=0x0000000100a22928, source=0x000070000e71f9c0, buffer=0x000070000e71fcc0) at Base.h:92:20
```

bzbarsky-apple, 2021-06-09:
Trivial to reproduce as follows:
1. Edit `scripts/tests/test_suites.sh` and replace `20202021` with `20202022` in the "pairing" command.
2. Run:
    ```
    % scripts/examples/gn_build_example.sh examples/all-clusters-app/linux out/debug chip_config_network_layer_ble=false
    % scripts/examples/gn_build_example.sh examples/chip-tool out/debug/standalone/
    % scripts/tests/test_suites.sh 1 
    ```

This will crash every time.

bzbarsky-apple, 2021-06-09:
Looks like the issue is that in this case we land in `PASESession::HandleMsg2_and_SendMsg3` and then do the following, in order:

1. Send msg3
2. Verify the keys and discover that verification fails.
3. Send an error message.

And it's this last that hits the VerifyOrDie.

pan-apple, 2021-06-09:
https://github.com/project-chip/connectedhomeip/pull/7502

andy31415, 2021-06-17:
Assuming closed on #7502 comment

closed 2021-06-17 by andy31415