project-chip/connectedhomeip #7390: chip-device-ctrl connected error

ChengqiangShe, 2021-06-04:
#### When I use chip-device-ctrl to configure a device, an error is displayed

chip-device-ctrl and device commit：test_event_3_2021_06_03
 #### bb8a6ef6ded5718caf855b1067300d19aba0a524 

@jmartinez-silabs 
Can you help me find out what caused it, how do I need to solve this problem here?

chip-device-ctrl logs：
```
(python_env) chengqiang@chengqiang-desktop:~/Desktop/connectedhomeip$ chip-device-ctrl 
[1622792402.317171][8360] CHIP:IN: local node id is 0x000000000001B669
[1622792402.318456][8360] CHIP:DL: MDNS failed to join multicast group on eth0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622792402.319958][8360] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622792402.320735][8360] CHIP:ZCL: Using ZAP configuration...
[1622792402.320837][8360] CHIP:ZCL: deactivate report event
[1622792402.320885][8360] CHIP:CTL: Getting operational keys
[1622792402.322067][8360] CHIP:CTL: Generating operational certificate for the controller
[1622792402.323233][8360] CHIP:CTL: Getting root certificate for the controller from the issuer
[1622792402.323720][8360] CHIP:CTL: Generating credentials
[1622792402.324034][8360] CHIP:CTL: Loaded credentials successfully
[1622792402.337505][8369] CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > 
chip-device-ctrl > connect -ble 3840 73141520 1232
Device is assigned with nodeid = 1232
[1622792446.882316][8364] CHIP:BLE: BLE removing known devices.
[1622792446.882395][8364] CHIP:BLE: BLE initiating scan.
[1622792447.368815][8364] CHIP:BLE: New device scanned: 4E:A7:F9:B4:3C:7A
[1622792447.369012][8364] CHIP:BLE: Device discriminator match. Attempting to connect.
[1622792447.373847][8364] CHIP:BLE: Scan complete notification without an active scan.
[1622792448.460473][8364] CHIP:DL: FAIL: ConnectDevice : GDBus.Error:org.bluez.Error.Failed: Software caused connection abort
Failed to establish secure session to device: 4050
CHIP Error 4050 (0x00000FD2): Timeout
[1622792476.864653][8368] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
Failed to establish secure session to device: 4050
chip-device-ctrl > 
chip-device-ctrl > connect -ble 3840 73141520 1232
Device is assigned with nodeid = 1232
[1622792520.528054][8364] CHIP:BLE: BLE removing known devices.
[1622792520.528139][8364] CHIP:BLE: BLE initiating scan.
[1622792521.280073][8364] CHIP:BLE: New device scanned: 4E:A7:F9:B4:3C:7A
[1622792521.280139][8364] CHIP:BLE: Device discriminator match. Attempting to connect.
[1622792521.283768][8364] CHIP:BLE: Scan complete notification without an active scan.
[1622792523.696654][8369] CHIP:BLE: subscribe complete, ep = 0xffffbbb24f40
[1622792523.696742][8369] CHIP:BLE: peripheral chose BTP version 3; central expected between 2 and 3
[1622792523.696766][8369] CHIP:BLE: using BTP fragment sizes rx 20 / tx 128.
[1622792523.696782][8369] CHIP:BLE: local and remote recv window size = 3
[1622792525.646112][8368] CHIP:EM: Received message of type 33 and protocolId 0 on exchange 39018
[1622792531.790033][8369] CHIP:EM: Received message of type 35 and protocolId 0 on exchange 39018
[1622792531.795485][8369] CHIP:IN: Sending msg from 0x000000000001B669 to 0x00000000000004D0 at utc time: 1066608 msec
[1622792531.795556][8369] CHIP:IN: Sending secure msg on generic transport
[1622792531.795602][8369] CHIP:IN: Secure msg send status No Error
[1622792537.006475][8368] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 1
[1622792537.006740][8368] CHIP:IN: Setting fabricID 4D0 on admin.
[1622792537.006804][8368] CHIP:IN: Since admin was modified, persisting changes to KVS
[1622792537.006951][8368] CHIP:EM: Received message of type 9 and protocolId 327680 on exchange 39019
[1622792537.007316][8368] CHIP:ZCL: OpCSRResponse:
[1622792537.007435][8368] CHIP:ZCL:   CSR: 204
[1622792537.007473][8368] CHIP:ZCL:   CSRNonce: 32
[1622792537.007506][8368] CHIP:ZCL:   VendorReserved1: 0
[1622792537.007651][8368] CHIP:ZCL:   VendorReserved2: 0
[1622792537.007755][8368] CHIP:ZCL:   VendorReserved3: 0
[1622792537.007854][8368] CHIP:ZCL:   Signature: 0
[1622792537.007960][8368] CHIP:CTL: Received certificate signing request from the device
[1622792537.008100][8368] CHIP:CTL: Generating operational certificate for device 4d0
[1622792537.010550][8368] CHIP:CTL: Getting intermediate CA certificate from the issuer
[1622792537.010625][8368] CHIP:CTL: GetIntermediateCACertificate returned 4190
[1622792537.010662][8368] CHIP:CTL: Intermediate CA is not needed
[... 395 lines cut ...]
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
00> <info  > [DL] Tx Confirmation received
00> <info  > [DL]  stop soft timer
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEIndicateConfirm
00> <info  > [DL] Char Write Req, char : 26
00> <detail> [DL] Write request/command received for CHIPoBLE RX characteristic (con 1, len 3)
00> <info  > [DL] _OnPlatformEvent kCHIPoBLEWriteReceived
(Connection lost)
```

jmartinez-silabs, 2021-06-04:
Hi @ChengqiangShe 

The secure session failed so you cannot send the zcl commands for the provissioning. Now I am not sure why it failed.

I see you are using mg21, I think it is a Heap issue, we had that happend on the lock app. RAM on MG21 is getting really slim and we have almost no room. You can try to bump the heap by 1k. I will try it and do a PR for that

Like this PR https://github.com/project-chip/connectedhomeip/pull/7241

I also what to point out that you are using the wrong endpoints for the commissioning zcl commands.

Here is the flow of commands you should be using

```
connect -ble 3840 73141520 1234

zcl NetworkCommissioning AddThreadNetwork 1234 **0** 0 operationalDataset=hex:0e080000000000000000000300000f35060004001fffe0020811111111222222220708fdc68a163468d333051000112233445566778899aabbccddeeff030e4f70656e54687265616444656d6f01021234041061e1206d2c2b46e079eb775f41fc72190c0402a0fff8 breadcrumb=0 timeoutMs=1000

zcl NetworkCommissioning EnableNetwork 1234 **0** 0 networkID=hex:1111111122222222 breadcrumb=0 timeoutMs=1000

close-ble

resolve 0 1234

zcl OnOff Toggle 1234 **1** 0
```
As you see except the OnOff command that use enpoint 1, the others are on endpoint 0

568760310, 2021-06-15:
@jmartinez-silabs 

I meet  the same issue on MG12 while pairing lock-app. Is the same reason  failed ? and how to fix this isuue?

568760310, 2021-06-15:
I also test the lock-app on nrf52840, the same issue happened.
```
ubuntu@ubuntuservercontroller:~$ chip-device-ctrl 
[1623756311.603052][11055] CHIP:CTL: SyncSetKeyValue on ExampleOpCredsCAKey
[1623756311.608230][11055] CHIP:IN: TransportMgr initialized
[1623756311.608277][11055] CHIP:DIS: Init admin pairing table with server storage
[1623756311.608327][11055] CHIP:IN: local node id is 0x000000000001B669
[1623756311.612231][11055] CHIP:DL: Avahi client registered
[1623756311.613888][11055] CHIP:ZCL: Using ZAP configuration...
[1623756311.613994][11055] CHIP:ZCL: deactivate report event
[1623756311.614037][11055] CHIP:CTL: Getting operational keys
[1623756311.615050][11055] CHIP:CTL: Generating operational certificate for the controller
[1623756311.616059][11055] CHIP:CTL: Getting root certificate for the controller from the issuer
[1623756311.616458][11055] CHIP:CTL: SyncSetKeyValue on CHIPAdmin0
[1623756311.616531][11055] CHIP:CTL: Generating credentials
[1623756311.616809][11055] CHIP:CTL: Loaded credentials successfully
[1623756311.617077][11063] CHIP:DL: CHIP task running
[1623756311.631259][11064] CHIP:DL: Platform main loop started.
Chip Device Controller Shell

chip-device-ctrl > connect -ble 3840 20202021  1
Device is assigned with nodeid = 1
[1623756364.280122][11055] CHIP:SC: Assigned local session key ID 0
[1623756364.280373][11055] CHIP:IN: Message appended to BLE send queue
[1623756364.280453][11055] CHIP:SC: Sent PBKDF param request
[1623756364.280395][11583] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1623756364.280672][11055] CHIP:CTL: SyncSetKeyValue on StartKeyID
[1623756364.280913][11059] CHIP:DL: TRACE: Bus acquired for name C-2b2f
[1623756364.292317][11064] CHIP:DL: PlatformBlueZInit init success
[1623756364.298864][11059] CHIP:BLE: BLE removing known devices.
[1623756364.298937][11059] CHIP:BLE: BLE initiating scan.
[1623756364.408275][11059] CHIP:BLE: Device 54:EF:44:26:58:0E does not look like a CHIP device.
[1623756364.429334][11059] CHIP:BLE: Device 7F:96:72:77:9C:17 does not look like a CHIP device.
[1623756364.431541][11059] CHIP:BLE: Device 65:EC:84:E4:D9:1B does not look like a CHIP device.
[1623756364.440751][11059] CHIP:BLE: Device 72:26:66:33:16:49 does not look like a CHIP device.
[1623756364.455182][11059] CHIP:BLE: Device 28:D1:27:29:8B:6B does not look like a CHIP device.
[1623756364.469305][11059] CHIP:BLE: Device 5C:E6:52:56:A1:8C does not look like a CHIP device.
[1623756364.492072][11059] CHIP:BLE: Device 28:D1:27:29:67:B9 does not look like a CHIP device.
[1623756364.494284][11059] CHIP:BLE: Device 5D:FD:FF:63:EE:A5 does not look like a CHIP device.
[1623756364.495993][11059] CHIP:BLE: Device 5B:95:BC:7A:46:32 does not look like a CHIP device.
[1623756364.497776][11059] CHIP:BLE: Device 84:2E:14:28:45:61 does not look like a CHIP device.
[1623756364.513851][11059] CHIP:BLE: Device CC:CC:CC:DB:FC:AC does not look like a CHIP device.
[1623756364.523375][11059] CHIP:BLE: Device 4A:72:3D:42:65:78 does not look like a CHIP device.
[1623756364.543388][11059] CHIP:BLE: Device 8C:85:90:C2:19:3B does not look like a CHIP device.
[1623756364.547021][11059] CHIP:BLE: Device 28:D1:27:29:2F:3E does not look like a CHIP device.
[1623756364.552294][11059] CHIP:BLE: Device 84:2E:14:78:C1:D0 does not look like a CHIP device.
[1623756364.555625][11059] CHIP:BLE: Device 28:D1:27:28:F0:4E does not look like a CHIP device.
[1623756364.557459][11059] CHIP:BLE: Device 7C:57:89:F5:A8:58 does not look like a CHIP device.
[1623756364.580314][11059] CHIP:BLE: Device 28:D1:27:29:83:5F does not look like a CHIP device.
[1623756364.583966][11059] CHIP:BLE: Device 74:9C:E7:45:22:1B does not look like a CHIP device.
[1623756364.586976][11059] CHIP:BLE: Device F9:9F:BE:EC:AA:7C does not look like a CHIP device.
[1623756364.588992][11059] CHIP:BLE: Device 74:10:04:9C:C3:1F does not look like a CHIP device.
[1623756364.619268][11059] CHIP:BLE: Device 54:EF:44:CC:D7:AE does not look like a CHIP device.
[1623756364.628333][11059] CHIP:BLE: Device 1F:CD:6D:81:DC:EF does not look like a CHIP device.
[1623756364.636453][11059] CHIP:BLE: New device scanned: F7:D8:C8:B3:46:53
[1623756364.636574][11059] CHIP:BLE: Device discriminator match. Attempting to connect.
[1623756364.654584][11059] CHIP:BLE: Scan complete notification without an active scan.
[1623756364.771029][11059] CHIP:DL: ConnectDevice complete
[1623756366.558490][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[1623756366.558655][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[1623756366.558696][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[1623756366.558724][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[1623756366.558762][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0007
[1623756366.558788][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[1623756366.558817][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0007
[1623756366.558838][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[1623756366.558863][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0007
[1623756366.558887][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[1623756366.558926][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0007
[1623756366.558948][11059] CHIP:DL: Char1 /org/bluez/hci0/dev_F7_D8_C8_B3_46_53/service0001
[... cut ...]
[1623756380.741952][11064] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623756383.369472][11064] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1623756386.117232][11059] CHIP:DL: Indication received, conn = 0xffff90064340
[1623756386.117461][11063] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1623756388.769677][11064] CHIP:DL: HandlePlatformSpecificBLEEvent 16387

```

the version of nrf
```
*** Booting Zephyr OS build v2.4.99-ncs1-4959-g4e63ea417176  ***
```
commit of CHIP:
```
b0eeffc345840f84a25a69b77c996d9e7fe1b915
```

jmartinez-silabs, 2021-06-16:
I am looking into this thank you. I will link to this issue when there is a fix

closed 2021-06-21 by woody-apple