project-chip/connectedhomeip #7394: Adaptive generated enumeration storage width

gjc13, 2021-06-04:
#### Problem

Currently the zap generated enumerations doesn't specify the storage width or must have unified storage width.

 #### Proposed Solution

Modify the ZAP generation engine to generate the storage width based on the enumeration values.

wqx6, 2021-06-15:
Maybe we can use `enum class ENUM_TYPE :uintx_t` when defining the enumeration and use `Put(...,static_cast<std::underlying_type_t<ENUM_TYPE>>)` when calling the enumeration. like https://en.cppreference.com/w/cpp/types/underlying_type. It seems a safe solution, but  it needs a lot of changes at where using the enumeration.

bzbarsky-apple, 2021-06-17:
Using `enum class` is orthogonal to the problem of figuring out what the right underlying type is for an enum....

Ideally that would just come from the spec or some other data source and feed into ZAP.

@tecimovic @vivien-apple

jmeg-sfy, 2021-06-18:
@bzbarsky-apple this is something i would like to explicity use and define in my xml file. casting is not a solution and is prone to runtime problem . what is the policy regarding enum and packed structure 
Also can we enforce preprocessing compiler checks on ZAP

closed 2021-06-18 by mspang

bzbarsky-apple, 2021-06-25:
@jmeg-sfy Sorry for the lag; I hope https://github.com/project-chip/connectedhomeip/pull/7746 does what you wanted...

jmeg-sfy, 2021-07-13:
> @jmeg-sfy Sorry for the lag; I hope #7746 does what you wanted...

yes all is solved now thanks !

PR #7746 (2021-06-18): Generate enumeration storage width
#### Problem
Currently enum are using `uint32_t` for their storage witdth

#### Change overview
 * Use the type specified in the enum definition
 
#### Testing
 * Manually tested by generating + compiling on Mac. Someone with a platform that was not compiling (for example `uint32_t` came from #7266 because of some compilation issues) may want to double check it

#Fixes #7394 ?