project-chip/connectedhomeip #7402: Using build_python.sh on darwin by default produces a chip-device-ctrl that can't do mdns

bzbarsky-apple, 2021-06-04:
#### Problem
1. Run ./scripts/build_python.sh on Mac
2. Try to use the result to BLE-pair with a device and then resolve the ip

Pairing works.  Resolving doesn't work.  When `chip-device-ctrl` starts, it outputs a bunch of:
```
[1622821483892] [0x290dc2e] CHIP: [DL] MDNS failed to join multicast group on en1 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622821483892] [0x290dc2e] CHIP: [DL] MDNS failed to join multicast group on en1 for address type IPv6: OS Error 2022 (0x000007E6): Invalid argument
```
which may be relevant, but may not be; I don't see it logging an error for en0, which is the interface we want to be using in the end.

It looks like #7061 defaulted the python build to using minimal mdns instead of platform, which is not what happens if various other build instructions (the ones used by TEs, for example) are followed, and doesn't seem to work on Darwin for some reason.

@PSONALl @andy31415 

 #### Proposed Solution
Change default for the script to platform mdns, until minimal works out of the box?

dhrishi, 2021-06-04:
> the ones used by TEs

Hi @bzbarsky-apple For TE2 we (and a few other platform folks as well IIRC) faced issues in resolve command and later it was discovered that the platform mDNS implementation needs to be used for that to work fine. So, we assumed that minimal was the default choice.

bzbarsky-apple, 2021-06-04:
The instructions I saw did not use this script in this form at all, and "platform" is the default value, so the thing people did ended up with platform mdns, no?

dhrishi, 2021-06-04:
Yes, I agree that `build_python.sh` was not used to build chip-device-ctrl. Although, see here: https://github.com/project-chip/connectedhomeip/issues/6329 and specifically https://github.com/project-chip/connectedhomeip/issues/6329#issuecomment-828297715
So Linux uses minimal mDNS as default and not default? And Darwin uses platform as the default?

bzbarsky-apple, 2021-06-04:
>  So Linux uses minimal mDNS as default and not platform? And Darwin used platform as the default before #7061?

`src/platform/device.gni` has:
```
  # Select DNS-SD implementation
  if (chip_device_platform == "linux" || chip_device_platform == "esp32") {
    chip_mdns = "minimal"
  } else if (chip_device_platform == "darwin" ||
             chip_device_platform == "cc13x2_26x2") {
    chip_mdns = "platform"
  } else {
    chip_mdns = "none"
  }
```
So yes, that is correct: default is minimal on Linux and platform on darwin.  So presumably the TEs have been using minimal, since they are using Raspberri Pi as the client, ok.

dhrishi, 2021-06-04:
Ohh okay. So need to change the script in such a way that it populates and sends chip_mdns only when it is passed through command line. Else, the default values for each platform would be appropriately chosen. Right? cc @PSONALl

bzbarsky-apple, 2021-06-04:
@dhrishi That would make the most sense to me.

PR #7420 (2021-06-07): Add support of chip_mdns without default value
#### Problem
* Currently python build is defaulted to use minimal mdns 
* It doesn't hold for other examples which use platform mdns as default
* Fixes #7402 

#### Change overview
* No any default value for chip_mdns variable is given
* Populates and sends chip_mdns only when it is passed through command line
* Else, the default values for each platform would be appropriately chosen

#### Testing
* Argument values populated in out/python_lib/args.gn file are verified.
closed 2021-06-07 by bzbarsky-apple