project-chip/connectedhomeip #7406: Minimal mdns fails on a Mac laptop

bzbarsky-apple, 2021-06-04:
#### Problem
Minimal mdns has the following general structure:

1. Loop over the network interfaces available, up to 30 of them.
2. For each one do `NewUDPEndPoint`, bind it, listen on it.
3. Try to `JoinMulticastGroup` on the endpoint.
4. When resolving, loop over all endpoints and try to broadcast on each one.  If the broadcast attempt fails with `ENETUNREACH`, log and keep going.  If it fails with anything else, bail out.

On my Mac laptop, I definitely have at least 30 network interfaces.  Luckily the one that actually matters here is in the first 30....  Anyway, when `ServerBase::Listen` runs I get the following log (after modifying the logging to log success, not just failure, and logging the endpoint indices):
<details>
<summary>Log for listening</summary> 

```
[1622842447962] [0x295cbf6] CHIP: [DL] MDNS failed 0 to join multicast group on en5 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447962] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 1 en5 IPv6
[1622842447962] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 2 en0 IPv4
[1622842447962] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 3 en0 IPv6
[1622842447963] [0x295cbf6] CHIP: [DL] MDNS failed 4 to join multicast group on en2 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447963] [0x295cbf6] CHIP: [DL] MDNS failed 5 to join multicast group on en2 for address type IPv6: OS Error 2022 (0x000007E6): Invalid argument
[1622842447964] [0x295cbf6] CHIP: [DL] MDNS failed 6 to join multicast group on en1 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447964] [0x295cbf6] CHIP: [DL] MDNS failed 7 to join multicast group on en1 for address type IPv6: OS Error 2022 (0x000007E6): Invalid argument
[1622842447964] [0x295cbf6] CHIP: [DL] MDNS failed 8 to join multicast group on en3 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447964] [0x295cbf6] CHIP: [DL] MDNS failed 9 to join multicast group on en3 for address type IPv6: OS Error 2022 (0x000007E6): Invalid argument
[1622842447965] [0x295cbf6] CHIP: [DL] MDNS failed 10 to join multicast group on en4 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447965] [0x295cbf6] CHIP: [DL] MDNS failed 11 to join multicast group on en4 for address type IPv6: OS Error 2022 (0x000007E6): Invalid argument
[1622842447966] [0x295cbf6] CHIP: [DL] MDNS failed 12 to join multicast group on bridge0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447966] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 13 bridge0 IPv6
[1622842447966] [0x295cbf6] CHIP: [DL] MDNS failed 14 to join multicast group on p2p0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447966] [0x295cbf6] CHIP: [DL] MDNS failed 15 to join multicast group on p2p0 for address type IPv6: OS Error 2022 (0x000007E6): Invalid argument
[1622842447967] [0x295cbf6] CHIP: [DL] MDNS failed 16 to join multicast group on awdl0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447967] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 17 awdl0 IPv6
[1622842447968] [0x295cbf6] CHIP: [DL] MDNS failed 18 to join multicast group on llw0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447968] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 19 llw0 IPv6
[1622842447968] [0x295cbf6] CHIP: [DL] MDNS failed 20 to join multicast group on utun0 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447968] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 21 utun0 IPv6
[1622842447969] [0x295cbf6] CHIP: [DL] MDNS failed 22 to join multicast group on utun1 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447969] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 23 utun1 IPv6
[1622842447970] [0x295cbf6] CHIP: [DL] MDNS failed 24 to join multicast group on utun2 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447970] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 25 utun2 IPv6
[1622842447971] [0x295cbf6] CHIP: [DL] MDNS failed 26 to join multicast group on utun4 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447971] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 27 utun4 IPv6
[1622842447971] [0x295cbf6] CHIP: [DL] MDNS failed 28 to join multicast group on en9 for address type IPv4: Inet Error 1016 (0x000003F8): Address not found
[1622842447971] [0x295cbf6] CHIP: [DL] MDNS SUCCESS 29 en9 IPv6
```

</details>

To summarize:
* Endpoints 0, 4, 6, 8, 10, 12, 14, 16, 18, 20, 22, 24, 26, 28 fail with `INET_ERROR_ADDRESS_NOT_FOUND`
* Endpoints 5, 7, 9, 11, 15 fail with `EINVAL`.
* Endpoints 1, 2, 3, 13, 17, 19, 21, 23, 25, 27, 29 sucessfully return from `JoinMulticastGroup`.

Then when `ServerBase::BroadcastSend` (the version without an interface argument) runs, I get a log like so:

<details>
<summary>Log for sending</summary>

```

[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 0 ipv4
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 0 errored out OS Error 2051 (0x00000803): Network is unreachable
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 1 ipv6
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 1 errored out OS Error 2065 (0x00000811): No route to host
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 2 ipv4
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 3 ipv6
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 4 ipv4
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 4 errored out OS Error 2051 (0x00000803): Network is unreachable
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 5 ipv6
[1622842777234] [0x295df50] CHIP: [DIS] Broadcasting 5 errored out OS Error 2022 (0x000007E6): Invalid argument
[1622842777235] [0x295df50] CHIP: [DIS] Broadcasting 6 ipv4
[... 37 lines cut ...]
[1622842777238] [0x295df50] CHIP: [DIS] Broadcasting 28 ipv4
[1622842777238] [0x295df50] CHIP: [DIS] Broadcasting 28 errored out OS Error 2051 (0x00000803): Network is unreachable
[1622842777238] [0x295df50] CHIP: [DIS] Broadcasting 29 ipv6
[1622842777238] [0x295df50] CHIP: [DIS] Broadcasting 29 errored out OS Error 2049 (0x00000801): Can't assign requested address
```

</details>

To summarize:
* Endpoints 0, 4, 6, 8, 10, 12, 14, 16, 18, 29, 22, 24, 26, 28 fail with `ENETUNREACH` 
* Endpoints 1 fails with `EHOSTUNREACH`
* Endpoints 5, 7, 9, 11, 15, fail with `EINVAL`
* Endpoints 13, 29 fail with `EADDRNOTAVAIL`
* Endpoints 2, 3, 17, 19, 21, 23, 25, 27 succeed in doing the broadcast send.
 
Comparing the two lists:
1. Everything that fails to `JoinMulticastGroup` also fails to broadcast (the things that failed with `ADDRESS_NOT_FOUND` now fail with `ENETUNREACH`, the things that failed with `EINVAL` still fail with `EINVAL`.
2. Of the things that succeeded at `JoinMulticastGroup`, three now fail to broadcst, with `EHOSTUNREACH` on "en5 IPv6" and `EADDRNOTAVAIL` on "bridge0 IPv6" and "en9 IPv6".

 #### Proposed Solution
I think we should do a few things:
1. Endpoints which fail to get through `JoinMulticastGroup` should be probably be `Free()`d and nulled out so we don't have to mess with them when we broadcast.
2. Broadcasting should keep going in the loop if it gets `EHOSTUNREACH` or `EADDRNOTAVAIL`, just like it already does for `ENETUNREACH`.  It's possible that it should just do this no matter what error it gets from a specific endpoint.

@andy31415

andy31415, 2021-06-11:
MinMdns was built for embedded and does not behave well with many interfaces but it does not try to either.

Will have to balance out here how much we want to make this work on linux/mac for testing purposes.

bzbarsky-apple, 2021-06-17:
#7555 helped some.  Now the problem is that at the end of the `ServerBase::Listen` loop we run out of space for endpoints, so return `CHIP_ERROR_NO_MEMORY` and clear out all the things we managed to set up so far...

closed 2021-06-17 by woody-apple