project-chip/connectedhomeip #7413: Lock-app on nRF52840 picks wrong SRP server address

retfie, 2021-06-05:
#### Problem
After provisioning on Thread network, lock-app unable to connect to SRP server:
```
I: 839 [DL]removing srp service: 000000014A77CBB3-00000000000011C2._chip._tcp
I: 846 [DL]OnSrpClientNotification: Last requested operation completed successfully
I: 853 [DL]advertising srp service: 000000014A77CBB3-00000000000011C2._chip._tcp
E: 2685 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 5779 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 11017 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 19891 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 34954 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 60523 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 103958 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 177771 [DL]OnSrpClientNotification: Timed out waiting on server response
```
```
uart:~$ ot srp client service
instance:"000000014A77CBB3-00000000000011C2", name:"_chip._tcp", state:Adding, port:11097, priority:0, weight:0
Done
```
```
uart:~$ ot ipaddr
fdf2:807d:e463:877b:bec6:2a22:40f5:c935
fd0a:3ab:437:11b0:0:ff:fe00:9004
fd0a:3ab:437:11b0:2f7f:3e07:e70d:6199
fe80:0:0:0:2099:f448:db3e:c09a
Done
```
Wrong server address and port
```
uart:~$ ot srp client server
[fd0a:3ab:437:11b0:0:ff:fe00:fc10]:64778
Done
```
After restart with another ipv6 (found with ot-ctl ipaddr), it successfully registers: 
```
uart:~$ ot srp client stop
Done
uart:~$  ot srp client state
Disabled
Done
uart:~$ ot srp client start fd0a:3ab:437:11b0:ee12:ab2b:7464:865 49154
Done
I: 69894 [DL]OnSrpClientNotification: Last requested operation completed successfully

uart:~$ ot srp client service
instance:"000000014A77CBB3-00000000000011C2", name:"_chip._tcp", state:Registered, port:11097, priority:0, weight:0
Done
```

```# OTBR
pi@raspberrypi:~ $ sudo ot-ctl ipaddr

fdf2:807d:e463:877b:3fec:11a7:617e:6da1
fd0a:3ab:437:11b0:0:ff:fe00:fc11
fd0a:3ab:437:11b0:0:ff:fe00:fc38
fd0a:3ab:437:11b0:0:ff:fe00:fc10                # <--------- not working
fd0a:3ab:437:11b0:0:ff:fe00:fc00
fd0a:3ab:437:11b0:0:ff:fe00:9000
fd0a:3ab:437:11b0:ee12:ab2b:7464:865            # <--------- working
fe80:0:0:0:cda:bc96:3ede:10d3
Done
```

 #### Proposed Solution

kkasperczyk-no, 2021-06-07:
Hi @retfie, what version of OTBR did you use? Was it openthread/otbr:latest or something else? I faced some time ago similar problem and it was handled in this issue: https://github.com/project-chip/connectedhomeip/issues/6726. However I verified that on current master branch it works with openthread/otbr:latest and in my case SRP client always selects Mesh local EID address (that one working in your case) instead of ALOC (ended with fc10).

retfie, 2021-06-07:
Hi @kkasperczyk-no , for OTBR I was following the steps from https://openthread.io/codelabs/openthread-border-router#1 (cloned a 2 days ago) and its HEAD was on one or two commits behind today's HEAD at  e6ec05260. Tried to checkout to one before 6f7fd7c21, and indeed beside the server address is still ALOC one (*fc10), it was able to successfully publish all its advertising:

```
I: 867 [DL]SRP Client was started, as detected server addressed: fd0a:3ab:437:11b0:0:ff:fe00:fc10
D: 879 [DL]OpenThread State Changed (Flags: 0x00000200)
D: 884 [DL]OpenThread State Changed (Flags: 0x00000001)
D: 889 [DL]   Thread Unicast Addresses:
D: 893 [DL]        fdf2:807d:e463:877b:bec6:2a22:40f5:c935/64 valid preferred
D: 900 [DL]        fd0a:3ab:437:11b0:0:ff:fe00:9001/64 valid preferred rloc
D: 907 [DL]        fd0a:3ab:437:11b0:2f7f:3e07:e70d:6199/64 valid preferred
D: 913 [DL]        fe80::2099:f448:db3e:c09a/64 valid preferred
E: 929 [DIS]Fabric ID not known. Using a default
I: 933 [DIS]Found admin pairing for admin 0, node 0x00000000000011C2
D: 940 [DIS]Using Thread extended MAC for hostname.
I: 944 [DIS]Advertise operational node 000000014A77CBB3-00000000000011C2
I: 951 [DL]removing srp service: 000000014A77CBB3-00000000000011C2._chip._tcp
I: 958 [DL]OnSrpClientNotification: Last requested operation completed successfully
I: 966 [DL]advertising srp service: 000000014A77CBB3-00000000000011C2._chip._tcp
I: 1240 [DL]OnSrpClientNotification: Last requested operation completed successfully

uart:~$
uart:~$ ot srp client server
[fd0a:3ab:437:11b0:0:ff:fe00:fc10]:49153
Done
uart:~$
uart:~$ ot srp client service
instance:"000000014A77CBB3-00000000000011C2", name:"_chip._tcp", state:Registered, port:11097, priority:0, weight:0
Done
```

while moving on  e6ec05260 or any other commit up-to current HEAD, only first attempt was successful:

```
I: 7552 [DL]SRP Client was started, as detected server addressed: fd0a:3ab:437:11b0:0:ff:fe00:fc10
D: 7564 [DL]OpenThread State Changed (Flags: 0x200002a4)
D: 7569 [DL]   Device Role: CHILD
D: 7572 [DL]   Partition Id: 0x20CE0D05
D: 7576 [DL]OpenThread State Changed (Flags: 0x00000001)
D: 7581 [DL]   Thread Unicast Addresses:
D: 7584 [DL]        fdf2:807d:e463:877b:bec6:2a22:40f5:c935/64 valid preferred
D: 7592 [DL]        fd0a:3ab:437:11b0:0:ff:fe00:9001/64 valid preferred rloc
D: 7598 [DL]        fd0a:3ab:437:11b0:2f7f:3e07:e70d:6199/64 valid preferred
D: 7605 [DL]        fe80::2099:f448:db3e:c09a/64 valid preferred
E: 7622 [DIS]Fabric ID not known. Using a default
I: 7626 [DIS]Found admin pairing for admin 0, node 0x00000000000011C2
D: 7633 [DIS]Using Thread extended MAC for hostname.
I: 7637 [DIS]Advertise operational node 000000014A77CBB3-00000000000011C2
I: 7644 [DL]removing srp service: 000000014A77CBB3-00000000000011C2._chip._tcp
I: 7651 [DL]OnSrpClientNotification: Last requested operation completed successfully
I: 7659 [DL]advertising srp service: 000000014A77CBB3-00000000000011C2._chip._tcp
E: 9490 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 12584 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 17824 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 26702 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 41765 [DL]OnSrpClientNotification: Timed out waiting on server response
E: 67335 [DL]OnSrpClientNotification: Timed out waiting on server response
```

For client I was using lock-app example from Nordic SDK at this commit:
*** Booting Zephyr OS build v2.4.99-ncs1-1829-ga9d3aef65424  ***

Probably it's based on little bit outdated OpenThread code base, so its worth to test it with example from CHIP repo.

kkasperczyk-no, 2021-06-07:
Ah yes, definitely that might be the reason. The examples located in the nRF Connect SDK often might be not up to date with the latest changes present in the CHIP upstream repo. That is because we update CHIP version available in nRF Connect SDK every few weeks. The OpenThread change in SRP that resulted in observed problems was merged few weeks ago, so using latest OTBR with older firmware version could lead to problems.

retfie, 2021-06-07:
Using the latest code for lock-app example, from CHIP repo, solves the problem, and always picks the correct IP address.

closed 2021-06-07 by retfie