project-chip/connectedhomeip #7416: [Cirque] See the 0xd error when running cirque mobile device test

yunhanw-google, 2021-06-06:
#### Problem
DeviceController is presenting 0xd error when testing with lighting example, maybe endpoint is changed?

2021-06-06 07:12:58,169 [PythonMatterControllerTEST] INFO Sending On/Off commands to device 1
[1622963578.169450][36] CHIP:DMG: ICR moving to [Initialize]
[1622963578.169488][36] CHIP:DMG: ICR moving to [AddCommand]
[1622963578.169516][36] CHIP:IN: Secure message was encrypted: Msg ID 15
[1622963578.169536][36] CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000000001 at utc time: 1411216 msec
[1622963578.169540][36] CHIP:IN: Sending secure msg on generic transport
[1622963578.169594][36] CHIP:IN: Secure msg send status No Error
[1622963578.169604][36] CHIP:DMG: ICR moving to [   Sending]
[1622963578.169634][49] CHIP:ZCL: DefaultResponse:
[1622963578.169637][49] CHIP:ZCL:   Transaction: 0x7f9c0a1f59b0
[1622963578.170392][49] CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
[1622963578.170395][49] CHIP:ZCL: emberAfDefaultResponseCallback: Missing success callback
[1622963578.170397][49] CHIP:ZCL: emberAfDefaultResponseCallback: Missing failure callback
[1622963578.170402][49] CHIP:DMG: ICR moving to [Uninitiali]
[1622963578.170442][49] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 0
[1622963578.170487][49] CHIP:EM: Received message of type 9 and protocolId 327680 on exchange 39025
[1622963578.170502][49] CHIP:DMG: InvokeCommand =
[1622963578.170506][49] CHIP:DMG: {
[1622963578.170509][49] CHIP:DMG: 	CommandList =
[1622963578.170512][49] CHIP:DMG: 	[
[1622963578.170514][49] CHIP:DMG: 		CommandDataElement =
[1622963578.170518][49] CHIP:DMG: 		{
[1622963578.170520][49] CHIP:DMG: 			CommandPath =
[1622963578.170523][49] CHIP:DMG: 			{
[1622963578.170526][49] CHIP:DMG: 				EndpointId = 0xe9,
[1622963578.170530][49] CHIP:DMG: 				ClusterId = 0x6,
[1622963578.170532][49] CHIP:DMG: 				CommandId = 0x1,
[1622963578.170535][49] CHIP:DMG: 			},
[1622963578.170539][49] CHIP:DMG: 			
[1622963578.170542][49] CHIP:DMG: 			StatusElement =
[1622963578.170545][49] CHIP:DMG: 			{
[1622963578.170547][49] CHIP:DMG: 				GeneralCode = 0xd,
[1622963578.170551][49] CHIP:DMG: 				ProtocolId = 0x0,
[1622963578.170578][49] CHIP:DMG: 				protocolCode = 0xffff,
[1622963578.170581][49] CHIP:DMG: 			},
[1622963578.170585][49] CHIP:DMG: 			
[1622963578.170587][49] CHIP:DMG: 		},
[1622963578.170595][49] CHIP:DMG: 		
[1622963578.170598][49] CHIP:DMG: 	],
[1622963578.170602][49] CHIP:DMG: 	
[1622963578.170604][49] CHIP:DMG: }
[1622963578.170696][49] CHIP:ZCL: DefaultResponse:
[1622963578.170716][49] CHIP:ZCL:   Transaction: 0x7f9c0a1f5898
[1622963578.170719][49] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
[1622963578.170722][49] CHIP:ZCL: emberAfDefaultResponseCallback: Missing success callback
[1622963578.170724][49] CHIP:ZCL: emberAfDefaultResponseCallback: Missing failure callback
[1622963578.170733][49] CHIP:EM: Sending Standalone Ack for MsgId:00000008
[1622963578.170751][49] CHIP:IN: Secure message was encrypted: Msg ID 16
[1622963578.170755][49] CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000000001 at utc time: 1411217 msec
[1622963578.170758][49] CHIP:IN: Sending secure msg on generic transport
[1622963578.170910][49] CHIP:IN: Secure msg send status No Error
[1622963578.170945][49] CHIP:EM: Flushed pending ack for MsgId:00000008
[1622963578.171726][36] CHIP:DMG: ICR moving to [Initialize]
[1622963578.171744][36] CHIP:DMG: ICR moving to [AddCommand]
[1622963578.171803][36] CHIP:IN: Secure message was encrypted: Msg ID 17
[1622963578.171810][36] CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000000001 at utc time: 1411218 msec
[1622963578.171813][36] CHIP:IN: Sending secure msg on generic transport
[1622963578.171844][36] CHIP:IN: Secure msg send status No Error
[1622963578.171849][36] CHIP:DMG: ICR moving to [   Sending]
[1622963578.172553][49] CHIP:ZCL: DefaultResponse:
[1622963578.172658][49] CHIP:ZCL:   Transaction: 0x7f9c0a1f5898
[1622963578.172664][49] CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
[1622963578.172667][49] CHIP:ZCL: emberAfDefaultResponseCallback: Missing success callback
[1622963578.172669][49] CHIP:ZCL: emberAfDefaultResponseCallback: Missing failure callback
[1622963578.172675][49] CHIP:DMG: ICR moving to [Uninitiali]
[1622963578.172742][49] CHIP:IN: Secure transport received message destined to fabric 0, node 0x000000000001B669. Key ID 0
[1622963578.172761][49] CHIP:EM: Received message of type 9 and protocolId 327680 on exchange 39026
[1622963578.172774][49] CHIP:DMG: InvokeCommand =
[1622963578.172777][49] CHIP:DMG: {
[1622963578.172780][49] CHIP:DMG: 	CommandList =
[1622963578.172783][49] CHIP:DMG: 	[
[1622963578.172786][49] CHIP:DMG: 		CommandDataElement =
[1622963578.172789][49] CHIP:DMG: 		{
[1622963578.172792][49] CHIP:DMG: 			CommandPath =
[1622963578.172795][49] CHIP:DMG: 			{
[1622963578.172799][49] CHIP:DMG: 				EndpointId = 0xe9,
[1622963578.172802][49] CHIP:DMG: 				ClusterId = 0x6,
[1622963578.172804][49] CHIP:DMG: 				CommandId = 0x0,
[1622963578.172807][49] CHIP:DMG: 			},
[1622963578.172811][49] CHIP:DMG: 			
[1622963578.172819][49] CHIP:DMG: 			StatusElement =
[1622963578.172840][49] CHIP:DMG: 			{
[1622963578.172844][49] CHIP:DMG: 				GeneralCode = 0xd,
[1622963578.172847][49] CHIP:DMG: 				ProtocolId = 0x0,
[1622963578.172850][49] CHIP:DMG: 				protocolCode = 0xffff,
[1622963578.172853][49] CHIP:DMG: 			},
[1622963578.172856][49] CHIP:DMG: 			
[1622963578.172858][49] CHIP:DMG: 		},
[1622963578.172862][49] CHIP:DMG: 		
[1622963578.172865][49] CHIP:DMG: 	],
[1622963578.172869][49] CHIP:DMG: 	
[1622963578.172871][49] CHIP:DMG: }

erjiaqing, 2021-06-07:
Caused by endpoint change.

erjiaqing, 2021-06-08:
Expected behavior since the corresponding test is

```
    FailIfNot(test.TestOnOffCluster(nodeid=1,
                                    endpoint=233,
                                    group=GROUP_ID), "Failed to test on off cluster on non-exist endpoint")
```

This is a expected behavior.

closed 2021-06-08 by erjiaqing