project-chip/connectedhomeip #7459: length attribute in cluster XML seems to be broken for cluster attributes of type ARRAY

bzbarsky-apple, 2021-06-08:
#### Problem
Consider this bit of XML we have in thread-network-diagnostics-cluster.xml:
```
    <attribute side="server" code="0x3B" define="SECURITY_POLICY" type="ARRAY" entryType="SecurityPolicy" length="1" writable="false" optional="false">SecurityPolicy</attribute>
```
I expect the author of that line expected it to mean "array with one entry".  What happens in practice is that the `typeSize` for this attribute ends up being set to `1` in ZAP.  As a result, the entry in `endpoint_config.h` inside `GENERATED_ATTRIBUTES` looks like this:
```
            { 0x003B, ZAP_TYPE(ARRAY), 1, 0, ZAP_EMPTY_DEFAULT() },                   /* SecurityPolicy */                         \
```
which seems very wrong: it's claiming this attribute is 1 _byte_ in size.  Whereas it should be either 2 bytes at least (to fit in the 16-bit length) or perhaps 5 bytes (length plus space for one element, and the element in this case is a 3-byte struct).

If I change the length in the XML to `3` instead of `1`, then I get a ZAP_LONG_DEFAULTS entry and that entry is 3 bytes long which once again seems wrong.

I expect this is what's causing the failures https://github.com/project-chip/connectedhomeip/pull/7278 is running into: that cluster has the only ARRAY attributes with `length="1"` in the tree, and they both fail in CI.  But oddly not when running locally....

 #### Proposed Solution
Figure out how to write a reliable test that will catch this, then fix the code generation.  I would _think_ that the right `typeSize` for an ARRAY should either always be 254 (which is what it is when you don't specify the length; not sure where that 254 is coming from) or should be `2 + length * entryType.typeSize` or something like that.  It should _definitely_ never be less than 2, since that's what we need to store the length.

@jmartinez-silabs @selissia @vivien-apple

selissia, 2021-06-09:
@bzbarsky-apple: Is there a specific XML schema used in CHIP? In Silabs XML schema an array is indicated by a boolean flag 
`<attribute side="server" code="0x0000" define="DEVICE_ID_LIST" type="INT16U" writable="false" optional="true" array="true">device id list</attribute>`

selissia, 2021-06-09:
I am told that currently ZAP does not support array-type attributes

bzbarsky-apple, 2021-06-09:
@selissia CHIP for now uses the XML quoted above, and generally speaking ZAP is supporting that, in that we are codegenning all sorts of code for such attributes right now....

What the official CHIP XML schema will look like is very much TBD.  :(

bzbarsky-apple, 2022-05-06:
This is no longer relevant because we are not storing lists in the attr store, and external things always get a typeSize of 0.

closed 2022-05-06 by bzbarsky-apple