project-chip/connectedhomeip #7460: [Cirque] Sometimes, EnableNetwork would have timeout issue

yunhanw-google, 2021-06-08:
#### Problem
Sometimes, EnableNetwork would have timeout issue

https://github.com/yunhanw-google/connectedhomeip/actions/runs/919164385

2021-06-08 18:04:23,560 [PythonMatterControllerTEST] INFO Send EnableNetwork command to device 1
[1623175463.560640][44] CHIP:DMG: ICR moving to [Initialize]
[1623175463.560707][44] CHIP:DMG: ICR moving to [AddCommand]
[1623175463.560804][44] CHIP:IN: Secure message was encrypted: Msg ID 9
[1623175463.560835][44] CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000000001 at utc time: 1065807 msec
[1623175463.560912][44] CHIP:IN: Sending secure msg on generic transport
[1623175463.561014][44] CHIP:IN: Secure msg send status No Error
[1623175463.561067][44] CHIP:DMG: ICR moving to [   Sending]
[1623175463.561164][56] CHIP:ZCL: DefaultResponse:
[1623175463.561222][56] CHIP:ZCL:   Transaction: 0x7fdd187790f8
[1623175463.561257][56] CHIP:ZCL:   status: EMBER_ZCL_STATUS_SUCCESS (0x00)
[1623175463.561322][56] CHIP:ZCL: emberAfDefaultResponseCallback: Missing success callback
[1623175463.561378][56] CHIP:ZCL: emberAfDefaultResponseCallback: Missing failure callback
[1623175463.561407][56] CHIP:DMG: ICR moving to [Uninitiali]
[1623175469.562966][56] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39022
2021-06-08 18:04:29,563 [PythonMatterControllerTEST] INFO Testing on off cluster
2021-06-08 18:04:29,563 [PythonMatterControllerTEST] INFO Sending On/Off commands to device 1 endpoint 1
[1623175469.563732][44] CHIP:DMG: ICR moving to [Initialize][1623175469.563734][56] CHIP:ZCL: DefaultResponse:
[1623175469.563791][56] CHIP:ZCL:   Transaction: 0x7fdd18779210

[1623175469.563798][56] CHIP:ZCL:   status: EMBER_ZCL_STATUS_FAILURE (0x01)
[1623175469.563827][44] CHIP:DMG: ICR moving to [AddCommand][1623175469.563847][56] CHIP:ZCL: emberAfDefaultResponseCallback: Missing success callback

[1623175469.563852][56] CHIP:ZCL: emberAfDefaultResponseCallback: Missing failure callback
[1623175469.563899][56] CHIP:DMG: ICR moving to [Uninitiali]
[1623175469.563936][44] CHIP:IN: Secure message was encrypted: Msg ID 10
[1623175469.563944][44] CHIP:IN: Sending msg from 0x000000000001B669 to 0x0000000000000001 at utc time: 1071810 msec
[1623175469.563951][44] CHIP:IN: Sending secure msg on generic transport
[1623175469.564029][44] CHIP:IN: Secure msg send status No Error
[1623175469.564035][44] CHIP:DMG: ICR moving to [   Sending]
[1623175475.566402][57] CHIP:DMG: Time out! failed to receive invoke command response from Exchange: 39023
2021-06-08 18:04:35,566 [PythonMatterControllerTEST] ERROR failed to send OnOff.On: error is 4050 with im responseNone
2021-06-08 18:04:35,566 [PythonMatterControllerTEST] CRITICAL Testfail: Failed to test on off cluster

 #### Proposed Solution
<suggested fix, suggested enhancement>

erjiaqing, 2021-06-09:
Also noticed this issue, might be an issue of ot-br-posix dbus interface.

erjiaqing, 2021-06-09:
@gjc13 Do you have any idea about this?
```
2021-06-08T12:35:07.897345406Z [1623155707.897267][55] CHIP:DMG: ICR moving to [   Sending]                                                                                                        
2021-06-08T12:35:07.897349065Z [1623155707.897279][55] CHIP:DMG: ICR moving to [Uninitiali]                                                                                                        
2021-06-08T12:35:07.906148585Z otbr-agent[50]: Handling method org.freedesktop.DBus.Properties.Set                                                                                                 
2021-06-08T12:35:07.906163730Z otbr-agent[50]: SetProperty io.openthread.BorderRouter.ActiveDatasetTlvs                                                                                            
2021-06-08T12:35:07.913451582Z otbr-agent[50]: [INFO]-MESH-CP-: Active dataset set                                                                                                                 
2021-06-08T12:35:07.913855085Z otbr-agent[50]: Replied to org.freedesktop.DBus.Properties.Set with result io.openthread.Error.OK                                                                   
2021-06-08T12:35:07.913880027Z otbr-agent[50]: Sending reply                                                                                                                                       
2021-06-08T12:35:07.913884468Z otbr-agent[50]: Reply queued                                                                                                                                        
2021-06-08T12:35:07.913888100Z otbr-agent[50]: [INFO]-CORE----: Notifier: StateChanged (0x101fc100) [KeySeqCntr Channel PanId NetName ExtPanId MstrKey PSKc SecPolicy ActDset]                     
2021-06-08T12:35:15.432395167Z [1623155715.432078][55] CHIP:DL: Long dispatch time: 7534 ms                                                                                                        
2021-06-08T12:35:15.432453177Z [1623155715.432205][55] CHIP:IN: Secure transport received message destined to fabric 0, node 0x0000000000000001. Key ID 0                                          
2021-06-08T12:35:15.432458855Z [1623155715.432253][55] CHIP:EM: Received message of type 16 and protocolId 0 on exchange 39022    
```
It waits for ~8seconds for the response.

gjc13, 2021-06-09:
Where does the line `2021-06-08T12:35:07.913884468Z otbr-agent[50]: Reply queued` come from?

erjiaqing, 2021-06-09:
> Where does the line `2021-06-08T12:35:07.913884468Z otbr-agent[50]: Reply queued` come from?
> 
> The client side may block at `dbus_connection_send_with_reply_and_block` which means the dbus message is lost?

I added some extra log on ot-br-posix, in `DBusRequest::ReplyOtResult`

From the log, seems the client does received the response, since dbus default timeout is 25s, but it only waits for 7.5s

yunhanw-google, 2021-06-11:
+++ b/src/dbus/client/thread_api_dbus.cpp
@@ -39,6 +39,7 @@
 namespace otbr {
 namespace DBus {
 
+#define DBUS_TIMEOUT_USE_DEFAULT_TEST 30000
 static ClientError NameToDeviceRole(const std::string &aRoleName, DeviceRole &aDeviceRole)
 {
     static std::pair<const char *, DeviceRole> sRoleMap[] = {
@@ -571,7 +572,7 @@ ClientError ThreadApiDBus::CallDBusMethodSync(const std::string &aMethodName)
     dbus_error_init(&error);
     VerifyOrExit(message != nullptr, ret = ClientError::ERROR_DBUS);
     reply = UniqueDBusMessage(
-        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT, &error));
+        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT_TEST, &error));
     VerifyOrExit(!dbus_error_is_set(&error), ret = DBus::ConvertFromDBusErrorName(error.message));
     VerifyOrExit(reply != nullptr, ret = ClientError::ERROR_DBUS);
     ret = DBus::CheckErrorMessage(reply.get());
@@ -589,7 +590,7 @@ ClientError ThreadApiDBus::CallDBusMethodAsync(const std::string &aMethodName, D
     DBusPendingCall * pending = nullptr;
 
     VerifyOrExit(message != nullptr, ret = ClientError::OT_ERROR_FAILED);
-    VerifyOrExit(dbus_connection_send_with_reply(mConnection, message.get(), &pending, DBUS_TIMEOUT_USE_DEFAULT) ==
+    VerifyOrExit(dbus_connection_send_with_reply(mConnection, message.get(), &pending, DBUS_TIMEOUT_USE_DEFAULT_TEST) ==
                      true,
                  ret = ClientError::ERROR_DBUS);
 
@@ -613,7 +614,7 @@ ClientError ThreadApiDBus::CallDBusMethodSync(const std::string &aMethodName, co
     VerifyOrExit(message != nullptr, ret = ClientError::ERROR_DBUS);
     VerifyOrExit(otbr::DBus::TupleToDBusMessage(*message, aArgs) == OTBR_ERROR_NONE, ret = ClientError::ERROR_DBUS);
     reply = DBus::UniqueDBusMessage(
-        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT, &error));
+        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT_TEST, &error));
     VerifyOrExit(!dbus_error_is_set(&error), ret = DBus::ConvertFromDBusErrorName(error.message));
     VerifyOrExit(reply != nullptr, ret = ClientError::ERROR_DBUS);
     ret = DBus::CheckErrorMessage(reply.get());
@@ -636,7 +637,7 @@ ClientError ThreadApiDBus::CallDBusMethodAsync(const std::string &           aMe
 
     VerifyOrExit(message != nullptr, ret = ClientError::ERROR_DBUS);
     VerifyOrExit(DBus::TupleToDBusMessage(*message, aArgs) == OTBR_ERROR_NONE, ret = ClientError::ERROR_DBUS);
-    VerifyOrExit(dbus_connection_send_with_reply(mConnection, message.get(), &pending, DBUS_TIMEOUT_USE_DEFAULT) ==
+    VerifyOrExit(dbus_connection_send_with_reply(mConnection, message.get(), &pending, DBUS_TIMEOUT_USE_DEFAULT_TEST) ==
                      true,
                  ret = ClientError::ERROR_DBUS);
 
@@ -667,7 +668,7 @@ ClientError ThreadApiDBus::SetProperty(const std::string &aPropertyName, const V
     VerifyOrExit(DBus::DBusMessageEncodeToVariant(&iter, aValue) == OTBR_ERROR_NONE, ret = ClientError::ERROR_DBUS);
 
     reply = DBus::UniqueDBusMessage(
-        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT, &error));
+        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT_TEST, &error));
 
     VerifyOrExit(!dbus_error_is_set(&error), ret = DBus::ConvertFromDBusErrorName(error.message));
     VerifyOrExit(reply != nullptr, ret = ClientError::ERROR_DBUS);
@@ -692,7 +693,7 @@ template <typename ValType> ClientError ThreadApiDBus::GetProperty(const std::st
     VerifyOrExit(message != nullptr, ret = ClientError::OT_ERROR_FAILED);
     otbr::DBus::TupleToDBusMessage(*message, std::tie(OTBR_DBUS_THREAD_INTERFACE, aPropertyName));
     reply = DBus::UniqueDBusMessage(
-        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT, &error));
+        dbus_connection_send_with_reply_and_block(mConnection, message.get(), DBUS_TIMEOUT_USE_DEFAULT_TEST, &error));
 
     VerifyOrExit(!dbus_error_is_set(&error), ret = DBus::ConvertFromDBusErrorName(error.message));
     VerifyOrExit(reply != nullptr, ret = ClientError::ERROR_DBUS);

yunhanw-google, 2021-06-23:
calling dbus_connection_send_with_reply_and_block in a thread and  another thread is blocking on dbus_connection_read_write_dispatch, the dbus connection maybe work properly.

yunhanw-google, 2021-06-23:
@kghost  as discussed, please post all set/get dbus properties call in thread stack management impl to the thread dbus mainloop's thread.

mspang, 2021-06-25:
> libdbus has multiple-thread stability issue.

Can you elaborate?

kghost, 2021-06-29:
Here is a message from Simon McVittie who is a contributor of dbus:

> DBusConnection is *meant to be* safe to share between threads, but we are
reasonably sure there are bugs in doing that, so it's safest to assume that
it is "owned" by one thread and should not be touched by other threads.

https://lists.freedesktop.org/archives/dbus/2017-September/017306.html

calls to `dbus_connection_send_with_reply_and_block` requires a `DBusConnection`, which is not thread-safe, I think it gives enough reason that we need to rework this part and do not use this API.

yunhanw-google, 2021-07-07:
https://github.com/project-chip/connectedhomeip/pull/7913

closed 2021-07-07 by yunhanw-google

bukepo, 2021-07-09:
Just found the root cause of this issue. The dbus mainloop in CHIP gives little chance for other threads to acquire the mutex of dbus connection. The issue is gone if adding `sleep(0);` after the following line:
https://github.com/project-chip/connectedhomeip/blob/9c01258cbd4342d2ca45f0d5912e3cfc3d4efe25/src/platform/Linux/ThreadStackManagerImpl.cpp#L101