project-chip/connectedhomeip #7468: Fix chip-tool's race conditions

mrjerryjohns, 2021-06-08:
#### Problem
 * chip-tool has a number of race conditions due to the main thread interacting with the CHIP thread while the stack itself is running on its own thread.
 
 #### Proposed Solution
* Adopting the approach of 'external synchronization' approach being espoused in #6841.

mrjerryjohns, 2021-06-08:
Using tsan to help validate the fixes.

TheCodeYoda, 2021-06-09:
How do you reproduce this issue ? [preferably using esp32-devkitC]

mrjerryjohns, 2021-06-09:
I've been building chip-tool for Mac, not sure if chip-tool is usable on other platforms. As long as that platform uses clang to compile, tsan should be usable.

1. Configure GN using 'is_tsan=true' passed in as args when configuring to build chip-tool.
2. Build normally.

When you run chip-tool there-after, should emit a ton of logs from tsan that show exactly where the race conditions appear:

Sample race (this is in the lighting-app though...):

```
WARNING: ThreadSanitizer: data race (pid=28651)
  Write of size 8 at 0x000105e54c18 by thread T1:
    #0 std::__1::deque<chip::DeviceLayer::ChipDeviceEvent, std::__1::allocator<chip::DeviceLayer::ChipDeviceEvent> >::push_back(chip::DeviceLayer::ChipDeviceEvent const&) deque:1933 (chip-all-clusters-app:x86_64+0x1000c85e4)
    #1 std::__1::queue<chip::DeviceLayer::ChipDeviceEvent, std::__1::deque<chip::DeviceLayer::ChipDeviceEvent, std::__1::allocator<chip::DeviceLayer::ChipDeviceEvent> > >::push(chip::DeviceLayer::ChipDeviceEvent const&) queue:305 (chip-all-clusters-app:x86_64+0x1000c5808)
    #2 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_PostEvent(chip::DeviceLayer::ChipDeviceEvent const*) GenericPlatformManagerImpl_POSIX.cpp:135 (chip-all-clusters-app:x86_64+0x1000c57b1)
    #3 chip::DeviceLayer::PlatformManager::PostEvent(chip::DeviceLayer::ChipDeviceEvent const*) PlatformManager.h:281 (chip-all-clusters-app:x86_64+0x1000c5092)
    #4 chip::DeviceLayer::Internal::GenericPlatformManagerImpl<chip::DeviceLayer::PlatformManagerImpl>::_ScheduleWork(void (*)(long), long) GenericPlatformManagerImpl.cpp:198 (chip-all-clusters-app:x86_64+0x1000c4fe3)
    #5 chip::DeviceLayer::PlatformManager::ScheduleWork(void (*)(long), long) PlatformManager.h:230 (chip-all-clusters-app:x86_64+0x10003219a)
    #6 chip::Shell::Engine::RunMainLoop() MainLoopDefault.cpp:193 (chip-all-clusters-app:x86_64+0x100121746)
    #7 ChipLinuxAppMainLoop()::$_0::operator()() const AppMain.cpp:119 (chip-all-clusters-app:x86_64+0x10005abfc)
    #8 decltype(std::__1::forward<ChipLinuxAppMainLoop()::$_0>(fp)()) std::__1::__invoke<ChipLinuxAppMainLoop()::$_0>(ChipLinuxAppMainLoop()::$_0&&) type_traits:3747 (chip-all-clusters-app:x86_64+0x10005ab40)
    #9 void std::__1::__thread_execute<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, ChipLinuxAppMainLoop()::$_0>(std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, ChipLinuxAppMainLoop()::$_0>&, std::__1::__tuple_indices<>) thread:280 (chip-all-clusters-app:x86_64+0x10005a9e8)
    #10 void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, ChipLinuxAppMainLoop()::$_0> >(void*) thread:291 (chip-all-clusters-app:x86_64+0x100059b0e)

  Previous read of size 8 at 0x000105e54c18 by main thread (mutexes: write M311):
    #0 std::__1::deque<chip::DeviceLayer::ChipDeviceEvent, std::__1::allocator<chip::DeviceLayer::ChipDeviceEvent> >::empty() const deque:1403 (chip-all-clusters-app:x86_64+0x1000cc9ba)
    #1 std::__1::queue<chip::DeviceLayer::ChipDeviceEvent, std::__1::deque<chip::DeviceLayer::ChipDeviceEvent, std::__1::allocator<chip::DeviceLayer::ChipDeviceEvent> > >::empty() const queue:291 (chip-all-clusters-app:x86_64+0x1000c6108)
    #2 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::ProcessDeviceEvents() GenericPlatformManagerImpl_POSIX.cpp:142 (chip-all-clusters-app:x86_64+0x1000c6079)
    #3 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::SysProcess() GenericPlatformManagerImpl_POSIX.cpp:214 (chip-all-clusters-app:x86_64+0x1000c5e01)
    #4 chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop() GenericPlatformManagerImpl_POSIX.cpp:228 (chip-all-clusters-app:x86_64+0x1000c58a9)
    #5 chip::DeviceLayer::PlatformManager::RunEventLoop() PlatformManager.h:235 (chip-all-clusters-app:x86_64+0x100059642)
    #6 ChipLinuxAppMainLoop() AppMain.cpp:124 (chip-all-clusters-app:x86_64+0x100059593)
    #7 main main.cpp:58 (chip-all-clusters-app:x86_64+0x100004fbd)

  Location is global 'chip::DeviceLayer::PlatformManagerImpl::sInstance' at 0x000105e54a08 (chip-all-clusters-app+0x00010013cc18)
```

TheCodeYoda, 2021-06-10:
Does this happend with the all-cluster-app (while running ./chip-tool onoff toggle 1)?

woody-apple, 2022-01-25:
Is there anything outstanding here? Please re-open if so!

closed 2022-01-25 by woody-apple