project-chip/connectedhomeip #7486: [nrfconnect] Thread network cannot consistently attach after CHIP provisioning

wgtdkp, 2021-06-09:
#### Problem
With both the latest OTBR and nrfconnect lighting app, the lighting app jumps between attached and
detached states after CHIP provisioning and finally stays in detach state.

------

 #### Reproduce
------
- OTBR commit: https://github.com/openthread/ot-br-posix/commit/faed2f0929121c1525b70af9c66aff7b2c9c80bc
- ot-nrf528xx commit (make `RCP` for OTBR): https://github.com/openthread/ot-rf528xx/commit/b00c50fed9b8353719ab1807f9d780fb595490b3
- nrf52840 dongle for use with OTBR: `PCA10059 1.0.0`.
- CHIP commit: 35b82baa08d82c09eea6b52ccff69dd10efbecab
- nRF52840 DK for building CHIP lighting app: `PCA10056 0.9.3`
------
1. Build and setup OTBR with the given OTBR and ot-nrf528xx commits by following https://openthread.io/codelabs/openthread-border-router.
2. Build the CHIP lighting app with the given commit by following https://github.com/project-chip/connectedhomeip/blob/master/examples/lighting-app/nrfconnect/README.md
3. Build and install the Android CHIPTool by following https://github.com/project-chip/connectedhomeip/blob/master/src/android/CHIPTool/README.md
4. Start the CHIP provisioning flow and save same Thread network parameters of OTBR to the lighting app.
5. The lighting app with attach as a `CHILD` and then go back to detached state repeats for several times.

------
Please find the logs and traffics:
[chip-lighting-app-nrfconnect-detached-log-traffic.zip](https://github.com/project-chip/connectedhomeip/files/6622913/chip-lighting-app-nrfconnect-detached-log-traffic.zip)

Damian-Nordic, 2021-06-10:
@wgtdkp I just tested it using the same commits you listed (thanks for very detailed description by the way!) and everything was very stable for me. Is it possible that you had other network nodes around which might interfere? What was the distance between the lighting device and OTBR during your testing?

wgtdkp, 2021-06-11:
Thanks @Damian-Nordic for your verification!

> Is it possible that you had other network nodes around which might interfere?

I scanned the Thread channel before forming network and selected an idle channel so I think it shouldn't be an issue.

> What was the distance between the lighting device and OTBR during your testing?

They are both by my side (distance < 0.5m).

@Damian-Nordic Would it be possible for you to share your firmwares (RCP, CHIP) and APK (I am using `arm64` arch) ? I am trying to verify if this is a hardware issue, Thanks!

Below are my firmwares:
[issue#7486-firmwares.zip](https://github.com/project-chip/connectedhomeip/files/6635405/issue.7486-firmwares.zip)

wgtdkp, 2021-06-16:
@Damian-Nordic a friendly reminder :)

Damian-Nordic, 2021-06-16:
Ah, sorry! :) Looking into it.

Damian-Nordic, 2021-06-16:
I used your firmware versions for all the components (except the Android CHIPTool as my low-end phone has 32-bit CPU :)) and everything is super stable even though I have the border router in a different room. I'll examine the PCAP you provided for some clue, but I noticed that I have newer nRF52840 DKs (2.0.1 revision). Do you have other DKs around to test if this could be a hardware issue? If not I'll try to find older ones, too and verify.

wgtdkp, 2021-06-16:
Thanks @Damian-Nordic  I also tried version `1.1.0` and the same issue.

Damian-Nordic, 2021-06-16:
I wonder if we could rule out Matter completely by using the OpenThread CLI sample. Could you maybe program the `ot-cli-ftd` image onto the nRF52840 DK and try to join the OTBR network using commands like:
```
ifconfig up
dataset channel 11
dataset panid 0x1234
dataset extpanid 1111111122222222
dataset networkname "Test"
dataset masterkey 00112233445566778899aabbccddeeff
dataset activetimestamp 1
dataset commit active
thread start
```
?

Basically, I would like to reduce the number of moving parts and narrow down the issue.

wgtdkp, 2021-06-17:
> Thanks @Damian-Nordic I also tried version 1.1.0 but the same issue (sorry I don't have newer boards).

I must apologize that I was using `1.1.0` DK as RCP as I didn't have two `1.1.0` DKs :(

This issue disappears when I switched to use `1.1.0` DK for the lighting app. @Damian-Nordic Thanks for your support!

closed 2021-06-17 by wgtdkp